zephyrproject-rtos/zephyr #13700: how to use math.h functions like ceil or floor?

zl3one6, 2019-02-23:
I notice that the zephyr project uses minimal c library? So I don't understand how to let zephyr support math.h which is from standard c library. I tried to enable the new libc in menuconfig, but when build the project, it shows an error about -ENOTSUP in kernel.h. The sample I used is the hello world sample. 

Thanks

SebastianBoe, 2019-02-25:
Can you post steps to reproduce and the error produced?

Note that you need to enable CONFIG_FLOAT when using the 'double' functions in math.h.

I got this error when I forgot to enable FLOAT, and a succesful run when it was enabled:

```
Booting from ROM..***** Booting Zephyr OS v1.14.0-rc1-511-g23ba884b2b *****
***** Floating point unit not enabled
Current thread ID = 0x00400060
eax: 0x00000206, ebx: 0x00000000, ecx: 0x00000000, edx: 0x00400060
esi: 0x00000000, edi: 0x00001d56, ebp: 0x004005d4, esp: 0x004005a0
eflags: 0x00000206 cs: 0x0008
call trace:
eip: 0x00001367
     0x00002e3b (0x4005f0)
     0x00001d65 (0x0)
Fatal fault in essential thread! Spinning...
Terminate emulator due to fatal kernel error
FAILED: zephyr/CMakeFiles/run 

```

zl3one6, 2019-03-05:
Thanks, I tried to compile it on Ubuntu environment and enable new lib, it passed.

closed 2019-03-05 by zl3one6