zephyrproject-rtos/zephyr #13701: x86 stack check fail w/posix-lib & newlib

galak, 2019-02-23:
When trying to run the posix tests w/newlib as enabled by PR #13685 

`./scripts/sanitycheck -T tests/posix/ -p qemu_x86`

We get:

SDK 0.9.5:
```
SeaBIOS (version rel-1.11.2-0-gf9626ccb91-prebuilt.qemu-project.org)
Booting from ROM..***** Booting Zephyr OS v1.14.0-rc1-501-ge03b7fad46 *****
***** Double Fault *****
***** Stack Check Fail! *****
Current thread ID = 0x00403160
eax: 0x00000001, ebx: 0x004011e0, ecx: 0x00000000, edx: 0x0043e00c
esi: 0x00401020, edi: 0x00400000, ebp: 0x004011e0, esp: 0x0043e000
eflags: 0x00000202 cs: 0x0008
call trace:
eip: 0x00002cc7
     corrupted? (bp=0x004011e0)
Fatal fault in essential thread! Spinning...
Terminate emulator due to fatal kernel error
```

SDK 0.10.0:
```
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS v1.14.0-rc1-501-ge03b7fad46 *****
***** Double Fault *****
***** Stack Check Fail! *****
Current thread ID = 0x00403160
eax: 0x00000001, ebx: 0x0043e064, ecx: 0x00000000, edx: 0x00000012
esi: 0x004011dc, edi: 0x0043e060, ebp: 0x0043e040, esp: 0x0043dfd8
eflags: 0x00000202 cs: 0x0008
call trace:
eip: 0x000026f3
     0x0000275e (0x401020)
     0x00001e26 (0x401020)
     0x000016ed (0x401020)
     0x00001568 (0x401020)
     0x0000a616 (0x14c4f)
     0x00003255 (0x124bd)
     0x0000a7f0 (0x4430ec)
     0x0000ac27 (0x43e1e4)
Fatal fault in essential thread! Spinning...
Terminate emulator due to fatal kernel error
```

nashif, 2019-02-23:
@galak why was this merged although it is known to be failing? Now all CI jobs are failing because of this.

galak, 2019-02-23:
> @galak why was this merged although it is known to be failing? Now all CI jobs are failing because of this.

My bad, wasn't thinking about the CI impact to everything else.

Is there someone who can look at the x86 issue.

andrewboie, 2019-02-24:
The error message is correct; this is a stack overflow in _main_thread.
It appears if CONFIG_ZTEST is enabled, CONFIG_MAIN_STACK_SIZE defaults to 512, which is not enough.

andrewboie, 2019-02-24:
@galak I've unpeeled this enough to see what is going on.

For this test, CONFIG_STDOUT_CONSOLE=y, which causes TC_PRINT() and related macros to map to libc printf() instead of the kernel's printk(). Newlib's printf() uses tons of stack space, even more than minimal libc, and LOTS more than printk(), and we are blowing stack in _ztest_run_test_suite() before the test can even run.

For some reason, CONFIG_STDOUT_CONSOLE is defaulting to Y when newlib is enabled:

```
config STDOUT_CONSOLE
	bool "Send stdout to console"
	depends on CONSOLE_HAS_DRIVER
	default NEWLIB_LIBC
	help
	  This option directs standard output (e.g. printf) to the console
	  device, rather than suppressing it entirely. See also EARLY_CONSOLE
	  option.
```

I don't even know why CONFIG_STDOUT_CONSOLE exists, or why we have it manually turned on for a lot of tests. The help for this is simply wrong; AFAICT printf() always works, and instead this option actually controls whether certain testcase related macros map to printf() or printk(). I don't see what we gain by this, why not just use printk() always??

PR #13723 (2019-02-26): libc: fix CONFIG_STDOUT_CONSOLE semantics
The intent of this Kconfig is to allow libc stdout
functions like printf() to send their output to the
active console driver instead of discarding it.

This somehow evolved into preferring to use
printf() instead of printk() for all test case output
if enabled. Libc printf() implementation for both
minimal libc and newlib use considerably more stack
space than printk(), with nothing gained by using
them.

Remove all instances where we are conditionally
sending test case output based on this config, enable
it by default, and adjust a few tests that disa [... cut]
commit feab3709 (2019-02-26): libc: fix CONFIG_STDOUT_CONSOLE semantics

The intent of this Kconfig is to allow libc stdout
functions like printf() to send their output to the
active console driver instead of discarding it.

This somehow evolved into preferring to use
printf() instead of printk() for all test case output
if enabled. Libc printf() implementation for both
minimal libc and newlib use considerably more stack
space than printk(), with nothing gained by using
them.

Remove all instances where we are conditionally
sending test case output based on this config, enable
it by default, and adjust a few tests that dis [... cut]
closed 2019-02-26 by galak