zephyrproject-rtos/zephyr #13703: Build error w/Flash driver enabled on hexiwear kw40

galak, 2019-02-23:
```
mkdir -p samples/drivers/flash_shell/build
cd samples/drivers/flash_shell/build
cmake -DBOARD=hexiwear_kw40z ..
make
```

We get:

```
make[3]: *** [zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/build.make:63: zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/s
make[3]: Target 'zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/build' not remade because of errors.
make[3]: Leaving directory '/home/galak/git/zephyr/sanity-out/hexiwear_kw40z/samples/drivers/flash_shell/test'
make[2]: *** [CMakeFiles/Makefile2:2254: zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/all] Error 2
[ 91%] Linking C static library app/libapp.a
make[3]: Leaving directory '/home/galak/git/zephyr/sanity-out/hexiwear_kw40z/samples/drivers/flash_shell/test'
[ 91%] Built target app
[ 92%] Linking C static library libkernel.a
make[3]: Leaving directory '/home/galak/git/zephyr/sanity-out/hexiwear_kw40z/samples/drivers/flash_shell/test'
[ 92%] Built target kernel
In file included from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_features.h:21:0,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_controller.h:13,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_flash.h:13,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_flash.c:10:
/home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_adapter.h:302:0: error: "SIM_FCFG1_REG" redefined [-Werror]
 #define SIM_FCFG1_REG SIM->FCFG1
 
In file included from /home/galak/git/zephyr/ext/hal/nxp/mcux/devices/MKW40Z4/fsl_device_registers.h:44:0,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_common.h:22,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_features.h:18,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_controller.h:13,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_flash.h:13,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_flash.c:10:
/home/galak/git/zephyr/ext/hal/nxp/mcux/devices/MKW40Z4/MKW40Z4.h:6829:0: note: this is the location of the previous definition
 #define SIM_FCFG1_REG(base)                      ((base)->FCFG1)
 
In file included from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_features.h:21:0,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_controller.h:13,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_controller.c:10:
/home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_adapter.h:302:0: error: "SIM_FCFG1_REG" redefined [-Werror]
 #define SIM_FCFG1_REG SIM->FCFG1
 
In file included from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_features.h:21:0,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_controller.h:13,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_flash.h:13,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_flash.c:10:
/home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_flash.c: In function ‘FLASH_Init’:
/home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_adapter.h:376:55: error: ‘FSL_FEATURE_FLASH_PFLASH_PROTECTION_REGIO
 #define FLASH0_FEATURE_PFLASH_PROTECTION_REGION_COUNT FSL_FEATURE_FLASH_PFLASH_PROTECTION_REGION_COUNT
                                                       ^
/home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_adapter.h:376:55: note: in definition of macro ‘FLASH0_FEATURE_PFLA
 #define FLASH0_FEATURE_PFLASH_PROTECTION_REGION_COUNT FSL_FEATURE_FLASH_PFLASH_PROTECTION_REGION_COUNT
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_adapter.h:376:55: note: each undeclared identifier is reported only
 #define FLASH0_FEATURE_PFLASH_PROTECTION_REGION_COUNT FSL_FEATURE_FLASH_PFLASH_PROTECTION_REGION_COUNT
                                                       ^
/home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_adapter.h:376:55: note: in definition of macro ‘FLASH0_FEATURE_PFLA
 #define FLASH0_FEATURE_PFLASH_PROTECTION_REGION_COUNT FSL_FEATURE_FLASH_PFLASH_PROTECTION_REGION_COUNT
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/galak/git/zephyr/ext/hal/nxp/mcux/devices/MKW40Z4/fsl_device_registers.h:44:0,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_common.h:22,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_features.h:18,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_controller.h:13,
                 from /home/galak/git/zephyr/ext/hal/nxp/mcux/drivers/kinetis/fsl_ftfx_controller.c:10:
/home/galak/git/zephyr/ext/hal/nxp/mcux/devices/MKW40Z4/MKW40Z4.h:6829:0: note: this is the location of the previous definition
 #define SIM_FCFG1_REG(base)                      ((base)->FCFG1)
 
```

galak, 2019-02-23:
Will possible block merging #13702

agansari, 2019-02-25:
@galak this errors come from different mcux sdk versions.
@MaureenHelm this should happen to kw30/20 as well, we could add a legacy_support.h file in ext/hal/nxp/mcux/devices/MKW40Z4 folder where we can undefine and define newer sdk changes or just keep multiple skds.

MaureenHelm, 2019-02-25:
> Will possible block merging #13702

You can use `depends_on` instead of a Kconfig filter, and then the PR won't be blocked.

> @galak this errors come from different mcux sdk versions.
> @MaureenHelm this should happen to kw30/20 as well, we could add a legacy_support.h file in ext/hal/nxp/mcux/devices/MKW40Z4 folder where we can undefine and define newer sdk changes or just keep multiple skds.

I'm considering removing the KW{40,30,20} altogether.

galak, 2019-02-26:
> You can use `depends_on` instead of a Kconfig filter, and then the PR won't be blocked.
> 
> >

`depends_on` what?

Even then we'd still have a build issue if flash is enabled on KW40.

MaureenHelm, 2019-02-26:
> `depends_on` what?

flash? soc_flash?

> Even then we'd still have a build issue if flash is enabled on KW40.

Not if we don't list flash in the supported list for kw40.

galak, 2019-02-26:
> Not if we don't list flash in the supported list for kw40.

If the driver isn't valid or supported on KW40, than I'd rather we fix Kconfig to say as much.  We don't have 'flash' or 'soc_flash' in the board yaml and I'd rather not add it.

galak, 2019-03-07:
Fixed by PR #14118

closed 2019-03-07 by galak