zephyrproject-rtos/zephyr #13708: No Arduino interface definition for Nordic dev. kits

overheat, 2019-02-24:
**Describe the bug**
When I try to compile "samples\shields\x_nucleo_iks01a1\" for my Nordic dev. kits(pca10040 and pca10056) on my hand, occurred this error:

```
Error: nrf52840_pca10056.dts.pre.tmp:437.1-13 Label or path arduino_i2c not found
FATAL ERROR: Syntax error parsing input tree
...
Call Stack (most recent call first):
...

-- Configuring incomplete, errors occurred!
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52840_pca10056
3. make
4. See error

**Screenshots or console output**
[image]

** Environment:**
 - OS: ( Windows)
 - Toolchain (Zephyr SDK, ...)
 - Commit SHA or Version used: 1.14-rc1

overheat, 2019-02-24:
I noticed that the Arduino interface, like default i2c and so on, not been included in the Nordic dev. kits, either pca10040 or pca10056. So I add this patch to nrf52_pca10040.dts and nrf52840_pca10056.dts:

```
@@ -97,7 +97,7 @@
        cts-pin = <7>;
 };

-&i2c0 {
+arduino_i2c: &i2c0 {
        status = "ok";
        sda-pin = <26>;
        scl-pin = <27>;
```
And it works.

Should every Nordic development board add the Arduino interface definition?

tanreb1, 2019-02-25:
That' s look good, but I think we should add all arduino interfaces. Not only i2c, but spi and uart too. 

+++ ../boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts
@@ -74,6 +74,10 @@
+arduino_i2c: &i2c0 {
    status = "ok";
};
+arduino_spi: &spi1 {
    status = "ok";
};
+arduino_serial: &uart0 {
    status = "ok";
};

overheat, 2019-02-28:
Hi,

This PR fix the issue and works fine right for my X-NUCLEO-IKS01A1 and 
nrf52840_pca10056!
[image]

As I also have pca10040 and pca10028 on my hand, I think I can help to fix the same issue on those boards.

carlescufi, 2019-11-13:
This should be fixed now.

closed 2019-11-13 by carlescufi