zephyrproject-rtos/zephyr #13718: [Feature Request] Updated Sensor API

microbuilder, 2019-02-24:
# Updated Sensor API

This issue is an attempt to start identifying improvements that
can be made to the current sensor API in Zephyr, with the goal of defining a
new API that will eventually replace the existing one.

A more formal proposal will be made in due time, but any weaknesses in the
current API can be listed here, or requirements for a new API, and based on
the discussion a specific sensor API proposal can be made if that path is
deemed useful of necessary.

The current proposals are based on existing issues and discussion during the
weekly API call, and are subject to revision.

## Key Revised API Goals

Two-tier sensor data API:

- Raw data at the lowest level:
  - Memory efficient (Two mandatory 32-bit values today)
  - Low latency
  - Higher processing burden in user land
- Standardised SI-based output at the higher-level:
  - Higher per-sample memory use
  - Higher latency
  - Low user processing burden (standard SI units and scale, filtering, etc.)
  - Smarter event callbacks based on richer data filtering, etc.

Both tiers must be implemented in a valid sensor driver.

## General Requirements

### Timestamp API

- Timestamping API for individual samples, required for DSP filters, logs, etc.

### Flexible Execution Context

- Allow execution context to be specified by user, with a goal of enabling
  a single thread or limited thread count for all sensor activity.

### Internal FIFO Management

- Internally cache sensor data via FIFO buffer, with threshold/full alert(s).
  Allows more efficient resource usage, lower user-burden.

### Sensor Data

- Enable raw data for smallest footprint with higher user burden.
- Drivers MUST implement raw->stdtype conversion function(s).
- Determine sensible SI unit/scale per channel or sensor type
  - Should scale be adjustable at the KConfig level? Auto-scaling?
- Where possible, flag missed sample(s) event between reads.
- Optionally associate variable meta-data (die temp, etc.) with sample.
  - Memory an issue here
  - Can be a pointer in the data struct that is `NULL` when metadata not
    present, and singly-linked list when present, though this still costs 4
    bytes per sample min.:
    `if (sdat->meta != NULL) { /* Traverse meta data slist. */};`

### Driver Attributes

A flexible, generic system is required here, though further discussion is
needed to define the initial attribute list.

- Review the current *attribute* system to accommodate more device types
  (integration time, etc.).
- Should a mechanism for *driver-defined* attributes be included, documented in
  the specific driver? `SENSOR_ATTR_USER_DEF_1`, `SENSOR_ATTR_USER_DEF_2`, etc.

### Triggers

- Review and revise current overly-narrow list of trigger events.
- Use a singly-linked list of callbacks that can be traversed asynchronously,
  starting in the IRQ handler but with the blocking SPI/I2C/ADC calls
  elsewhere. An slist allows this to be dynamically adjusted at runtime.

### Channels

- Allow multiple instances of a channel.
- How to assign meta-data to channels (ex. associate a wavelength value or
  range to an instance of the `light` channel, etc.)?

### Features API?

There is currently no mechanism to benefit from device-specific functions
(sensor fusion, etc.), which is a major limitation designing resource-efficient
devices.

- How to generically expose features like sensor fusion, tap-detection, etc.?
- Singly-linked list of device-specific 'features' with some sort of numeric
  identifier for a feature that is described at the driver level ('Feature 1'
  = sensor fusion).
- How to handle parameters for each *feature*?
- Enable device specific user function overrides on raw data, etc. NULL goes
  to the internal function, can be overridden by the end over though.

### Calibration API?

- Common API to calculate, load/store and apply calibration coefficients?

### Logging/Storage API?

- Centralised logging API for sensor data to make storing and off-loading
  sensor data at a later date easier? Persist data in a standard format to
  NVM, etc.?

### Device Simulation?

While perhaps not restricted to the sensor API, a design goal should be
enabling simulation of specific sensors when HW isn't available, or specific
edge cases are difficult to reproduce on real HW (to properly test algorithms
or application code that relies on sensor data).

- Allow a sensor driver to 'register' itself at a specific I2C address, and
  in a simulated environment when a generic I2C read/write request is made, the
  I2C handler can redirect the request to the simulated function in the driver.
- The following functions would be required for I2C sensors, for example:
  - Read request handler
  - Write request handler
  - Init function to register the simulated device with the generic I2C layer.

> For an example of this using I2C, see [mcu_sim_i2c.h](https://github.com/apache/mynewt-core/blob/master/hw/mcu/native/include/mcu/mcu_sim_i2c.h) from Apache
  Mynewt, and [tsl2591_sim.c](https://github.com/apache/mynewt-core/blob/master/hw/drivers/sensors/tsl2591/src/tsl2591_sim.c) for a partial implementation of
  this concept.

## Related Issues

#1387 : Lack of a performant sensor API
#13455 : Improvements to `drivers/sensors` for light sensors

bfrog, 2019-02-25:
I think looking at the linux IIO subsystem and how it works would be very useful in designing a performant at usable API for both drivers and applications.

Some nice things IIO does from my day exploration into it

* Static list of channel per device driver to describe said channels
* Channels may be read or write
* Attach buffers to channels with a pollable buffer, may be circular or not.
* Timestamps are optionally there per sample
* Direct DMA transfers to the buffer are possible from the devices, avoiding unnecessary copying. I think this is really a key point in zephyr land, as it would reduce memory usage and improve performance by offloading work to the hardware when possible.
* Each channel and device has a set of attributes, accessible by char* names, which may be manipulated.

Since each device describes its own channels with a static array of structs channel descriptors, problems relating to the current API limiting channels to 1 one of each kind is no longer a problem.

microbuilder, 2019-02-25:
Thanks for the feedback. I agree with most of the comments above personally, and I'll dig into IIO a bit on my end having never worked with it. This doesn't address the **Features API** idea, which I think needs to be addressed, but it definitely seems like a good starting point architecturally.

pabigot, 2019-02-25:
I'm generally in favor of re-using existing architectural solutions like IIO, but I'm more in favor of starting with a clear understanding of stakeholder needs.  It may be that IIO is based on assumptions that don't apply to the application domains that Zephyr's intended to support.

My primary application is low-power sensors.  I always know what sensor (HTU21D, BME280, SDP810, ...) is providing the readings, so I really don't care about the ability to ask for a temperature or a pressure independent of what provides the value.

I want to collect at a relatively low frequency (at most 1 Hz) or on detected change, so while some DMA features might be useful if they get in my way they're not wanted.

I want a single observation structure that provides a space-efficient representation of all the measurements produced by that specific device, so I know (for example) that the relative humidity I'm using was captured synchronously with the temperature I'm using, or that the three axes of acceleration were from the same observation.  My application will know whether the pressure is Pa in a `uint32_t` or cPa in an `int16_t`.

I want notification of relevant events: new observation, threshold change, calibration completion.  At this time I want that through a callback invoked from the GPIO or alarm interrupt.  I'll decide if I want to do the handoff to the application through kpoll, kwork, or simply setting a flag to be processed by the main loop, and whether I need to synchronously do some other change (such as reconfiguring the interrupt management or triggering another observation).  If I need to timestamp a captured observation I'll do that myself, since I know what clock will provide the precision and range I'll want to use for the specific application.

I want full control of the sensor's specific capabilities.  Proper use of a CCS811 involves tracking and restoring baseline values at various points in the device and application life cycle, as well as updating stored environmental conditions.  That's not going to be supported by a generic API in any usable manner.  See #11993, as well as #12056 which it motivated.

In short, I think Zephyr's too hung up on providing cross-platform/cross-device API.  All that abstraction costs code, RAM, and power, and I believe many embedded use cases won't want to pay for it.  Perhaps I'm wrong.

My [wip/sensor2 branch](https://github.com/pabigot/zephyr/commits/wip/sensor2) has my initial very rough thoughts from some weeks ago.  I haven't looked at it recently, but some of it might be useful.

bfrog, 2019-02-25:
I actually disagree somewhat on the Feature API, Features per device could be done via named attributes. It would require some devices providing their named attributes in a public header. In some scenarios there might be additional channels of information that can be subscribed to as well, which themselves allow the device user to take advantage of those features.

Sensor Fusion example might be AHRS channels that are available but disabled by default. Enabling one, enables them all (they are tied together). The driver is responsible for reconfiguring the device with sensor fusion enabled and filling the buffer appropriately.

In a similar manner gestures might be a channel of enumerated data with a timestamp given. So again additional channels of information are available from the device, like I imagine numerous TDK 9 axis sensors, which are feature rich IMU chips, might have the following channels...

* die temp (1 channel)
* accel x,y,z (3 channels)
* gyro x,y,z (3 channels)
* magn x,y,z (3 channels)
* aux (5 channels)
* ahrs (3 channels) 
* gestures (1 channel with int values describing gesture)
* pedometer (1 channel, with a timer trigger)

Now enabling/disabling these channels provides nearly all the functionality of the chip available with a pollable buffer and optionally timestamps, by enabling or disabling those channels.

Certainly there's some attributes for groups of these channels that should be mutable in some way.

* Gyro, Accel, Mag filtering settings
* Gyro, Accel, Mag range settings
* Gyro, Accel, Mag rate settings
* Aux (I2C master) register settings
* Gestures to notify of

Those could be attributes available on a channel or set of channels. Commonly named ones could be defined in a common header. Device specific ones would have to be exported by the driver in a different header, which is perfectly sensible.

bfrog, 2019-02-25:
@pabigot I think using name, typified attributes solves most of those problems. I agree, notification of events in a convienent and compact manner is important. Doing things in a callback is ok, but in Zephyr the context of the callback matters and that's partially why there's the trigger options now at compile time for k_work or k_thread callback contexts. Providing a callback in the IRQ context leads to problems in my experience as I experimented with that. I think channels that are pollable are quite workable, would allow the kernel to sleep saving power, and provide all the potential event notifications we could want.

microbuilder, 2019-02-25:
@pabigot I think this highlights why we need a two-tier approach to sensor data. There are certainly many occasions where you have full knowledge and control over the hardware you are using -- gain, integration time, etc., are known -- and memory efficiency is primary, in which case you should opt for a *RAW* channel and interface. I would propose that all sensor drivers, for example, should have a mandatory **channel 0** that is always the raw data.

But there are situations where a high level of abstraction is definitely useful, especially in networks of long-lived devices. A key problem in the embedded space is obviously parts availability, and sensors that may be available today may be EOL, or with long lead times later, meaning I need to substitute in a replacement part to ship a new batch of devices. A high level of abstraction makes it relatively easy to replace accelerometer A with model B, C, or D with very little overall impact on the end node itself, or on the logging and management systems above it.

Both approaches are necessary and valid, but in my own case I generally am concerned with reasonably large sensor networks and collecting compatible data points that can easily be statistically analysed, etc., once a critical volumes is reached, and sensor availability is certain to change over time. The extra overhead of so much 'cross compatabiity' is very valuable to me, as is 'fast and efficient' access to raw data when I need to access something like accelerometer data at 1kHz, or something similar.

bfrog, 2019-02-25:
cross compatibility and high performance are also important to me, as is the ability to reconfigure the device at run time into various modes of operation for power savings and different triggers

avisconti, 2019-02-28:
This thread is really interesting to me. I am currently exploring the possibility to add FIFO capability to one of ST sensor (LSM6DSO), but there is a significant lack in the sensor APIs. I agree with all the requests listed in the @microbuilder description.

I am particularly interested into following stuff:

- timestamp management.
 Currently there is no way to pass timestamp to application, but this should be a task
 performed by driver.

- FIFO management
Absolutely mandatory.

- Virtual sensors handling
This request seems to match with 'features API' in the description.
We should be able to handle step_counter/step_detector/tilt/motion/orientation/pickup and so on virtual sensors

- Event management
Sometimes it is necessary to report events to application (e.g. step detection). 
Propably triggers is the way and we just need to populate the list of possible 'trigger events'.

- sensor calibration
I would need accel/gyro/mag calibration algo.

avisconti, 2019-02-28:
Concerning already existing code I think that linux IIO framework is good, but I also suggest taking
a look to Google contexthub, which is an Apache 2.0 licensed work:

https://android.googlesource.com/device/google/contexthub/

microbuilder, 2019-02-28:
@avisconti Thanks for the heads up on contexthub. It's also worth listing `libhardware` which contexthub relies on, and is a fairly mature API at this point considering the number of sensors it has had thrown at at:

- https://android.googlesource.com/platform/hardware/libhardware/+/master/modules/sensors/
- Channels, event, etc.: [sensors-base.h](https://android.googlesource.com/platform/hardware/libhardware/+/master/include/hardware/sensors-base.h)
- Core structs like `sensors_event_t ` [sensors.h](https://android.googlesource.com/platform/hardware/libhardware/+/master/include/hardware/sensors.h)

This was the basis of the Adafruit unified sensor API years back, but it has definitely evolved since I first looked at it ages ago.

avisconti, 2019-02-28:
@microbuilder 
Yes, I think that libhardware is the contexthub counterpart in Android which run on the AP, while contexthub itself is a firmware that runs on a companion micro.

But, yes, all that work is pretty standard now and runs on many Android based mobile phones.

avisconti, 2019-02-28:
Anyway, I personally think that this new sensor API should be scheduled with high priority for v1.15.

microbuilder, 2019-03-02:
An initial proof of concept repo has been created here -- along with some useful labels and issues -- to enable collaborative development until this is ready to advance to a proper PR and RFC in the `zephyr` repo: https://github.com/microbuilder/zsensor

I'm, of course, happy to add anyone as a collaborator, just dropping me here or in the zsensor repo so I know the right github username to add.

avisconti, 2019-03-04:
@microbuilder 
I think that, as a first step, we would need to reconsider the data structures associated to sensor.
Currenty everything is just a senor_value with val1 as integer part and val2 as fractional part.

Looking at google sensorhub I see that there are more complex data structures.
Take a look to https://android.googlesource.com/device/google/contexthub/+/refs/heads/master/firmware/os/inc/sensors.h

Data can be of folloqing types:
- NUM_AXIS_EMBEDDED
- NUM_AXIS_ONE 
- NUM_AXIS_THREE

Embedded is used for algo data, like step_counter data, tilt/sig_motion events, gesture events and so on. Axis_one for things like environmental (pressure, temperature, humidity ...). Axis_three for accel/gyro/mag/...

Each one of the 3 types has a different structure, with timestamp if it is the case.
Every sensor generates at a given time generates an array of this samples, which may be of size one or
more than that (in case for example of FIFO enabled).

microbuilder, 2019-03-04:
@avisconti Sorry that this kind of spans a couple PRs and issues at this point, but you can see some initial thoughts in code here, along with some other ideas by @bfrog earlier: https://github.com/zephyrproject-rtos/zephyr/pull/14008#issuecomment-469055223

I agree the current system is extremely limiting, though!

nordic-krch, 2019-12-19:
I see that sensor API discussions cooled down. I recently also encountered significant limitations there described in #21515

yperess, 2022-02-08:
I've nearly finished my first draft for a proposal on https://docs.google.com/document/d/1UsElJJHRudeWXb-tYwXA6qt9gF6PH8wwS7PjhqO1MLI/edit# @MaureenHelm @nashif @keith-zephyr 

I believe it's ready for review. There are a few pending TODOs to make sure we support the full Linux Kernel's iio features, but I believe I'm happy with how it's shaping up.

Note that this is a minimal design. When possible, it punts the grunt work to the application and/or a sensor framework. Many features described in this issue are really issues that shouldn't be handled in the individual driver, but instead in the framework that will follow after.

jgl-meta, 2023-04-26:
@MaureenHelm @microbuilder Can you provide a status update for this task? Will this be ready for release 3.4?

teburd, 2023-07-19:
I believe much of this will have been resolved with the inclusion of #60063 while more still will come from the sensing subsystem

microbuilder, 2023-07-19:
@teburd Agreed. This is already ancient, and I think any future work probably deserves a new RFC. Thanks for the work put into this so far, along with @yperess.

teburd, 2023-08-04:
I think this should remain open *until* #60063 is fully merged, as until then its still very much a valid thing.

yperess, 2023-08-29:
https://github.com/zephyrproject-rtos/zephyr/pull/61022 resolves the portions of this issue about multiple channels of the same type.

PR #60063 (2023-11-10): sensors: Add streaming APIs
Introduce a streaming API that uses the same data path as the async API.

This includes features to the decoder:
* Checking if triggers are present

Adding streaming features built ontop of existing triggers:
* Adding 3 operations to be done on a trigger
  * include - include the data with the trigger information
  * nop - do nothing
  * drop - drop the data (flush)
* Add a new sensor_stream() API to mirror sensor_read() but add an
optional handler to be able to cancel the stream.

Fixes #1387 
Fixes #13718
closed 2023-11-10 by MaureenHelm