zephyrproject-rtos/zephyr #13725: drivers: ssd1306: When 128x32 is used, only half of the screen is output.

KwonTae-young, 2019-02-25:
I have a `128x32` SSD1306 module.
SSD1306 Module(128x32): [Adafruit PiOLED - 128x32 Monochrome OLED Add-on for Raspberry Pi](https://www.adafruit.com/product/3527)

There is height in dts of SSD1306.
``` yaml
height:
  type: int
  category: required
  description: Height in pixel of the panel driven by the controller
  generation: define
```

So I put `32` in the height as shown below.
However, only half of the screen is printed.
```
ssd1306@3c {
        compatible = "solomon,ssd1306fb";
        reg = <0x3c>;
        label = "SSD1306";
        width = <128>;
        height = <32>;
        segment-offset = <0>;
        page-offset = <0>;
        display-offset = <0>;
        segment-remap;
        com-invdir;
        prechargep = <0x22>;
};
```

Below is the output of the screen according to the `height` value.
I tested it with `cfb_shell`.
```
cfb init
cfb print 0 0 1234567890
```

| height = <64> | height = <32> |
| --------------| ------------- |
| [image]  | [image] |

I tried to modify it as below.
The results are slightly different, but still half the output.
``` diff
diff --git a/drivers/display/ssd1306.c b/drivers/display/ssd1306.c
index 3c91e3d812..b354d6e4e3 100644
--- a/drivers/display/ssd1306.c
+++ b/drivers/display/ssd1306.c
@@ -32,7 +32,7 @@ LOG_MODULE_REGISTER(ssd1306);
 #define SSD1306_PANEL_NUMOF_PAGES      (DT_SOLOMON_SSD1306FB_0_HEIGHT / 8)
 #define SSD1306_CLOCK_DIV_RATIO                0x0
 #define SSD1306_CLOCK_FREQUENCY                0x8
-#define SSD1306_PANEL_MUX_RATIO                63
+#define SSD1306_PANEL_MUX_RATIO                (DT_SOLOMON_SSD1306FB_0_HEIGHT  - 1)
 #define SSD1306_PANEL_VCOM_DESEL_LEVEL 0x20
 #define SSD1306_PANEL_PUMP_VOLTAGE     SSD1306_SET_PUMP_VOLTAGE_90
``` 
[image]

Am I using dts incorrectly?
Or is it a driver bug?

jfischer-phytec-iot, 2019-09-11:
@KwonTae-young please check #19073

KwonTae-young, 2019-09-11:
@jfischer-phytec-iot 
I am currently in holiday season.
I'll check it next Monday. :)

KwonTae-young, 2019-09-16:
@jfischer-phytec-iot 
It works fine.
[image]

jfischer-phytec-iot, 2019-09-16:
@KwonTae-young :+1: maybe you would like to try `samples/gui/lvgl`, lvgl is more sophisticated as cfg

KwonTae-young, 2019-09-16:
@jfischer-phytec-iot I checked using lvgl. Works fine. :smile:
[image]

PR #19073 (2019-09-17): shields: add generic shield for SSD1306 based 128x32 display
Add DT properties to support 128x32 display.
Fixes: #13725

Add generic shield for SSD1306 based 128x32 pixel
monochrome displays.
commit 2388e316 (2019-09-17): drivers: ssd1306: add DT properties to support 128x32 display

Add DT properties to support 128x32 display.

Fixes: #13725
closed 2019-09-17 by MaureenHelm