zephyrproject-rtos/zephyr #13727: MIMXRT1020_EVK subsys.logging.logger report unaligned memory access

hakehuang, 2019-02-25:
**Describe the bug**
logger report unaligned memory access

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1020_evk ..
3. make
4. make flash

**Expected behavior**
case can run well

**Impact**
logger subsystem can not work well

**Screenshots or console output**
```
start application at 0x600053d5 !!!

***** Booting Zephyr OS 1.14.0-rc1 *****

Module logging showcase.

[00:00:00.130,586] [0m<inf> sample_module: log in test_module 11[0m

[00:00:00.130,630] [0m<inf> sample_module: Inline function.[0m

Disabling logging in the sample_module module

Function called again but with logging disabled.

Instance level logging showcase.

[00:00:00.132,265] [0m<inf> sample_instance.inst1: Inline call.[0m

[00:00:00.132,309] [0m<inf> sample_instance.inst1: counter_value: 0[0m

[00:00:00.132,371] [1;33m<wrn> sample_instance.inst1: Example of hexdump:

                                                01 02 03 04             |....    [0m

[00:00:00.132,423] [0m<inf> sample_instance.inst2: Inline call.[0m

[00:00:00.132,467] [0m<inf> sample_instance.inst2: counter_value: 0[0m

[00:00:00.132,528] [1;33m<wrn> sample_instance.inst2: Example of hexdump:

                                                01 02 03 04             |....    [0m

Changing filter to warning on sample_instance.inst1 instance.

[00:00:00.133,402] [1;33m<wrn> sample_instance.inst1: Example of hexdump:

                                                01 02 03 04             |....    [0m

[00:00:00.133,464] [0m<inf> sample_instance.inst2: Inline call.[0m

[00:00:00.133,509] [0m<inf> sample_instance.inst2: counter_value: 1[0m

[00:00:00.133,568] [1;33m<wrn> sample_instance.inst2: Example of hexdump:

                                                01 02 03 04             |....    [0m

Disabling logging on both instances.

Function call on both instances with logging disabled.

Severity levels showcase.

[00:00:01.146,297] [1;31m<err> main: Error message example.[0m

[00:00:01.146,337] [1;33m<wrn> main: Warning message example.[0m

[00:00:01.146,378] [0m<inf> main: Info message example.[0m

***** USAGE FAULT *****

  Unaligned memory access

***** Hardware exception *****

Current thread ID = 0x80000004
Faulting instruction address = 0x60002956

Fatal fault in thread 0x80000004! Aborting.
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain  Zephyr SDK
 - Commit SHA: v1.14.0_rc1

hakehuang, 2019-02-25:
@MaureenHelm

MaureenHelm, 2019-02-26:
I'm not able to reproduce this

galak, 2019-02-26:
Wondering if there are toolchain version differences?

hakehuang, 2019-02-27:
I find this issue may be related that the stack point is not set by zephyr application which depends on the board ROM to set,  @MaureenHelm , @galak does zephyr set the stack point, defined in the first 4 bytes of the address

hakehuang, 2019-02-27:
MPU region config issue before the application runs. not the application issue

closed 2019-02-27 by hakehuang

hakehuang, 2019-03-07:
unaligned memory access is caused by MPU not configure the flexspi-nor area as normal, in device mode, unaligned access is not allowed.