zephyrproject-rtos/zephyr #13729: sanitycheck --coverage failed

cinlyooi-intel, 2019-02-25:
**Describe the bug**
sanitycheck cannot generate coverage report.

**To Reproduce**
Steps to reproduce the behavior:
`scripts/sanitycheck --coverage --coverage-platform mps2_an385 -p mps2_an385  --coverage-platform qemu_x86 -p qemu_x86 --gcov-tool /opt/zephyr-sdk-0.9.5/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elf/i586-zephyr-elf-gcov -T tests/kernel/ -T tests/subsys/logging/
`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Cannot generate coverage report for Triaging.

**Screenshots or console output**
```
...

total complete:  147/ 147  100%  failed:  147
Generating coverage files...
geninfo: WARNING: no .gcda files found in /home/cinlyooi/Documents/vanilla/zephyr.git/sanity-out - skipping!
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/Documents/vanilla/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/Documents/vanilla/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/Documents/vanilla/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/Documents/vanilla/zephyr.git/sanity-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/cinlyooi/Documents/vanilla/zephyr.git/sanity-out/coverage.info
0 of 147 tests passed with 0 warnings in 105 seconds
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain :zephyr
 - Commit SHA or Version used: 23ba884b2bb5af8801a4874804800cc93ece7930

**Additional context**

Without coverage options, sanity check runs fine:
`scripts/sanitycheck  -p mps2_an385  -p qemu_x86  -T tests/kernel/ -T tests/subsys/logging/
`

AdithyaBaglody, 2019-02-25:
Issue is because of ccad9d0d099f38db97db95abd65e03a43a7f8ee6 
incorrect movement of the files into a new place.
@nashif advice on the solution.

cinlyooi-intel, 2019-03-01:
@nashif Any advice please? Many thanks

nashif, 2019-03-04:
this is still broken even after #13982

nashif, 2019-03-04:
at least on ARM

closed 2019-03-04 by nashif

Jocelyn-Li, 2019-03-04:
@wentongwu please continue checking about this.

cinlyooi-intel, 2019-03-05:
> at least on ARM

Hi @nashif : Does mps2_an385 counts as ARM? I do realize at present it is a QEMU emulation but I managed to get a html report using mps2_an385. 

```
scripts/sanitycheck \
   --coverage --coverage-platform mps2_an385 -p mps2_an385 \
   --gcov-tool /path/to/zephyr-elf-gcov \
   -T tests/subsys/logging
```

It also works for me if I substitutes `mps2_an385` with `qemu_x86`.

wentongwu, 2019-03-05:
> this is still broken even after #13982

the problem is that sanitycheck cannot generate coverage report. From Cinly's console output, all test cases failed because of cmake not finding coverage folder to generate build files. So after the #13982,  sanitycheck can work like before.  Also before the pull request, I do the test locally and it can fix Cinly's problem.

PR #13982 (2019-03-04): tests: move coverage code into subsys
move tests/coverage to subsys/testsuite to make cmake runs well when doing coverage enabled sanitycheck
Fixes #13729
commit 0ef2408e (2019-03-04): tests: move tests/coverage to subsys/testsuite

This part of tests was forgotten when we move to subsys/

Fixes #13729