zephyrproject-rtos/zephyr #13734: tests/subsys/settings/fcb/src/settings_test_save_unaligned.c fail with assertion failure on nrf52_pca10040

arun1joshi, 2019-02-25:
The tests in tests/subsys/settings/fcb/src/settings_test_save_unaligned.c fail with assertion failure on nrf52_pca10040

The test started failing from this commit: [b1a7e71]

**Arch**: arm
**Platforms**: nrf52_pca10040

**Error on console log**:
_nrf52_pca10040:arm_
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-506-g406e942949 (delayed boot 1000ms) *****
===================================================================
starting test - test_config_save_fcb_unaligned

Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/subsys/settings/fcb/src/settings_test_save_unaligned.c:40: test_config_save_fcb_unaligned: (val8_un == 33 is false)
bad value read
FAIL - test_config_save_fcb_unaligned
===================================================================
Test suite test_config_fcb failed.
PROJECT EXECUTION FAILED

nvlsianpu, 2019-02-27:
@arun1joshi At sha b1a7e71 this test as not implemented yet.

nvlsianpu, 2019-02-27:
Pleas check your repository tree status - it must have been not well (clean) checkout-ed.
I closed this as this bug can not appear for SHA you pointed, and it is not present for master HEAD as well.

closed 2019-02-27 by nvlsianpu

arun1joshi, 2019-03-01:
@nvlsianpu, can you please double check and re-open the issue?
still failing for the latest commit [9ec9c430b3].

nvlsianpu, 2019-03-01:
> ***** Booting Zephyr OS v1.14.0-rc1-669-g9ec9c430b312 *****
Running test suite test_config_fcb
===================================================================
starting test - config_empty_lookups
PASS - config_empty_lookups
===================================================================
starting test - test_config_insert
PASS - test_config_insert
===================================================================
starting test - test_config_getset_unknown
PASS - test_config_getset_unknown
===================================================================
starting test - test_config_getset_int
PASS - test_config_getset_int
===================================================================
starting test - test_config_getset_int64
PASS - test_config_getset_int64
===================================================================
starting test - test_config_commit
PASS - test_config_commit
===================================================================
starting test - test_config_save_fcb_unaligned
PASS - test_config_save_fcb_unaligned
===================================================================
starting test - test_config_empty_fcb
PASS - test_config_empty_fcb
===================================================================
starting test - test_config_save_1_fcb
PASS - test_config_save_1_fcb
===================================================================
starting test - test_config_insert2
PASS - test_config_insert2
===================================================================
starting test - test_config_save_2_fcb
PASS - test_config_save_2_fcb
===================================================================
starting test - test_config_insert3
PASS - test_config_insert3
===================================================================
starting test - test_config_save_3_fcb
PASS - test_config_save_3_fcb
===================================================================
starting test - test_config_compress_reset
PASS - test_config_compress_reset
===================================================================
starting test - test_config_save_one_fcb
PASS - test_config_save_one_fcb
===================================================================
starting test - test_config_compress_deleted
PASS - test_config_compress_deleted
===================================================================
Test suite test_config_fcb succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

> ***** Booting Zephyr OS v1.14.0-rc1-669-g9ec9c430b312 *****
Running test suite test_config_fcb
===================================================================
starting test - test_settings_encode
PASS - test_settings_encode
===================================================================
starting test - test_setting_raw_read
PASS - test_setting_raw_read
===================================================================
starting test - test_setting_val_read
PASS - test_setting_val_read
===================================================================
starting test - config_empty_lookups
PASS - config_empty_lookups
===================================================================
starting test - test_config_insert
PASS - test_config_insert
===================================================================
starting test - test_config_getset_unknown
PASS - test_config_getset_unknown
===================================================================
starting test - test_config_getset_int
PASS - test_config_getset_int
===================================================================
starting test - test_config_getset_int64
PASS - test_config_getset_int64
===================================================================
starting test - test_config_commit
PASS - test_config_commit
===================================================================
starting test - test_config_save_fcb_unaligned
PASS - test_config_save_fcb_unaligned
===================================================================
starting test - test_config_empty_fcb
PASS - test_config_empty_fcb
===================================================================
starting test - test_config_save_1_fcb
PASS - test_config_save_1_fcb
===================================================================
starting test - test_config_insert2
PASS - test_config_insert2
===================================================================
starting test - test_config_save_2_fcb
PASS - test_config_save_2_fcb
===================================================================
starting test - test_config_insert3
PASS - test_config_insert3
===================================================================
starting test - test_config_save_3_fcb
PASS - test_config_save_3_fcb
===================================================================
starting test - test_config_compress_reset
PASS - test_config_compress_reset
===================================================================
starting test - test_config_save_one_fcb
PASS - test_config_save_one_fcb
===================================================================
starting test - test_config_compress_deleted
PASS - test_config_compress_deleted
===================================================================
Test suite test_config_fcb succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

This are Results I have (RAW, BASE64 build).