zephyrproject-rtos/zephyr #13737: Where can I find tutorial to make my own device driver for a device under I2C bus?

hgijeon, 2019-02-25:
I'm trying to make my own device driver. the board has ADC chip connected to I2C pins.
Is there any tutorial to make device driver from scratch in zephyr?
I'm trying to find code example from devicetree through Kconfig and application usage, but I couldn't find yet.

chris-makaio, 2020-04-23:
I know it's hard to start with drivers. I suggest to have a look at /drivers/adc/adc_nrfx.c to see an actual implementation of the ADC API.
And for the I2C stuff, I recommend to look at a sensor driver for a starting point, most of them use I2C. 

To enable configuration of your driver in devicetree check /dts/bindings/ 
The yamls contain the definition of what can be configured via DT.

pabigot, 2020-08-19:
Closing as this is old, and the question has been answered (there isn't one).  The best available advice remains look at examples for similar drivers and follow what they do.

closed 2020-08-19 by pabigot