zephyrproject-rtos/zephyr #13748: CONFIG_SOC_MCIMX7D_M4 is undefined but referenced

ulfalizer, 2019-02-25:
Referenced in `ext/hal/nxp/imx/devices/MCIMX6X/device_imx.h`.

stanislav-poboril, 2019-02-25:
@ulfalizer ,

How does the bug manifest itself? It asks if the symbol is defined, which isn't, therefore it just skips some code.

The file is in `ext` folder, coming from FreeRTOS BSP for i.MX processors.
The files in `ext` folder are usually not modified, but in this case `CPU_MCIMX6X_M4` has been changed to `CONFIG_SOC_MCIMX6X_M4` and similarly for CPU_MCIMX7D_M4.

The original file:

    /*
     * Include the cpu specific register header files.
     *
     * The CPU macro should be declared in the project or makefile.
     */
    #if defined(CPU_MCIMX6X_M4)
    
        /* CMSIS-style register definitions */
        #include "MCIMX6X/include/MCIMX6X_M4.h"
        #define RDC_SEMAPHORE_MASTER_SELF   (5)
        #define SEMA4_PROCESSOR_SELF        (1)
    
    #elif defined(CPU_MCIMX7D_M4)
    
        /* CMSIS-style register definitions */
        #include "MCIMX7D/include/MCIMX7D_M4.h"
    
        #define RDC_SEMAPHORE_MASTER_SELF   (6)
        #define SEMA4_PROCESSOR_SELF        (1)
    
    #else
        #error "No valid CPU defined!"
    #endif

Perhaps the `#if` directives could be removed and only the relevant branch could stay there.
But I wanted to make only minimal modifications to the external code.

ulfalizer, 2019-02-25:
@stanislav-poboril 
IMO, dead code is a bug in itself, because people normally assume that all code is used (and might need to be updated).

Could leave it in if you think it'd be more work than it's worth to maintain a "fork" though.

stanislav-poboril, 2019-02-26:
I understand. The legacy FreeRTOS BSP and this file is probably not going to be updated, or updated frequently.

Bigger issue I think is that this file (and possibly others) is duplicated in `ext` for i.MX6 and i.MX7. If it would not duplicated but shared, there would not be a dead code.  I will have a look if some of those files in `ext` could be shared between i.MX6/7 instead of removing the dead code first.

stanislav-poboril, 2019-02-27:
Fixed in pr #13818

PR #13818 (2019-02-27): ext: hal: nxp: imx: Fixed duplicit file device_imx.h
The file was in two subfolders as copied from the FreeRTOS BSP
for i.MX6 and i.MX7. Unified the file and moved it into the
devices folder, like in the FreeRTOS BSPs.

Fixes: #13748
commit ee1bdc12 (2019-02-27): ext: hal: nxp: imx: Fixed duplicit file device_imx.h

The file was in two subfolders as copied from the FreeRTOS BSP
for i.MX6 and i.MX7. Unified the file and moved it into the
devices folder, like in the FreeRTOS BSPs.

Fixes: #13748
closed 2019-02-27 by galak