zephyrproject-rtos/zephyr #13753: *_UART_NS16550_PORT_{2,3}_* seems to be a (possibly broken) Kconfig/DTS mismash

ulfalizer, 2019-02-25:
These symbols are referenced but never defined:

```
CONFIG_UART_NS16550_PORT_2_BAUD_RATE
CONFIG_UART_NS16550_PORT_2_IRQ_FLAGS
CONFIG_UART_NS16550_PORT_2_IRQ_PRI
CONFIG_UART_NS16550_PORT_2_NAME
CONFIG_UART_NS16550_PORT_3_BASE_ADDR
CONFIG_UART_NS16550_PORT_3_BAUD_RATE
CONFIG_UART_NS16550_PORT_3_CLK_FREQ
CONFIG_UART_NS16550_PORT_3_IRQ
CONFIG_UART_NS16550_PORT_3_IRQ_FLAGS
CONFIG_UART_NS16550_PORT_3_IRQ_PRI
CONFIG_UART_NS16550_PORT_3_NAME
```

Some of the `CONFIG_*` macros seem to get defined with a `DT_*` prefix instead:

```
$ git grep UART_NS16550_PORT_3_BASE_ADDR
drivers/serial/uart_ns16550.c:  .port = CONFIG_UART_NS16550_PORT_3_BASE_ADDR,
soc/arc/snps_arc_iot/dts_fixup.h:#define DT_UART_NS16550_PORT_3_BASE_ADDR DT_NS16550_80014300_BASE_ADDRESS
soc/x86/apollo_lake/soc.c:MMU_BOOT_REGION(CONFIG_UART_NS16550_PORT_3_BASE_ADDR, 0x1000,
```

dcpleung, 2019-02-25:
@vonhust Could you take a look at soc/arc/snps_arc_iot/dts_fixup.h to see if those are defined correctly?

ulfalizer, 2019-02-25:
Would be nice to get rid of those `CONFIG_*` prefixes on stuff that isn't from Kconfig in any case.

dcpleung, 2019-02-25:
Previous renames from `CONFIG_*` to `DT_*` missed a few. Looking into the remaining ones now...

PR #13760 (2019-02-27): uart/ns16550: Use DT_ prefix for remaining device configs
Previous rename from CONFIG_* to DT_* left a few remaining
CONFIG_*. So rename them manually now.

Fix #13753
commit 603f0686 (2019-02-27): uart/ns16550: Use DT_ prefix for remaining device configs

Previous rename from CONFIG_* to DT_* left a few remaining
CONFIG_*. So rename them manually now.

Fix #13753
closed 2019-02-27 by galak