zephyrproject-rtos/zephyr #13754: Error: "West version found in path does not support '/usr/bin/make flash', ensure west is installed and not only the bootstrapper"

Olivier-ProGlove, 2019-02-25:
I cannot flash the simple hello world on nRF52840 using the latest Zephyr (tested commit `d30c9aeafd`, Fri Feb 22).

To duplicate the issue:

1. Open a new terminal:

        $ cd /tmp
        $ env | grep ZEPHYR
        ZEPHYR_SDK_INSTALL_DIR=/home/olivier/zephyr-sdk
        ZEPHYR_TOOLCHAIN_VARIANT=zephyr
        $ pip3 install --upgrade --user west
        $ pip list | grep west
        west (0.5.1)

2. Clone the repository

        $ git clone git@github.com:zephyrproject-rtos/zephyr.git
        $ export ZEPHYR_BASE=/tmp/zephyr/

3. Build it

        $ mkdir build && cd build
        $ cmake -DBOARD=nrf52840_pca10056 ../zephyr/samples/hello_world/

4. And flash it

        $ make flash
        Scanning dependencies of target flash
        West version found in path does not support '/usr/bin/make flash', ensure west is installed and not     only the bootstrapper. run 'west init' to fetch west.
        Built target flash

MaureenHelm, 2019-02-28:
Did you run `west init`?

mbolivar, 2019-03-01:
Closing as a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/12812

Hi @Olivier-ProGlove, flashing has been implemented using west for many months. The copy of west that was inside the zephyr repository was deleted when standalone west was included in the main workflow. You need a west installation to use `ninja flash` or `make flash`.

closed 2019-03-01 by mbolivar