zephyrproject-rtos/zephyr #13759: Implement system call fuzz testing

andrewboie, 2019-02-25:
We'd like to have fuzz testing implemented for system calls to give us more confidence in the correctness of our system call handler checks.

We could develop our own framework, or use something which exists and adapt it to Zephyr.

Reading material:
https://github.com/google/syzkaller
https://lwn.net/Articles/677764/
https://en.wikipedia.org/wiki/Fuzzing
https://codemonkey.org.uk/projects/trinity/

ceolin, 2019-02-25:
I've taken a look on this, it's worth to add afl-unicorn https://hackernoon.com/afl-unicorn-fuzzing-arbitrary-binary-code-563ca28936bf

afl-unicorn can be used to fuzz virtually everything, not only syscalls and it seems the easiest one to use on Zephyr.

d3zd3z, 2019-12-02:
@ceolin does this have as a prerequisite porting Zephyr to Unicorn? This does seem like it might be a good path forward for fuzzing on Zephyr, though. We'd talked about something with Qemu, but if these people have already put in effort with Unicorn, it might get us a lot closer.

mped-oticon, 2019-12-05:
Kinda related: FWIW, we have made an internal "CPU load fuzzer", a dummy load to determine system stabillity if/when we loose service (Bluetooth connection for us) or even crash.

{mean, variance}-randomized {period, duty cycle} over {coop, irq} loads.
Not meant for security but good as a canary in the coalmine for merges.

Let me know if you would be interested, I could throw up a RFC PR.

jettr, 2020-10-14:
I would like to see this happen as well. Is the 2.4 target realistic for adding fuzz support?

andrewboie, 2020-10-14:
@jettr I don't know if anyone is currently working on this.
I myself am occupied with delivering demand paging and some other MMU-related work for 2.5.
@ceolin is this still in your scope?