zephyrproject-rtos/zephyr #13765: mimxrt1060_evk tests/kernel/workq/work_queue meets Illegal use of the EPSR

hakehuang, 2019-02-26:
**Describe the bug**
Illegal use of the EPSR

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make
4. See error

**Expected behavior**
test pass

**Impact**
work queue usage

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

Running test suite workqueue

===================================================================

starting test - H=扂?4?#?齪d飈<?簑&Dw?夠8D悓FGyq香跲髹拍Q ?汢o縧s嚬凘?4玤烼a??槼?乨+弳?H 0?滇8??,KBO,

l傖趑
降E馼氅桡=??西{頃畷..?wzY^咣??;厔寸繒g簙?`詪錈?r褵蟼??)J?Y唗=o劕種

***** USAGE FAULT *****

  Illegal use of the EPSR

***** Hardware exception *****

Current thread ID = 0x800001a8

Faulting instruction address = 0x8000031c

Fatal fault in essential thread! Spinning...

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (Zephyr SDK)
 - Commit SHA or Version used:v1.14.0_rc1

hakehuang, 2019-02-26:
@MaureenHelm same with #13735

hakehuang, 2019-02-26:
mix with kernel/fatal failure.

MaureenHelm, 2019-02-26:
I'm not able to reproduce this. @dleach02, @agansari can you please give it a try?

hakehuang, 2019-02-27:
@MaureenHelm , @dleach02 , @agansari sorry this is the same root cause as #13727, zephyr does not reinitialize the sp. please ignore this

closed 2019-02-27 by hakehuang