zephyrproject-rtos/zephyr #13766: mimxrt1060_evk tests/kernel/fatal meet many unwanted exceptions

hakehuang, 2019-02-26:
**Describe the bug**
the exceptions are not expected

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make
4. See error

**Expected behavior**
all exception is correctly catched

**Impact**
exception handling

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

Running test suite fatal

===================================================================

starting test - test_fatal

test alt thread 1: generic CPU exception

***** USAGE FAULT *****

  Illegal use of the EPSR

***** Hardware exception *****

Current thread ID = 0x80000000

Faulting instruction address = 0x80000a7c

Caught system error -- reason 0

test alt thread 2: initiate kernel oops

***** Kernel OOPS! *****

Current thread ID = 0x80000000

Faulting instruction address = 0x6000261c

Caught system error -- reason 4

test alt thread 3: initiate kernel panic

***** Kernel Panic! *****

Current thread ID = 0x80000000

Faulting instruction address = 0x60002634

Caught system error -- reason 5

test stack overflow - timer irq

posting 2048 bytes of junk to stack...

busy waiting...

ULT *****

  Data Access Violation

  MMFAR Address: 0x60003b4c

***** Hardware exception *****

Current thread ID = 0x60003b25

Faulting instruction address = 0x60003f2a

Caught system error -- reason 0

ASSERTION FAIL [!(thread->base.user_options & ((1UL << (0))))] @ /build/src/workspace/rt1060_master-M47H37VYJLYH6RPUQ3ZIDFUP2YVNEZ4DCSCRKPANJY7327MRFXVA/arch/arm/core/thread_abort.c:37

	essential thread aborted

ULT *****

ULT *****

ULT *****

  Data Access Violation

  MMFAR Address: 0x60003b4c

***** Hardware exception *****

Current thread ID = 0x60003b25

Faulting instruction address = 0x60002e92

Caught system error -- reason 0

ASSERTION FAIL [!(thread->base.user_options & ((1UL << (0))))] @ /build/src/workspace/rt1060_master-M47H37VYJLYH6RPUQ3ZIDFUP2YVNEZ4DCSCRKPANJY7327MRFXVA/arch/arm/core/thread_abort.c:37

	essential thread aborted***** Booting Zephyr OS 1.14.0-rc1 *****

Running test suite fatal
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain ( Zephyr SDK)
 - Commit SHA or Version used: v1.14.0_rc1

hakehuang, 2019-02-26:
[serial_2019-02-26-12-18-22 (1).txt](https://github.com/zephyrproject-rtos/zephyr/files/2903782/serial_2019-02-26-12-18-22.1.txt)

hakehuang, 2019-02-26:
@MaureenHelm

MaureenHelm, 2019-02-26:
I think this is expected behavior. @andrewboie is that correct?

hakehuang, 2019-02-27:
according to the log, the board is keeping rebooting. @MaureenHelm

MaureenHelm, 2019-04-03:
@hakehuang Looking at this again, I see you're right that the reset is not expected behavior. The problem seems to have been resolved between rc1 and rc3, but I'm having trouble bisecting due to #14497. Can you rerun this test on your side?

closed 2019-04-10 by MaureenHelm

hakehuang, 2019-04-22:
@MaureenHelm fixed in d18ec535755210a738f56db24933f5f6bdacb88e