zephyrproject-rtos/zephyr #13779: mimxrt1060_evk tests/kernel/mem_pool/mem_pool_threadsafe meets Imprecise data bus error

hakehuang, 2019-02-26:
**Describe the bug**
Imprecise data bus error

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make
4. See error

**Expected behavior**
Test pass

**Impact**
memory pool

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

Running test suite mpool_threadsafe

===================================================================

starting test - test_mpool_threadsafe

***** BUS FAULT *****

  Imprecise data bus error

***** Hardware exception *****

Current thread ID = 0x800000d0

Faulting instruction address = 0x600056ae

Fatal fault in ISR! Spinning..
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK)
 - Commit SHA or Version used: v1.14.0_rc1

**Additional context**
Add any other context about the problem here.

hakehuang, 2019-02-26:
@MaureenHelm

andyross, 2019-02-27:
This sounds a lot like a duplicate of #12559, which has a fix up in #13770.  Could you try with that applied and verify it works for you?

hakehuang, 2019-02-27:
@andyross thanks it fixed

closed 2019-02-27 by hakehuang