zephyrproject-rtos/zephyr #13780: mimxrt1060_evk tests/crypto/tinycrypt_hmac_prng meets Unaligned memory access

hakehuang, 2019-02-26:
**Describe the bug**
Unaligned memory access

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make
4. See error

**Expected behavior**
test PASS

**Impact**
tinycrypt

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

Running test suite test_hmac_prng_fn

===================================================================

starting test - test_hmac_prng

***** USAGE FAULT *****

  Unaligned memory access

***** Hardware exception *****

Current thread ID = 0x80000050

Faulting instruction address = 0x6000265c

Fatal fault in thread 0x80000050! Aborting.
```

**Environment (please complete the following information):**
 - OS: ( Linux)
 - Toolchain (Zephyr SDK)
 - Commit SHA or Version used: v1.14.0_rc1

hakehuang, 2019-02-26:
@MaureenHelm

MaureenHelm, 2019-02-26:
I'm not able to reproduce either of these failures.

hakehuang, 2019-02-27:
@MaureenHelm , this is the same root cause as #13727, SP is not set by zephyr itself.

closed 2019-02-27 by hakehuang