zephyrproject-rtos/zephyr #13813: Test suite mslab_threadsafe fails randomly

ulfalizer, 2019-02-27:
Been seeing this in a few different unrelated PRs now:

```
Running test suite mslab_threadsafe
===================================================================
starting test - test_mslab_threadsafe
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x0
***** Hardware exception *****
Current thread ID = 0x200000b0
Faulting instruction address = 0x49ea
Fatal fault in ISR! Spinning...
---out-3nd-pass/mps2_an385/tests/kernel/mem_slab/mslab_threadsafe/kernel.memory_slabs/handler.log---
0 of 1 tests passed with 0 warnings in 83 seconds
```

ulfalizer, 2019-03-05:
Haven't seen it in a while now. Not sure what changed.

andyross, 2019-03-06:
There was a "nonatomic swap" fix that went in about a week ago that IIRC mslab would have tickled.  This is one of the tests that **does** fail sporadically in CI, but it's definitely not failing reliably for me right now on unloaded systems, and I'm regularly running such tests to check now as part of #12553.  Will close, please reopen if you see it.

andrewboie, 2020-07-15:
I am able to reproduce this issue.
https://github.com/andrewboie/zephyr/tree/mslab-threadsafe-issue
reproduces with the `qemu_x86_coverage` target.
This is on x86 but the error is we get a pagefault when we try to execute a NULL address:
```
Running test suite mslab_threadsafe
===================================================================
START - test_mslab_threadsafe
 PASS - test_mslab_threadsafe
===================================================================
Test suite mslab_threadsafe succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
E: Page fault at address 0x0 (error code 0x11)
E: Access violation: supervisor thread not allowed to execute
E:   PDE: 0x0000000000136827 Writable, User, Execute Enabled
E:   PTE: 0x8000000000000013 Writable, Supervisor, Execute Disable
E: EAX: 0x00154680, EBX: 0x00000000, ECX: 0x00000246, EDX: 0x00156060
E: ESI: 0x00000000, EDI: 0x00104766, EBP: 0x001489e8, ESP: 0x001489e4
E: EFLAGS: 0x00000206 CS: 0x0008 CR3: 0x0013a220
E: call trace:
E: EIP: 0x00000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x0013a128 (unknown)
E: Halting system
```

I can confirm a few things:
- This situation is **very** timing sensitive.
- My branch reproduces 100% of the time on my system. It also failed in CI when I sent out #. We are using icount, so my guess is that other code changes in my PR which might change timings are causing this to get unlucky in whatever race is going on, and since it's icount execution is deterministic; it either always works or it doesn't.
- I swear my PR is orthogonal to this, other than I/O mapping happens at different times.
- Doesn't reproduce in any other build target
- Extremely minor changes to the `tmslab_api()` function cause the crash to go away. like adding a printk() at the end, or removing the `if (block[i])` check and freeing the block unconditionally, or changing the timeout arg to k_mem_slab_alloc() from TIMEOUT to K_FOREVER. It's that subtle.
- The crash is always one of the tdata threads. The stack is almost empty save for a memory address inside tmslab_api.
- The crash always happens after 
- There's no stack overflows and we're not re-using thread objects

I thought I was going insane until I found this GH ticket.
This branch is the same as the other branch I linked, plus a cleanup patch to the mslab_threadsafe test https://github.com/andrewboie/zephyr/tree/mslab-threadsafe-fixup there are comments that show where you can disturb the timings.

andrewboie, 2020-07-15:
I think I can rule out any changes in my PR, nothing should be accessing any memory in the NULL page. This does seem like this obscure bug popping up again.

I don't have time right now to dive deeper, but leaving this open. I submitted a patch which cleans up the test code a little and adds a delay which makes the problem stop failing the test. https://github.com/zephyrproject-rtos/zephyr/pull/26883

andrewboie, 2020-07-15:
Spent some more time looking at this. The patch that added a delay that fixed it for qemu_x86_coverage causes the test to now fail on qemu_x86_nommu with a general protection fault or invalid opcode exception. In this case, instead of EIP being 0, EIP points to some location on the interrupt stack, and the faulting thread is main() after it has exited.

The test does something unusual: it sets `CONFIG_TIMESLICE_SIZE=1`. There is only one other test that does this: tests/kernel/mem_pool/mem_pool_threadsafe. Guess what? This test **also** has been known to fail intermittently for reasons nobody has been able to figure out, see https://github.com/zephyrproject-rtos/zephyr/issues/19701

I've spent hours staring at the test code. There's literally nothing wrong with it, it's only 100 lines of code. My suspicion right now that there is a subtle race condition within the kernel that tends to only show itself (sometimes) if the timeslicing rate is high. Basically, bolstering my suspicion that this might be due to a badly timed timer interrupt corrupting something. I think the HPET driver is not the culprit since historically we have seen this on other arches.

So in sum:
- `CONFIG_TIMESLICE_SIZE=1` I think is a strong clue
- I think the test code itself is not related to the problem
- Seems to be stack corruption that causes a bad branch to happen as a thread is exiting, but I haven't pinpointed where in code it happens, I just know its somewhere along the thread exit path

andrewboie, 2020-07-15:
Just for laughs I cloned a fresh tree and applied this patch to the master branch, I wanted to see if stuff starts to fail if we enable 1ms timeslicing by default:
```
apboie@apboie-mobl4:~/projects/zephyr2/zephyr master /home/apboie/projects/zephyr2/zephyr
$ git diff
diff --git a/kernel/Kconfig b/kernel/Kconfig
index 702d908d4b..fc94ca3111 100644
--- a/kernel/Kconfig
+++ b/kernel/Kconfig
@@ -404,7 +404,7 @@ config TIMESLICING
 
 config TIMESLICE_SIZE
        int "Time slice size (in ms)"
-       default 0
+       default 1
        range 0 2147483647
        depends on TIMESLICING
        help
```

I then ran `sanitycheck -T tesks/kernel` to see what shakes out.

I then got a reproducible failure in tests/kernel/sched/deadline/kernel.scheduler.deadline on qemu_riscv32 where we are also jumping to an illegal location:
```
*** Booting Zephyr OS build zephyr-v2.3.0-1056-g568211738d30  ***
Running test suite suite_deadline
===================================================================
START - test_deadline
E: Exception cause Illegal instruction (2)
E: Faulting instruction address = 0x204060ac
E:   ra: 0x204061a8  gp: 0x00000000  tp: 0x00000000  t0: 0x00000000
E:   t1: 0x00000000  t2: 0x00000000  t3: 0x00000000  t4: 0x00000000
E:   t5: 0x00000000  t6: 0x00000000  a0: 0x000186a0  a1: 0x00000000
E:   a2: 0x204066f8  a3: 0x00000021  a4: 0x00000000  a5: 0x00000008
E:   a6: 0x00000000  a7: 0x00000000

E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x800004d8 (unknown)
E: Halting system
```

With this change I noticed a few of tests failing on qemu_x86_64 due to a page fault at address 0xB9, which means an un-suspended thread object was attempted to be resumed. These seem sporadic, but qemu_x86_64 does not use icount. AFAICT the tests that failed are not racing to re-use thread objects so I suspect this might be related?

Looking through test configurations, we do not have any dedicated tests for timeslicing. Only three tests change the default value of CONFIG_TIMESLICE_SIZE=0, which means unless I'm missing something, only these tests are exercising the timeslice feature, and all of them are actually testing something else:

- tests/kernel/mem_slab/mslab_threadsafe/prj.conf sets it to 1
- tests/kernel/mem_pool/mem_pool_threadsafe/prj.conf sets it to 1
- tests/kernel/mem_protect/syscalls/prj.conf sets it to 10 (for the syscall torture test)

andrewboie, 2020-07-15:
I went another step further, and in addition to the patch to apply timeslicing to all tests, added this to disable icount for all emulators:
```
diff --git a/cmake/emu/qemu.cmake b/cmake/emu/qemu.cmake
index 26e7622dee..a05ca93cd3 100644
--- a/cmake/emu/qemu.cmake
+++ b/cmake/emu/qemu.cmake
@@ -60,12 +60,6 @@ endif()
 # Connect monitor to the console chardev.
 list(APPEND QEMU_FLAGS -mon chardev=con,mode=readline)
 
-if(CONFIG_QEMU_ICOUNT)
-  list(APPEND QEMU_FLAGS
-         -icount shift=${CONFIG_QEMU_ICOUNT_SHIFT},align=off,sleep=off
-         -rtc clock=vm)
-endif()
-
 # Add a BT serial device when building for bluetooth, unless the
 # application explicitly opts out with NO_QEMU_SERIAL_BT_SERVER.
 if(CONFIG_BT)
```

I then ran sanitycheck again to see what fails, and why. There were 43 total failures when testing `sanitycheck -T tests/kernel` which seems abnormal even with icount disabled. I looked through the logs:

- There were a number of expected timing ordering failures, maybe 5-10
- Several tests failed with no error in the output, as if the output was truncated or the emulator exited suddenly
- I also saw lots of crashes. The majority of the failures were with qemu_cortex_m0 or qemu_cortex_r5, here are some examples:

qemu_cortex_r5            tests/kernel/mbox/mbox_api/kernel.mailbox.api
```
START - test_mbox_async_put_to_waiting_get
E: ***** HARD FAULT *****
E: r0/a1:  0xfffffffd  r1/a2:  0xe000ed00  r2/a3:  0xffffffff
E: r3/a4:  0xffffffff r12/ip:  0x00000000 r14/lr:  0x00003f87
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x00005dec
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200005a0 (unknown)
E: Halting system
```

qemu_cortex_m0            tests/kernel/sched/schedule_api/kernel.scheduler.multiq
```
START - test_lock_preemptible
E: ***** HARD FAULT *****
E: r0/a1:  0xfffffffd  r1/a2:  0xe000ed00  r2/a3:  0xffffffff
E: r3/a4:  0xffffffff r12/ip:  0x00000000 r14/lr:  0x00003ee3
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x00005a84
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000758 (unknown)
E: Halting system
```

qemu_cortex_m0            tests/kernel/sched/schedule_api/kernel.scheduler
```
START - test_lock_preemptible
E: ***** HARD FAULT *****
E: r0/a1:  0xfffffffd  r1/a2:  0xe000ed00  r2/a3:  0xffffffff
E: r3/a4:  0xffffffff r12/ip:  0x00000000 r14/lr:  0x00003ee3
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x00005aa4
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000758 (unknown)
E: Halting system
```

qemu_cortex_r5            tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem.nouser
```
START - test_sem_take_multiple
ASSERTION FAIL [thread->base.pended_on] @ WEST_TOPDIR/zephyr/kernel/sched.c:475
@ WEST_TOPDIR/zephyr/lib/os/assert.c:45
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x000060b0 (unknown)
E: Halting system
```

qemu_cortex_r5            tests/kernel/mem_pool/mem_pool_threadsafe/kernel.memory_pool.threadsafe
```
START - test_mpool_threadsafe
E: ***** PREFETCH ABORT *****
E: Permission Fault @ 0xffb41092
E: r0/a1:  0x00004c58  r1/a2:  0x00004ec0  r2/a3:  0x00004ec0
E: r3/a4:  0x00000000 r12/ip:  0x00000002 r14/lr:  0x00003098
E:  xpsr:  0x600001ff
E: Faulting instruction address (r15/pc): 0xffb41092
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x00004ec0 (main)
E: Halting system
```

I think at this point I've convinced myself that this is a race condition in the timeslicing implementation.

andrewboie, 2020-07-21:
caught this failing a timeout assertion on x86_64:

```
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.3.0-1198-g849ff583968e  ***
Running test suite mpool_threadsafe
===================================================================
START - test_mpool_threadsafe
ASSERTION FAIL [!sys_dnode_is_linked(&to->node)] @ WEST_TOPDIR/zephyr/kernel/timeout.c:103
	
E: RAX: 0x0000000000000004 RBX: 0x0000000000109390 RCX: 0x0000000000000001 RDX: 0x0000000000000046
E: RSI: 0x0000000000000067 RDI: 0x000000000010847b RBP: 0x0000000000000015 RSP: 0x0000000000111088
E: Page fault at address 0xb9 (error code 0x11)
E:  R8: 0x0000000000000001  R9: 0x0000000000000020 R10: 0x0000000000000202 R11: 0x0000000000111098
E: Access violation: supervisor thread not allowed to execute
E: R12: 0xfffffffffffffffe R13: 0x00000000001034c3 R14: 0x0000000000000016 R15: 0x0000000000000010
E: PTE: 0x0000000000000000 -> 0x0000000000000000: RW XD 
E: RAX: 0x0000000000000001 RBX: 0x0000000000111050 RCX: 0x0000000000000001 RDX: 0x0000000000000046
E: RSI: 0x0000000000000000 RDI: 0x000000000010bc64 RBP: 0x00000000001110d0 RSP: 0x0000000000111048
E:  R8: 0x000000000010bb30  R9: 0x0000000000000001 R10: 0x0000000000000202 R11: 0x0000000000109da0
E: RSP: 0x0000000000111088 RFLAGS: 0x0000000000000006 CS: 0x0018 CR3: 0x0000000000141000
E: R12: 0x0000000000000000 R13: 0x0000000000000001 R14: 0x000000000010bb10 R15: 0x000000000013c1c0
E: RSP: 0x0000000000111048 RFLAGS: 0x0000000000000247 CS: 0x0018 CR3: 0x0000000000141000
E: RIP: 0x00000000000000b9
E: RIP: 0x0000000000100dde
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 1
E: Attempt to resume un-suspended thread object
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x0000000000109360 (unknown)
E: Current thread: 0x0000000000109360 (unknown)
E: Halting system

```

andrewboie, 2020-08-12:
opened another ticket now that the root cause is found

closed 2020-08-12 by andrewboie