zephyrproject-rtos/zephyr #13821: tests/kernel/sched/schedule_api: Assertion failed for test_slice_scheduling

cinlyooi-intel, 2019-02-27:
**Describe the bug**
'''
 Assertion failed at src/test_slice_scheduling.c:51: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && ((int)p1 == thread_idx)) is false
'''

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf51_pca10028 -DARCH=arm 
3. make
4. make run

**Screenshots or console output**
```
===================================================================
starting test - test_slice_scheduling
AB
A
Assertion failed at zephyr.git/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:51: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && ((int)p1 == thread_idx)) is false

B
Assertion failed at zephyr.git/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:51: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && ((int)p1 == thread_idx)) is false

 Assertion failed at zephyr.git/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:51: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && ((int)p1 == thread_idx)) is false

FAIL - test_slice_scheduling
======================================================
```
**Environment (please complete the following information):**
 - OS: Fedora 29
 - Toolchain : zephyr
 - Commit SHA or Version used: dd5ff3957ceece428e3465dd4e23fb926f61691c

gnuless, 2019-03-06:
@cinlyooi-intel Are you running this test on real hardware?

andyross, 2019-03-06:
And is this reliable?  I was looking at exactly this test on a nrf52840_pca10056 ~2-3 weeks ago and it was running great for me.  If it's failing, even spuriously, it's likely to be a recent regression.

andyross, 2019-03-06:
Read more carefully, this is a duplicate of #11721, which got fixed by #13916 last week but is not included in the SHA referenced in the report above.

closed 2019-03-06 by andyross