zephyrproject-rtos/zephyr #13822: Invalid USB state: powered after cable is disconnected

pdunaj, 2019-02-27:
**Describe the bug**
Replicated with nrf_desktop on both PCA20041 and PCA10056 (here you have to enable USB support as it is not there in default configuration).
Connect to USB - observe `powered` event. HID reports are transmitted and everything seems fine.
Disconnect USB now - observe `disconnected` event (so far so good) and then `powered` event which clearly is wrong as USB cable is not connected. This behavior is there since Jan and it seems latest change done to USB is not fixing it.

**To Reproduce**
See description.

**Expected behavior**
When cable is disconnected driver should not inform that it is connected.

**Impact**
Application cannot work around this so we assume cable is connected. This is showstopper as device behavior is improper.

**Screenshots or console output**
```
[01074764] <inf> event_manager: e: usb_state_event id:0x20000195 state:POWERED
[01074764] <inf> hid_state: Subscriber 0x20000195 connected
[01074764] <inf> hid_state: Active subscriber 0x20000195
[01074795] <wrn> usb_state: USB_DC_CONNECTED when USB is not disconnected
[01084472] <inf> usb_state: USB HID enabled
[01084475] <inf> event_manager: e: usb_state_event id:0x20000195 state:ACTIVE
[01084478] <inf> event_manager: e: hid_report_subscription_event MOUSE report notification enabled by 0x20000195
[01085358] <wrn> usb_hid: len 126 doesn't match Report Descriptor size
[01414924] <inf> usb_state: USB HID disabled
[01414926] <wrn> usb_state: USB_DC_RESET when USB is disconnected
[01414930] <inf> event_manager: e: hid_report_subscription_event MOUSE report notification disabled by 0x20000195
[01414930] <inf> hid_state: Clear report data (0)
[01414932] <inf> event_manager: e: usb_state_event id:0x20000195 state:DISCONNECTED
[01414932] <inf> hid_state: Subscriber 0x20000195 disconnected
[01414932] <inf> hid_state: Active subscriber 0x20003560
[01414934] <inf> event_manager: e: usb_state_event id:0x20000195 state:POWERED
[01414934] <inf> hid_state: Subscriber 0x20000195 connected
```

**Environment (please complete the following information):**
ncs: bd591571f66c17a8693755465b1067320eeff19d
ncs zephyr: a869285d262d57c382eddce928b46e3b18daa197 (from upstream ec424b7a455cc98c36e3cab80be7e152ef2c5217)

Newest change done in https://github.com/zephyrproject-rtos/zephyr/pull/13132 does not help.

**Additional context**
N/A

finikorg, 2019-03-08:
@pdunaj I assume this is bug against nrf platform, right?

carlescufi, 2019-03-08:
Yes but this should be fixed now. Checking

PR #14473 (2019-03-13): drivers: usb: nordic: Fix for events when cable is detached
When user disconnects the USB cable, peripheral should be
immediately disabled, howewer a delay may occur when driver
events are processed from a workqueue or higher-priority
thread/ISR is active. This may lead to a fake resume/reset
event (peripheral-specific behavior). This fix drops such
events when cable is detached.

Fixes #13822
commit 79dc1fee (2019-03-13): drivers: usb: nordic: Fix for events when cable is detached

When user disconnects the USB cable, peripheral should be
immediately disabled, howewer a delay may occur when driver
events are processed from a workqueue or higher-priority
thread/ISR is active. This may lead to a fake resume/reset
event (peripheral-specific behavior). This fix drops such
events when cable is detached.

Fixes #13822
closed 2019-03-13 by carlescufi