zephyrproject-rtos/zephyr #13823: tests/kernel/arm_irq_vector_table: test case cannot quit displaying "isr 0 ran!"

cinlyooi-intel, 2019-02-27:
**Describe the bug**
Test case did not quit. It kept displaying "isr 0 ran!" until it times out

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52_pca10040 -DARCH=arm 
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
...
isr0 ran!
isr0 ran!
isr0 ran!
isr0 ran!
isr0
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain : zephyr
 - Commit SHA or Version used: dd5ff3957ceece428e3465dd4e23fb926f61691c

ioannisg, 2019-02-27:
Thanks for reporting this, @cinlyooi-intel 
The problem is that the IRQ line-0 is wired to the CLOCK_CONTROL IRQ in nRF52x series of SOCs, and that peripheral is generating interrupts.

ioannisg, 2019-02-27:
Provided a fix in #13903, @cinlyooi-intel pls, test if you have time.

PR #13903 (2019-02-28): Tests kernel arm irq vector table fix
This PR does some minor refactoring of the arm_irq_vector_table test,
and provides the fixes so the test can successfully run in nRF platforms.

Fixes #13823 .
commit 1e740076 (2019-02-28): tests: kernel: arm_irq_vector_table: add clock ISR in the IRQ vector

This commit adds the Clock Control Interrupt Service
Routine into the customized vector table, when building
for nRF52X-based platforms. As a result, the interrupts
generated by the clock control will not interfere with
the test.

Fixes #13823.
closed 2019-02-28 by nashif