zephyrproject-rtos/zephyr #13842: [Coverity CID :190978]Control flow issues in /drivers/spi/spi_intel.c

mandarcthorat1, 2019-02-27:
Static code scan issues seen in File: /drivers/spi/spi_intel.c
 Category: Control flow issues
 Function: push_data
 Component: Drivers
 CID: 190978
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

pfalcon, 2019-02-28:
Also handled by @maxxlife in https://github.com/zephyrproject-rtos/zephyr/pull/13806, which was earlier. (@maxxlife, that shows that if you start working on a ticket, it makes sense to add a comment to it about that).

maxxlife, 2019-02-28:
> 
> 
> Also handled by @maxxlife in #13806, which was earlier. (@maxxlife, that shows that if you start working on a ticket, it makes sense to add a comment to it about that).

Ok. I just didn't see that thread, maybe it appeared later.

PR #13899 (2019-02-28): spi: spi_intel: Fix missing 'break' in case
Coverity scan found issue with a missing 'break' statement.  Fix
push_data by adding the break after handling the 1 byte case.

Coverity CID: 190978
Fixes #13842
commit 846dfd4b (2019-02-28): spi: spi_intel: Fix missing 'break' in case

Coverity scan found issue with a missing 'break' statement.  Fix
push_data by adding the break after handling the 1 byte case.

Coverity CID: 190978
Fixes #13842
closed 2019-02-28 by andrewboie