zephyrproject-rtos/zephyr #13855: [Coverity CID :190961]Error handling issues in /subsys/bluetooth/host/mesh/prov.c

mandarcthorat1, 2019-02-27:
Static code scan issues seen in File: /subsys/bluetooth/host/mesh/prov.c
 Category: Error handling issues
 Function: prov_data
 Component: Bluetooth
 CID: 190961
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

jhedberg, 2019-02-28:
This is marked as fixed by #12338 in Coverity. Why was this issue opened?

mandarcthorat1, 2019-03-13:
Commit was not fixing the issue reported by coverity. The problem is still valid and detected by latest scan

jhedberg, 2019-03-13:
@mandarcthorat1 seems like I may have gotten two different issues mixed up. Anyway, this is a simple fix - will submit a PR in a moment.

PR #14451 (2019-03-13): Bluetooth: Mesh: Fix checking for bt_mesh_provision() error
There isn't much we can do if this fails, but at least log an error
about it.

Fixes #13855
commit 0299db73 (2019-03-13): Bluetooth: Mesh: Fix checking for bt_mesh_provision() error

There isn't much we can do if this fails, but at least log an error
about it.

Fixes #13855
closed 2019-03-13 by galak