zephyrproject-rtos/zephyr #13873: [Coverity CID :190940]Memory - corruptions in /soc/arm/microchip_mec/mec1701/soc.c

mandarcthorat1, 2019-02-27:
Static code scan issues seen in File: /soc/arm/microchip_mec/mec1701/soc.c
 Category: Memory - corruptions
 Function: soc_init
 Component: Other
 CID: 190940
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

galak, 2019-02-27:
Issue related to treating a pointer as an array:

```
        for (girc_enable_set = (uint32_t *)&INTS_INST->GIRQ08_EN_SET;
             girc_enable_set <= &INTS_INST->GIRQ15_EN_SET;
             girc_enable_set += 5) {
			*girc_enable_set = 0xFFFFFFFF;
		}
```

If we want we could just remove the loop and do:
```
		INTS_INST->GIRQ08_EN_SET = 0xFFFFFFFF;
		INTS_INST->GIRQ09_EN_SET = 0xFFFFFFFF;
		INTS_INST->GIRQ10_EN_SET = 0xFFFFFFFF;
		INTS_INST->GIRQ11_EN_SET = 0xFFFFFFFF;
        ...
		INTS_INST->GIRQ15_EN_SET = 0xFFFFFFFF;
```

But marking this as something we aren't going to change and isn't an issue.

closed 2019-02-27 by galak

ceolin, 2019-03-01:
there are other places that coverity is complaining the same thing. I vote classify it as intentional and close the issue.