zephyrproject-rtos/zephyr #13890: stm32: serial: Data is not read properly at a certain baud rate

KwonTae-young, 2019-02-27:
**Describe the bug**
UART data is not read properly at a certain `baud rate`.

**To Reproduce**
Board: [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html)
Barcode module: [EM1399](http://www.newlandaidc.com/h-pd-84.html#keyword=EM1399&pfc=%7B%22lid%22%3A1%2C%22prgmid%22%3A399%2C%22kw%22%3A%22EM1399%22%2C%22sc%22%3A%7B%22key%22%3A%22name%22%2C%22desc%22%3Afalse%7D%7D&_pp=0_399)  
[image]

I was conducting a test with [EM1399](http://www.newlandaidc.com/h-pd-84.html#keyword=EM1399&pfc=%7B%22lid%22%3A1%2C%22prgmid%22%3A399%2C%22kw%22%3A%22EM1399%22%2C%22sc%22%3A%7B%22key%22%3A%22name%22%2C%22desc%22%3Afalse%7D%7D&_pp=0_399) to read the barcode.
[EM1399](http://www.newlandaidc.com/h-pd-84.html#keyword=EM1399&pfc=%7B%22lid%22%3A1%2C%22prgmid%22%3A399%2C%22kw%22%3A%22EM1399%22%2C%22sc%22%3A%7B%22key%22%3A%22name%22%2C%22desc%22%3Afalse%7D%7D&_pp=0_399) is connected to `UART1`.

```
west init zephyrproject -m https://github.com/KwonTae-young/zephyr/ --mr code
cd zephyrproject/zephyr
source zephyr-env.sh
mkdir samples/barcode/build
cd samples/barcode/build/
cmake -DBOARD=stm32f4_disco ..
make -j8 flash
```

**Expected behavior**
I expected the scanned barcode information to be output.

**Impact**
At certain baud rate, barcode data is not read properly through UART.

**Screenshots or console output**
I have tested on a total of two boards.([stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html), [nrf52840_pca10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html))
[stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html) does not read the data properly.
[nrf52840_pca10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html) reads the data correctly.

1. UART setup of [EM1399](http://www.newlandaidc.com/h-pd-84.html#keyword=EM1399&pfc=%7B%22lid%22%3A1%2C%22prgmid%22%3A399%2C%22kw%22%3A%22EM1399%22%2C%22sc%22%3A%7B%22key%22%3A%22name%22%2C%22desc%22%3Afalse%7D%7D&_pp=0_399)

|                   | [EM1399](http://www.newlandaidc.com/h-pd-84.html#keyword=EM1399&pfc=%7B%22lid%22%3A1%2C%22prgmid%22%3A399%2C%22kw%22%3A%22EM1399%22%2C%22sc%22%3A%7B%22key%22%3A%22name%22%2C%22desc%22%3Afalse%7D%7D&_pp=0_399) |
| ------------ | ------ |
| baud rate    | 9600   |
| stop bit     | 1      |
| data bit     | 8      |
| parity       | none   |
| flow control | none   |

2. [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html)
[image]
- pin connection

| [EM1399](http://www.newlandaidc.com/h-pd-84.html#keyword=EM1399&pfc=%7B%22lid%22%3A1%2C%22prgmid%22%3A399%2C%22kw%22%3A%22EM1399%22%2C%22sc%22%3A%7B%22key%22%3A%22name%22%2C%22desc%22%3Afalse%7D%7D&_pp=0_399)       | [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html) |
| ------------ | ------------- |
| VDD(2)       | 3V            |
| GND(3)       | GND           |
| TX(5)        | PB7(UART1_RX)           |
| nTrig(12)    | custom button |

- console
```
***** Booting Zephyr OS zephyr-v1.13.0-5106-g5d6c36b85f *****
Sample app running on: stm32f4_disco
uart_init() done
uart_config.baudrate=9600
uart_config.parity=0
uart_config.stop_bits=1
uart_config.data_bits=3
uart_config.flow_ctrl=0
uart_isr: 38 (1 bytes)
uart_isr: 38 (1 bytes)
uart_isr: 30 (1 bytes)
uart_isr: 30 (1 bytes)
uart_isr: 32 (1 bytes)
uart_isr: 37 (1 bytes)
uart_isr: 31 (1 bytes)
[... 317 lines cut ...]
uart_config.parity=0
uart_config.stop_bits=1
uart_config.data_bits=3
uart_config.flow_ctrl=0
uart_isr: 38 (1 bytes)
uart_isr: 38 (1 bytes)
uart_isr: 34 (1 bytes)
```

9. baud rate: 115200
```
***** Booting Zephyr OS zephyr-v1.13.0-5106-g34550e11e9 *****
Sample app running on: stm32f4_disco
uart_init() done
uart_config.baudrate=115200
uart_config.parity=0
uart_config.stop_bits=1
uart_config.data_bits=3
uart_config.flow_ctrl=0
uart_isr: 38 (1 bytes)
uart_isr: 38 (1 bytes)
```

I'm sorry I do not have enough English.
Thank you.

erwango, 2019-02-27:
Can you try the following, in `boards/arm/stm32f4_disco/pinmux.c`:
```
#ifdef CONFIG_UART_1
	{STM32_PIN_PB6, STM32F4_PINMUX_FUNC_PB6_USART1_TX},
-	{STM32_PIN_PB7, STM32F4_PINMUX_FUNC_PB7_USART1_RX},
+	{STM32_PIN_PB7, STM32F4_PINMUX_FUNC_PB7_USART1_RX | STM32_OSPEEDR_VERY_HIGH_SPEED},
#endif	/* CONFIG_UART_1 */
```

KwonTae-young, 2019-02-28:
Hi. @erwango
I applied and tested it.
But the result was the same.
```
***** Booting Zephyr OS zephyr-v1.13.0-5106-g34550e11e9 *****
Sample app running on: stm32f4_disco
uart_init() done
uart_config.baudrate=9600
uart_config.parity=0
uart_config.stop_bits=1
uart_config.data_bits=3
uart_config.flow_ctrl=0
uart_isr: 38 (1 bytes)
uart_isr: 38 (1 bytes)
uart_isr: 30 (1 bytes)
uart_isr: 30 (1 bytes)
uart_isr: 32 (1 bytes)
uart_isr: 37 (1 bytes)
uart_isr: 31 (1 bytes)
uart_isr: 36 (1 bytes)
uart_isr: 0a (1 bytes)
```

KwonTae-young, 2019-02-28:
## Additional testing
I have tested some more about [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html) and [nrf52840_pca10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html).
This test was done using [samples/nfc/nfc-hello](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/nfc/nfc_hello).
The baudrate was tested to `9600`.

**loopback test**
- [nrf52840_pca10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html)(loopback)
[image]
Connect P0.26(Tx) and P0.27(Rx) to each other.
Below is the log of [nrf52840_pca10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html).
```
***** Booting Zephyr OS zephyr-v1.13.0-5106-g34550e11e9 *****
Sample app running on: arm
uart1_init() done
uart1_isr: 00 (1 bytes)
uart1_isr: 00 (1 bytes)
uart1_isr: 00 (1 bytes)
uart1_isr: 04 (1 bytes)
uart1_isr: 20 (1 bytes)
uart1_isr: 00 (1 bytes)
uart1_isr: 01 (1 bytes)
uart1_isr: 00 (1 bytes)
```

- [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html)(loopback)
[image]
Connect PB6(Tx) and PB7(Rx) to each other.
Below is the log of [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html).
```
***** Booting Zephyr OS zephyr-v1.13.0-5106-g34550e11e9 *****
Sample app running on: arm
uart1_init() done
uart1_isr: 00 (1 bytes)
```

**stm32f4_disco+ nrf52840_pca10056 test**
- [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html)(Rx) + [nrf52840_pca10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html)(Tx)
[image]
Connect PB7(Rx) and P0.26(Tx) to each other.
Below is the log of [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html).
```
***** Booting Zephyr OS zephyr-v1.13.0-5106-g34550e11e9 *****
Sample app running on: arm
uart1_init() done
uart1_isr: 00 (1 bytes)
uart1_isr: 00 (1 bytes)
uart1_isr: 00 (1 bytes)
uart1_isr: 04 (1 bytes)
uart1_isr: 00 (1 bytes)
uart1_isr: 00 (1 bytes)
```

- [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html)(Tx) + [nrf52840_pca10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html)(Rx)
[image]
Connect PB6(Tx) and P0.27(Rx) to each other.
Below is the log of [nrf52840_pca10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html).
```
***** Booting Zephyr OS zephyr-v1.13.0-5106-g34550e11e9 *****
Sample app running on: arm
uart1_init() done
uart1_isr: 00 (1 bytes)
```

I was not using `Tx` because it was a barcode scanner.
But from the above results, `Tx` also seems strange.
I added `STM32_OSPEEDR_VERY_HIGH_SPEED` but the result was the same.

erwango, 2019-02-28:
@KwonTae-young , can you provide test sample you're using for loopback test so I can test on my side?

KwonTae-young, 2019-02-28:
@erwango, I used [samples/nfc/nfc-hello](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/nfc/nfc_hello).
Pressing the reset button will output the defined reset command.
https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/nfc/nfc_hello/src/main.c#L65-L69
``` c
/*
 * Peer will receive: 0x00 0x00 0x00 0x04 0x20 0x00 0x01 0x00
 *	                nci_reset size   +    nci_reset cmd
 */
uart_fifo_fill(uart1_dev, tx_buf, sizeof(u32_t) + sizeof(nci_reset));
```

I created a [loopback](https://github.com/KwonTae-young/zephyr/tree/loopback) branch that I tested.
loopback branch: https://github.com/KwonTae-young/zephyr/tree/loopback
```
west init zephyrproject_loopback -m https://github.com/KwonTae-young/zephyr/ --mr loopback
cd zephyrproject_loopback/zephyr
source zephyr-env.sh
mkdir samples/nfc/nfc_hello/build
cd samples/nfc/nfc_hello/build/
cmake -DBOARD=stm32f4_disco ..
make -j8 flash
```

Below is a test video.
1. stm32f4_disco
https://youtu.be/J9NJ-enYM5I

2. nrf52840_pca10056
https://youtu.be/9FMTosBAdEE

erwango, 2019-02-28:
There is at least an issue in fifo_fill.
I've proposed a fix: https://github.com/zephyrproject-rtos/zephyr/pull/13952, so the loopback mode is now working.

Then, about your use case, can you have a try to disable printf in uart1_isr and see what happens?

erwango, 2019-02-28:
@KwonTae-young , actually, it appears that uart_fifo_fill call in nfc_hello does not respect the API and behavior is undefined, cf: 
https://github.com/zephyrproject-rtos/zephyr/blob/687e02de8451856ceb8bfb3c7737b12db0a63407/include/uart.h#L697-L698

pfalcon, 2019-02-28:
uart_fifo_read, uart_fifo_fill, etc. are internal APIs, deprecated for use by application. That's because it's very hard to use them correctly by an application.

Please use higher-level API, I suggest https://github.com/zephyrproject-rtos/zephyr/blob/master/include/tty.h

KwonTae-young, 2019-02-28:
@erwango, @pfalcon
I am on vacation until March 3rd.
I will test the following on Monday next week.
1. Apply #13952 to read looback and barcode data.
2. Use 'https://github.com/zephyrproject-rtos/zephyr/blob/master/include/tty.h' to read the barcode data.

I thought [samples/nfc/nfc_hello](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/nfc/nfc_hello) was a simple example using uart.
I currently can not find an example using https://github.com/zephyrproject-rtos/zephyr/blob/master/include/tty.h.
So I will refer to https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/console/getchar.c.

KwonTae-young, 2019-03-04:
@erwango
I applied #13952.
And I modified [samples/nfc/nfc-hello](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/nfc/nfc_hello) as follows.
``` diff
diff --git a/samples/nfc/nfc_hello/src/main.c b/samples/nfc/nfc_hello/src/main.c
index 0f2f6fa358..926c9bb45c 100644
--- a/samples/nfc/nfc_hello/src/main.c
+++ b/samples/nfc/nfc_hello/src/main.c
@@ -19,6 +19,9 @@ static u8_t rx_buf[BUF_MAXSIZE];
 static u8_t tx_buf[BUF_MAXSIZE];
 static u8_t nci_reset[] = {0x20, 0x00, 0x01, 0x00};
 
+static u8_t save_buf[BUF_MAXSIZE];
+static int count = 0;
+
 static void msg_dump(const char *s, u8_t *data, unsigned len)
 {
        unsigned i;
@@ -35,7 +38,17 @@ static void uart1_isr(struct device *x)
        int len = uart_fifo_read(uart1_dev, rx_buf, BUF_MAXSIZE);
 
        ARG_UNUSED(x);
-       msg_dump(__func__, rx_buf, len);
+
+       // length is always 1
+       memcpy(save_buf + count++, rx_buf, 1);
+
+       // The reset command has a length of 8
+       if (count == 8) {
+               printf("[loopback test] save_buf: ");
+               for (int i = 0; i < count ; i++) {
+                       printf("%02x ", save_buf[i]);
+               }
+       }
 }
 
 static void uart1_init(void)
```

Then the loopback test was good.
Works well on all baudrates.
```
***** Booting Zephyr OS zephyr-v1.13.0-5107-geded5c2720 *****
Sample app running on: arm
uart1_init() done
[loopback test] save_buf: 00 00 00 04 20 00 01 00
```

I tried to run a bardcode scanner in a similar way.
[image]
Barcode data is read normally.
```
***** Booting Zephyr OS zephyr-v1.13.0-5107-geded5c2720 *****
Sample app running on: arm
uart1_init() done
[loopback test] save_buf: 38 38 30 39 30 39 32 38 37 34 31 34 36 0d 0a
```

<br/>

@pfalcon
I created the following program using the [tty API](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/tty.h).
``` c
#include <zephyr.h>
#include <stdio.h>
#include <uart.h>
#include <device.h>
#include <console.h>
#include <tty.h>

static struct device *uart_dev;
static struct tty_serial console_serial;

void main(void)
{
	printf("Sample app running on: %s\n", CONFIG_BOARD);

	uart_dev = device_get_binding("UART_1");
	tty_init(&console_serial, uart_dev);
}
```

However, an error occurs during compilation.
What is my mistake?
I used [tty.h](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/tty.h).
I have uploaded the tested source to the [tty_api_test](https://github.com/KwonTae-young/zephyr/tree/tty_api_test) branch.
``` 
[  1%] Built target driver_validation_h_target
[  4%] Built target syscall_list_h_target
[  5%] Built target syscall_macros_h_target
[  6%] Built target kobj_types_h_target
[  7%] Built target offsets
[  8%] Built target offsets_h
[  9%] Built target linker_script_target
[ 19%] Built target lib__libc__minimal
[ 20%] Built target app
[ 22%] Built target arch__arm__core__cortex_m__mpu
[ 29%] Built target arch__arm__core__cortex_m
[ 42%] Built target arch__arm__core
[ 44%] Built target drivers__gpio
[ 63%] Built target kernel
[ 64%] Built target drivers__pwm
[ 67%] Built target boards__arm__stm32f4_disco
[ 91%] Built target zephyr
[ 93%] Built target drivers__serial
[ 94%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       16764 B         1 MB      1.60%
            SRAM:        4460 B       192 KB      2.27%
        IDT_LIST:         152 B         2 KB      7.42%
../app/libapp.a(main.c.obj): In function `main':
/work/zephyrproject_loopback/zephyr/samples/tty_api/src/main.c:22: undefined reference to `tty_init'
collect2: error: ld returned 1 exit status
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:97: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
CMakeFiles/Makefile2:458: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

gon1332, 2019-03-04:
@KwonTae-young  You should add these in your proj.conf:
```
CONFIG_CONSOLE_SUBSYS=y
CONFIG_CONSOLE_GETCHAR=y
```

KwonTae-young, 2019-03-04:
@gon1332
Thank you. After adding the setting, the error disappears.  :smile:

KwonTae-young, 2019-03-05:
I created a program that reads the barcode using the [tty API](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/tty.h).
The source has been uploaded to the [tty_api_test](https://github.com/KwonTae-young/zephyr/tree/tty_api_test) branch.
``` c
/*
 * Copyright (c) 2015 Intel Corporation
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr.h>
#include <stdio.h>
#include <uart.h>
#include <device.h>
#include <console.h>
#include <tty.h>
#include <string.h>

#define BARCODE_UART_PORT	"UART_1"

static struct device *uart_dev;
static struct tty_serial console_serial;

static u8_t console_rxbuf[CONFIG_CONSOLE_GETCHAR_BUFSIZE];
static u8_t barcode_buf[64];
static u8_t length = 0;

int barcode_init()
{
	struct uart_config uart_cfg;
	int ret;

	uart_dev = device_get_binding(BARCODE_UART_PORT);
	if (uart_dev == NULL) {
		printf("Failed to get %s\n", BARCODE_UART_PORT);
		return -1;
	}

	ret = uart_config_get(uart_dev, &uart_cfg);
	if (!ret) {
		printf("\n======== [%s] ========\n", BARCODE_UART_PORT);
		printf("[%s] uart_config.baudrate=%d\n", BARCODE_UART_PORT, uart_cfg.baudrate);
		printf("[%s] uart_config.parity=%d\n", BARCODE_UART_PORT, uart_cfg.parity);
		printf("[%s] uart_config.stop_bits=%d\n", BARCODE_UART_PORT, uart_cfg.stop_bits);
		printf("[%s] uart_config.data_bits=%d\n", BARCODE_UART_PORT, uart_cfg.data_bits);
		printf("[%s] uart_config.flow_ctrl=%d\n", BARCODE_UART_PORT, uart_cfg.flow_ctrl);
	} else {
		printf("uart_config_get() error\n");
		return -1;
	}

	tty_init(&console_serial, uart_dev);
	tty_set_rx_buf(&console_serial, console_rxbuf, sizeof(console_rxbuf));

	return 0;
}

int get_barcode()
{
	int size = 0;

	size = tty_read(&console_serial, console_rxbuf, 1);
	if (size > 0) {
		barcode_buf[length++] = console_rxbuf[0];
		if (console_rxbuf[0] == 0x0a) {
			return 0;
		}
	}

	return -1;
}

void main(void)
{
	int i, ret;

	printf("tty API sample app running on(for barcode): %s\n", CONFIG_BOARD);

	ret = barcode_init();
	if (ret) {
		printf("barcode_init() error\n");
		return;
	}

	while (1) {
		if (!get_barcode()) {
			printf("barcode data: ");
			for (i = 0; i < length; i++) {
				printf("%02x ", barcode_buf[i]);
			}
			printf("\n");

			length = 0;
			memset(console_rxbuf, 0x00, sizeof(console_rxbuf));
			memset(barcode_buf, 0x00, sizeof(barcode_buf));
		}
	}

	return;
}
```

I created two barcodes for testing.
It is read well using the [tty API](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/tty.h).
[image]
[image]

```
***** Booting Zephyr OS zephyr-v1.13.0-5108-gd6e60d5cb0 *****
tty API sample app running on(for barcode): stm32f4_disco

======== [UART_1] ========
[UART_1] uart_config.baudrate=9600
[UART_1] uart_config.parity=0
[UART_1] uart_config.stop_bits=1
[UART_1] uart_config.data_bits=3
[UART_1] uart_config.flow_ctrl=0
barcode data: 31 32 33 34 35 36 37 38 39 30 0d 0a
barcode data: 31 32 33 34 35 36 37 38 39 30 0d 0a
barcode data: 31 32 33 34 35 36 37 38 39 30 0d 0a
barcode data: 5a 65 70 68 79 72 0d 0a
barcode data: 5a 65 70 68 79 72 0d 0a
barcode data: 5a 65 70 68 79 72 0d 0a
```

pfalcon, 2019-03-05:
@KwonTae-young:

> It is read well using the tty API.

Glad to hear!

erwango, 2019-03-05:
Txs @pfalcon for help on this. @KwonTae-young, please close the issue when you think it is ok.

closed 2019-03-05 by KwonTae-young