zephyrproject-rtos/zephyr #13913: MPU policy interactions with SAM E70 flash driver

andrewboie, 2019-02-27:
Logging from a message from JK on slack:

 Hi, Im using SAME70 (Cortex-M7) platform and I have a problem with the FLASH driver when MPU is enabled.  The basic problem is that MPU is protecting FLASH before writing - this is what we actually want. How ever the FLASH controller does not count with this behavior and need to perform the write operation into the FLASH address area, which cause write into internal latch registers. So when the MPU is enabled and protecting the FLASH area, this operation cause MPU to rise exception - as expected. However this write can be anywhere where the FLASH memory is expected (but it dont have to be installed)  There is no simply workaround, i see three options:
1) Disable before write and then enable MPU   - with multi threading this is might not be a good idea. I have a limited knowledge here.
2) Shrink down the MPU area to make a free space which is not covered by MPU. But the size of MPU area is defined as 2^x - so shrinking down is pretty complex.
3) Create workaround which will not cover all FLASH variant  (I can write beyond the installed flash area, but for example with SAME70 this is working only for 515KB and 1MB flash and it will not work for 2MB flash as I can't write beyond the 2BM flash area)
We can create all patch variants, But Im not sure which one is preferred by Zephyr community.  Can some help me to make a good decision.

aurel32, 2019-02-27:
When the MPU is enabled, if you want to be able to write to flash, you need to enabled the `MPU_ALLOW_FLASH_WRITE` option. Is it enabled in your case?

ioannisg, 2019-02-28:
I think this is only enabled, by default in Bluetooth -supported builds. So I doubt that it is.

ioannisg, 2019-03-08:
Could  this issue be closed?

kubiaj, 2019-03-27:
Hi,
as I didnt file the question I didnt get the notification about your purpose. 

The catch here is that user normally dont want to have the ability to write to the flash (when NAND flash is used) - that is why we have MPU. We want to catch the program exception when something goes wrong - when it really try write into FLASH area because some pointer is wrong. 

In our SW we need sometime update backup firmware or update system environment variables.  NAND memories have controller to write/read/erase memory and for SAM device you are filling the shadow register via write to FLASH area but you are not really writing into the flash. When this shadow register is filled then you can issue a command to write the shadow register into the specific FLASH address. The problem here is that I must fill the shadow register via writing to the flash area. 

From my point of view setting MPU_ALLOW_FLASH_WRITE is a workaround and not true solution - it leaves FLASH open for writing but in real it is not possible.

My understanding of MPU_ALLOW_FLASH_WRITE is that can use this with NOR memories when I can do direct FLASH write without use special flash controller.

ioannisg, 2019-05-31:
Any solution we pick here, @aurel32 will eat at least one MPU region, AFAICS.

kubiaj, 2019-05-31:
Hi,
I dont know. However Im almost convinced that the only one good solution might be to temporarily enable MPU_MPU_FLASH_WRITE or temporarily disable MPU. I have discovered that the Nordic driver behave in the same way.

aurel32, 2019-05-31:
> Any solution we pick here, @aurel32 will eat at least one MPU region, AFAICS.

Agreed, we still need to keep one region for the program execution and then we need a new region to program the flash.

> I have discovered that the Nordic driver behave in the same way.

The behavior is not at the flash driver level, but at the MPU level. Therefore the behavior is the same for all SoCs.