zephyrproject-rtos/zephyr #13919: tests/crypto/mbedtls reports some errors without failing

andrewboie, 2019-02-27:
Observe output of tests/crypto/mbedls:

```
Running test suite test_mbedtls_fn
===================================================================
starting test - test_mbedtls
starting test - Performing mbedTLS crypto tests:
the snprintf implementation is broken

 MBEDTLS_SELF_TEST not defined.
  Executed 0 test suites

  [ All tests PASS ]

PASS - test_mbedtls
===================================================================
===================================================================
PROJECT EXECUTION SUCCESSFUL
PASS - test_mbedtls
===================================================================
Test suite test_mbedtls_fn succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

Need to understand these two error messages and why they are printed. I am very unsure on priority.

> the snprintf implementation is broken

>  MBEDTLS_SELF_TEST not defined.
>   Executed 0 test suites

andrewboie, 2019-02-28:
As far as I can tell, almost this entire test case is a no-op since 
MBEDTLS_SELF_TEST is not defined

andrewboie, 2019-03-03:
I saw the patch #14016 
Does this also fix the `the snprintf implementation is broken` message?

nashif, 2019-03-04:
> Does this also fix the `the snprintf implementation is broken` message?

ah, I missed this one.

commit 3b745cc6 (2019-03-05): mbedtls: enable self-test in config-tls-generic.h

MBEDTLS_SELF_TEST is needed when setting CONFIG_MBEDTLS_TEST

Fixes #13919
closed 2019-03-05 by andrewboie