zephyrproject-rtos/zephyr #13934: tests/kernel/fatal: test_fatal rv equal to TC_FAIL

cinlyooi-intel, 2019-02-28:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=quark_se_c1000_ss_devboard -DARCH=arc 
3. make
4. See run

**Screenshots or console output**
```

***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-60-g9072d34f30 (delayed boot 1000ms) *****
Running test suite fatal
===================================================================
starting test - test_fatal
test alt thread 1: generic CPU exception
Exception vector: 0x8, cause code: 0x0, parameter 0x0
Address 0x40000550
Current thread ID = 0xa8000400
Faulting instruction address = 0x40000550
Caught system error -- reason 0
test alt thread 2: initiate kernel oops
***** Kernel OOPS! *****
Current thread ID = 0xa8000400
Caught system error -- reason 4
test alt thread 3: initiate kernel panic
***** Kernel Panic! *****
Current thread ID = 0xa8000400
Caught system error -- reason 5
test stack overflow - timer irq
posting 1024 bytes of junk to stack...
busy waiting...
FAIL - stack_thread1@132. should never see this
***** Stack Check Fail! *****
Current thread ID = 0xa8000400
Caught system error -- reason 2

 Assertion failed at zephyr.git/tests/kernel/fatal/src/main.c:236: test_fatal: (rv equal to TC_FAIL)

thread was not aborted
FAIL - test_fatal
===================================================================
Test suite fatal failed.
===================================================================
RunID: ci-190212-1245-422:uuur
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: 9072d34f30

andrewboie, 2019-03-06:
If I understand this bug report properly, this is a test with CONFIG_STACK_SENTINEL enabled that is not working on ARC. The result ought to be reproducible with the nsimdrv emulator using the nsim_em target.

wentongwu, 2019-03-13:
I checked the latest code, and running fail code should not be compiled with ARC board configured. please retest this case with latest zephyr version.  And the version(9072d34) you use is dated on Feb 9, after that there is changes for this case.

Thanks

cinlyooi-intel, 2019-03-13:
Passed for 6368ed4edf5c5ffbccf9f904efe836ef5ca5d84a. Closing case.

closed 2019-03-13 by cinlyooi-intel