zephyrproject-rtos/zephyr #13937: tests/net/tcp: Page fault

cinlyooi-intel, 2019-02-28:
**Describe the bug**
Although the test case ran to completion and report that it is successful. There is a "Page Fault" error

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2.cmake -DBOARD=qemu_x86 -DARCH=x86
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-60-g9072d34 (delayed boot 1000ms) *****
starting test - test TCP init
passed
starting test - test TCP register/unregister port cb
passed
starting test - test TCP context init
passed
starting test - test IPv6 TCP reset packet creation
passed
starting test - test IPv4 TCP reset packet creation
passed
starting test - test IPv6 TCP syn packet creation
passed
starting test - test IPv4 TCP syn packet creation
passed
starting test - test IPv6 TCP synack packet create
passed
starting test - test IPv4 TCP synack packet create
passed
starting test - test IPv6 TCP fin packet creation
passed
starting test - test IPv4 TCP fin packet creation
passed
starting test - test IPv6 TCP seq check
passed
starting test - test IPv4 TCP seq check
passed
starting test - test TCP seq validity
passed
starting test - test TCP reply context init
passed
starting test - test TCP accept init
passed
starting test - test cleanup
passed
===================================================================
RunID: ci-190212-1245-422:fls2
PROJECT EXECUTION SUCCESSFUL
[00:00:14.123,179] &#1b;[1;31m<err> net_conn: Address families different&#1b;[0m
***** CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x00000030
PDE: 0x025 Present, Read-only, User, Execute Enabled
PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
Current thread ID = 0x004047fc
eax: 0x00000030, ebx: 0x0040a358, ecx: 0x00000030, edx: 0x0040a358
esi: 0x00000028, edi: 0x00401428, ebp: 0x0041cf70, esp: 0x0041cf6c
eflags: 0x00000206 cs: 0x0008
call trace:
eip: 0x0000db84
0x0000e570 (0x13c0f)
0x0000e6b8 (0x40a358)
0x00004486 (0x4038f8)
0x0000440e (0x4047e0)
Fatal fault in thread 0x004047fc! Aborting.
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: b75115a49ef20d6131752b79aeda94e9a476f380

jukkar, 2019-03-01:
I could not reproduce this. There was no crash at the end of the test run, there is error print that says
`[00:00:10.215,052] <err> net_conn: Address families different` but no crashes.
The head of my git tree was b17d9eefc5897c5bca71b50241c7cdcb272f7c10
I am using SDK 0.9.5 so the qemu version is most probably different which could explain the different result.

rveerama1, 2019-03-08:
I tried this today in my environment.
===================================================================
PROJECT EXECUTION SUCCESSFUL
[00:00:02.540,000] <err> net_conn: Address families different

OS: Ubuntu 18.04
Toolchain: Zephyr
SDK: zephyr-sdk-0.10.0-rc3
git head: https://github.com/rveerama1/zephyr/commit/2384ebc2e32572afea5fee99c850e4cf8943dee8

rveerama1, 2019-03-11:
> Also fyi, #14069 has a patch for TCP tests are migrated to new net_pkt api.

PR merged and I tried it with today latest https://github.com/rveerama1/zephyr/commit/7967829a261ef8ae21622710dd5699d30110af11.

Hence closing this issue, re-open if you still find.

closed 2019-03-11 by rveerama1