zephyrproject-rtos/zephyr #13955: stm32: Implement async uart api

erwango, 2019-02-28:
**Is your feature request related to a problem? Please describe.**
UART_ASYNC_API is not available yet on stm32  target.
Implement it

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

StefJar, 2019-03-26:
I done a demo implementation
which could be used as a [blueprint](https://github.com/StefJar/zephyr_stm32_uart3_dma_driver)

erwango, 2021-01-25:
Fixed: https://github.com/zephyrproject-rtos/zephyr/pull/30917

closed 2021-01-25 by erwango