zephyrproject-rtos/zephyr #13984: samples: console/echo: It doesn't echo

gon1332, 2019-03-01:
**Describe the bug**
The echo sample doesn't seem to work as expected.
The motivation to try this sample, was after I wrote a sample using tty.h and it could not receive data through `tty_read()`. My sample is a driver for a camera module. I've also mocked the camera behaviour via FTDI but nothing this function doesn't seem to response. I've also tried different UART ports.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo_l496zg -s samples/subsys/console/echo
2. west flash
3. See error

**Expected behavior**
It should echo back the input.

**Screenshots or console output**
In the output below, you can also see this artifact. It happens every time when the sample starts.
```
***** Booting Zephyr OS v1.14.0-rc1-669-g9ec9c430b312 *****
You should see another line with instructions below. If not,
the (interrupt-driven) console device doesn't work as expected:
Start typing characters to see them echoed back
�I speak to the void, but it doesn't echo back to me..
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit hash: 9ec9c430b

pfalcon, 2019-03-01:
If that's actual console-shot, then it actually shows characters echoed back ;-).

Otherwise, as the sample suggest, there's an issue with interrupt-driven UART functions implementation.

Can you try other BOARD's, starting with qemu_x86 ?

gon1332, 2019-03-04:
Damn, you're right. Sorry for that one.

closed 2019-03-04 by gon1332