zephyrproject-rtos/zephyr #13991: net: Spurious driver errors due to feeding packets into IP stack when it's not fully initialized (assumed reason)

pfalcon, 2019-03-01:
A typical situation:
~~~
***** Booting Zephyr OS zephyr-v1.13.0-5295-g5f2099fb63ba *****
[00:00:02.362,040] <err> eth_smsc911x: net_recv_data: -22
[00:00:02.364,062] <inf> net_config: Initializing network
[00:00:02.364,637] <inf> net_config: IPv4 address: 192.0.2.1
~~~

It's pretty clear (though not confirmed) that driver (on a typical busy network) receives and feeds a packet into stack when it's not yet fully initialized.

It's pretty easy to reproduce that with the default qemu_x86/SLIP setup, where I regularly get 2 screenfuls of errors like that.

We should do something about that. And the most obvious solution is to have proper initializing priorities, where drivers are initialized only after the core stack can accept packets from them.

pfalcon, 2019-03-07:
> It's pretty easy to reproduce that with the default qemu_x86/SLIP setup, where I regularly get 2 screenfuls of errors like that.

Here's a typical case:

~~~
SeaBIOS (version rel-1.11.2-0-gf9626ccb91-prebuilt.qemu-project.org)
Booting from ROM..***** Booting Zephyr OS zephyr-v1.13.0-5473-gae63b2411fe4 *****

--- 64 messages dropped ---
[00:00:00.350,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.360,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.360,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.360,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.370,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.370,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.370,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.370,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.380,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.380,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.390,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.390,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.400,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.400,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.410,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.410,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.420,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.420,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.420,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.430,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.430,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.440,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.440,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.440,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.450,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.460,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.500,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.500,000] <err> slip: [0x00400660] cannot allocate pkt
[00:00:00.540,000] <inf> net_config: Initializing network
[00:00:00.540,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.640,000] <inf> net_config: IPv6 address: fe80::200:5eff:fe00:5306
[00:00:00.650,000] <inf> net_config: IPv6 address: fe80::200:5eff:fe00:5306
[00:00:03.660,000] <inf> net_echo_server_sample: Run echo server
[00:00:03.660,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6)...
[00:00:03.660,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
[00:00:03.660,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:03.660,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv4)...
~~~

pfalcon, 2019-03-07:
@jukkar, Any comments?

jukkar, 2019-03-07:
If the upper network stack is not able to receive any packets, then driver drops the packet as it is doing right now. What are you proposing we start to do at this point? Simple option is to just ignore the issue and not print the error as there is nothing wrong at this point.

Anyway, making this an enhancement and not a bug.

jukkar, 2020-01-28:
Going through old issues. This is a good candidate for closing. Can be re-opened if really needed.

closed 2020-01-28 by jukkar