zephyrproject-rtos/zephyr #14010: logger: timestamp resets after 35.7 seconds on K64F

mike-scott, 2019-03-02:
**Describe the bug**
Timestamps will wrap from 35.7 seconds to 0 on the FRDM K64F board.

I haven't bisected the issue yet, but it seems to have been around for more than a week.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/subsys/shell/shell_module/
2. mkdir build && cd build
3. cmake -DBOARD=frdm_k64f ..
4. make
5. make flash
6. [connect to the console on K64F and enter] log_test start demo

**Expected behavior**
Expect the timestamps not to reset to 0 after 35.7 seconds.

**Impact**
Not sure what actual impact this has or where the bug lies.

**Screenshots or console output**
```
[00:00:35.616,064] <inf> app: Timer expired.
[00:00:35.616,067] <inf> app_test: info message
[00:00:35.616,069] <wrn> app_test: warning message
[00:00:35.616,072] <err> app_test: err message
[00:00:00.024,673] <inf> app: Timer expired.
[00:00:00.024,676] <inf> app_test: info message
[00:00:00.024,685] <wrn> app_test: warning message
[00:00:00.024,688] <err> app_test: err message
```

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: Current master as of 3/2/2019 (630af8a65f114b6b247dd26fbf47e9ebeb4672d1)

mike-scott, 2019-03-02:
Assigned @andyross due to his familiarity with the latest timer changes in Zephyr.  Please assign to someone else if appropriate.

mike-scott, 2019-03-02:
If I get time, I'll try and bisect a bit later.

nashif, 2019-03-03:
duplicate of #9043

PR #14074 (2019-03-06): logging: Use k_uptime_get_32 for high frequency system clock
If system clock runs fast k_cycle_get_32(), which is the timestamp
source, wraps often (few minutes). This patch changes default
timestamp function to use k_uptime_get_32() if system clock
frequency is higher than 1 MHz and k_cycle_get_32() otherwise.

If system clock runs at 1 MHz, counter will wrap every 71.5 minutes.

Fixes #9043.
Fixes #14010.
closed 2019-03-03 by nashif