zephyrproject-rtos/zephyr #14028: [flash] question about the flash partition

summer20100514, 2019-03-04:
Hello, I am just curious about how can I add one custom flash partition into the dts? I used one overlay file to achieve that but fails with the recent branch.

summer20100514, 2019-03-04:
@galak I am eager to hear from you, thank you for your patience.

erwango, 2019-03-04:
@summer20100514 , please provide more information on your issue (base SHA1, SHA1 in use when it was working, overlay in use...). I suggest to use issue template (which you erased when entering your issue).
It is quite hard to help when we don't have inforamtion

summer20100514, 2019-03-04:
OK, I solved my problem just now. 
`#if defined(CONFIG_FS_FLASH_STORAGE_PARTITION)
		storage_partition: partition@fc000 {
			label = "storage";
			reg = <0x000fc000 0x00002000>;
		};
		storage_partition_2: partition@fe000 {
			label = "storage2";
			reg = <0x000fe000 0x00002000>;
		};
#endif`
it seemed like that when parsed the overlay file, the macro CONFIG_FS_FLASH_STORAGE_PARTITION is not defined using the lastest branch, however after compilation, the CONFIG_FS_FLASH_STORAGE_PARTITION do exist.

closed 2019-03-04 by summer20100514