zephyrproject-rtos/zephyr #14034: support for ppp protocol

shlomow, 2019-03-04:
Hi! I saw that there is a porting for a cellular device. I wonder if there is any idea for implementing the ppp layer 2 stack, since it is very convenient to handle sockets riding on ppp.

Thanks!
Shlomi

nashif, 2019-03-26:
@laperie fyi

jukkar, 2019-04-03:
I am actually working on this right now. Current plan is to implement a dedicated PPP L2 layer in networking stack, and initially support LCP, IPCP and IPv6CP. Target release would be 1.15.

shlomow, 2019-04-03:
Thanks for the reply! Since ppp is mostly used by cellular devices there will be a generic library to handle 3gpp at commands?

llisauskas, 2019-04-08:
@jukkar Are you targeting cellular devices? 
Most often embedded devices communicate with cellular modems via UART and designating it only for PPP limits modem features severely (env. monitoring such as serving and/or visible cells, SMS operations, incoming/outgoing calls and so on are unavailable if port is used by PPP). Have you thought about implementing CMUX (3GPP TS 07.10 multiplexer) to overcome this?

PR #16217 (2019-07-29): Initial PPP implementation
Initial implementation for Point-to-point protocol (PPP). It implements LCP, IPCP and IPV6CP modules. See these RFCs for details.
* LCP : https://tools.ietf.org/html/rfc1661
* HDLC : https://tools.ietf.org/html/rfc1662
* IPCP : https://tools.ietf.org/html/rfc1332
* IPV6CP : https://tools.ietf.org/html/rfc5072

This is still work-in-progress and following things need more TLC:
- [ ] Authentication support missing
- [ ] LCP non-default option negotiation
- [ ] MRU/MTU handling not yet done

Fixes #14034
commit f95938da (2019-07-29): net: ppp: Initial support for point-to-point protocol

This implements ppp L2 component, LCP and IPCP modules.

Fixes #14034
closed 2019-07-29 by jukkar