zephyrproject-rtos/zephyr #14042: MCUboot fails to boot STM32L4 device

kzyapkov, 2019-03-04:
**Describe the bug**
We've been unable to adapt our code to the latest master branches of MCUboot and zephyr, which worked fine around 1.13 and a few weeks after that. After some digging through #13249, #13408, #13433 I enabled MULTITHREADING and expected everything to just work, but, alas, when jumping to the reset vector of the app we get a crash.

**To Reproduce**
I've prepared a small git repo with the needed configs and instructions for reproducing this on a Nucleo L476RG:

https://github.com/kzyapkov/mcuboot-stm32-fault

**Expected behavior**
MCUboot starts the application which successfully says "hello".

**Impact**
Blocker.

**Screenshots or console output**

Playing with Kconfig options for both MCUboot and the app I am able to produce various types of hardware faults, but no success:

```
[00:00:00.005,196] <inf> mcuboot: Starting bootloader
[00:00:00.011,798] <inf> mcuboot: Image 0: magic=unset, copy_done=0x3, image_ok=0x3
[00:00:00.019,705] <inf> mcuboot: Scratch: magic=unset, copy_done=0x0, image_ok=0x3
[00:00:00.027,776] <inf> mcuboot: Boot source: slot 0
[00:00:00.048,246] <inf> mcuboot: Swap type: none
[00:00:00.126,291] <inf> mcuboot: Bootloader chainload address offset: 0x10000
[00:00:00.133,764] <inf> mcuboot: Jumping to the first image slot
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0xea45cc18
***** Hardware exception *****
Current thread ID = 0x20005024
Faulting instruction address = 0x80054e6
Fatal fault in essential thread! Spinning...
```
or

```
***** HARD FAULT *****
***** Hardware exception *****
Current thread ID = 0x20005024
Faulting instruction address = 0x8001b04
Fatal fault in essential thread! Spinning...
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK v0.9.5
 - zephyr@823d86097fd ; mcuboot@b26a4a9c40

**Additional context**
This is reliably reproducible with the Zephyr sample provided in MCUboot. I've only added the necessary changes so it builds for the Nucleo board.

erwango, 2019-03-06:
@kzyapkov, as a start, have you been able to verify this is working on another platform/board?

kzyapkov, 2019-03-06:
Nope, but reports in the bugtracker indicate that it does. I don't have dev boards of other platforms handy right now.

kzyapkov, 2019-03-06:
Okay, so the change from #14086 also fixes this issue. The DTS file (or overlay) needs

```
/ {
	chosen {
		zephyr,code-partition = &slot0_partition;
	};
};
```

nashif, 2019-03-06:
@kzyapkov which documentation were you following?

kzyapkov, 2019-03-07:
Coming back to this now, I find:

https://docs.zephyrproject.org/latest/guides/dts/index.html#linking-zephyr-within-a-partition

I was mostly following examples, documentation about using MCUboot with zephyr is spread over a few pages... I vaguely remember that selecting "zephyr,code-partition" wasn't necessary (an overlay was added by zephyr when BOOTLOADER_MCUBOOT=y), and it is somewhat confusing when we're using the same DTS overlay for the mcuboot build and the app.

Closing this.

closed 2019-03-07 by kzyapkov