zephyrproject-rtos/zephyr #14044: BLE HID sample fails to reconnect on Windows 10 tablets (follow-up)

Olivier-ProGlove, 2019-03-04:
Same issue as https://github.com/zephyrproject-rtos/zephyr/issues/12362 but as the other one has been closed and is quite long and I have also managed to narrow down the issue. I am re-opening this new one.

Even with @cvinayak 's fix #13983 the issue still exists.

**To Reproduce**
1. Build and flash peripheral_hids on Nordic nRF52840 dev kit (PCA10056).
2. Pair the Windows 10 tablet to the device (all good!)
3. Reset the Nordic dev kit by pressing 'BOOT/RESET'

**Expected behavior**
Windows 10 should automatically reconnect to the dev kit as it advertises on restart

* Observed result: BT_HCI_ERR_TERM_DUE_TO_MIC_FAIL errors on the first packets

**Environment (please complete the following information):**
- Latest Zephyr: b17d9ee
- @cvinayak 's fix: #13983
- Test: zephyr/samples/bluetooth/peripheral_hids/ on nrf52840_pca10056 using Windows 10 Surface Pro 4.

The issue seems to be in the Link Layer and it is a dynamic issue. That could explain why I thought @cvinayak's fix fixed my issue: https://github.com/zephyrproject-rtos/zephyr/pull/13983#issuecomment-468639617
I am not familiar enough with the Link Layer to says what is the issue. But here is what I observed with the same binary on a failing case and successful case:

# Failing case

See `test7-fail-2-with-LL_LENGTH_REQ.pcapng` for the complete trace from [sniffer-traces.zip](https://github.com/zephyrproject-rtos/zephyr/files/2927420/sniffer-traces.zip).

[image]

```
Master > Slave
    Data Header: 0x170f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [OK]
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 23
    Control Opcode: LL_ENC_REQ (0x03)

Slave > Master
    Data Header: 0x091b
        .... ..11 = LLID: Control PDU (0x3)
        .... .0.. = Next Expected Sequence Number: 0
        .... 1... = Sequence Number: 1 [OK]
        ...1 .... = More Data: True
        000. .... = RFU: 0
        Length: 9
    Control Opcode: LL_LENGTH_REQ (0x14)

Master > Slave
    Data Header: 0x020f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [OK]
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 2
    Control Opcode: LL_UNKNOWN_RSP (0x07)
    Unknown Type: LL_LENGTH_REQ (0x14)

Master > Slave
    Data Header: 0x020f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [Wrong]
            [Expert Info (Note/Sequence): Retransmission]
                [Retransmission]
                [Severity level: Note]
                [Group: Sequence]
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 2
    Control Opcode: LL_UNKNOWN_RSP (0x07)
    Unknown Type: LL_LENGTH_REQ (0x14)

Master > Slave
    Data Header: 0x020f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [Wrong]
            [Expert Info (Note/Sequence): Retransmission]
                [Retransmission]
                [Severity level: Note]
                [Group: Sequence]
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 2
    Control Opcode: LL_UNKNOWN_RSP (0x07)
    Unknown Type: LL_LENGTH_REQ (0x14)
```

# Successful case

See `test8-successful-2-with-LL_LENGTH_REQ.pcapng` for the complete trace from [sniffer-traces.zip](https://github.com/zephyrproject-rtos/zephyr/files/2927420/sniffer-traces.zip).

[image]

```
Slave > Master
    Data Header: 0x0917
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 0... = Sequence Number: 0 [OK]
        ...1 .... = More Data: True
        000. .... = RFU: 0
        Length: 9
    Control Opcode: LL_LENGTH_REQ (0x14)

Master > Slave
    Data Header: 0x171f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [OK]
        ...1 .... = More Data: True
        000. .... = RFU: 0
        Length: 23
    Control Opcode: LL_ENC_REQ (0x03)

Master > Slave
    Data Header: 0x0203
        .... ..11 = LLID: Control PDU (0x3)
        .... .0.. = Next Expected Sequence Number: 0
        .... 0... = Sequence Number: 0 [OK]
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 2
    Control Opcode: LL_UNKNOWN_RSP (0x07)
    Unknown Type: LL_LENGTH_REQ (0x14)

Slave > Master
    Data Header: 0x0d1b
        .... ..11 = LLID: Control PDU (0x3)
        .... .0.. = Next Expected Sequence Number: 0
        .... 1... = Sequence Number: 1 [OK]
        ...1 .... = More Data: True
        000. .... = RFU: 0
        Length: 13
    Control Opcode: LL_ENC_RSP (0x04)
```

More Wireshark Sniffer traces can be found in this archive [sniffer-traces.zip](https://github.com/zephyrproject-rtos/zephyr/files/2927420/sniffer-traces.zip). I have also learned how to only save a subset of the Wireshark traces. So no more thousand lines traces :-)
Tests traces are prefixed with `testN-successful-` when the reconnection succeeded and `testN-fail-` when reconnection failed. It looks like they all follow the same pattern as my extracted traces above.

A workaround to this issue seems to be to disable all `LL` commands before `LL_ENC_REQ`:

```
# Disable LL_SLAVE_FEATURE_REQ:
CONFIG_BT_CTLR_SLAVE_FEAT_REQ=n

# Disable LL_PHY_REQ
CONFIG_BT_PHY_UPDATE=n

# Disable LL_LENGTH_REQ
CONFIG_BT_DATA_LEN_UPDATE=n
```

cc: @carlescufi @pdunaj @Qbicz @jhn-nordic @wiminordic @jhedberg @FrankGorgenyi @joerchan @vcgomes cvinayak

Olivier-ProGlove, 2019-03-04:
It could be noticed in the failing case a retransmission request from Windows 10 BLE controller:

```
Master > Slave
    Data Header: 0x020f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [Wrong]
            [Expert Info (Note/Sequence): Retransmission]
                [Retransmission]
                [Severity level: Note]
                [Group: Sequence]
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 2
    Control Opcode: LL_UNKNOWN_RSP (0x07)
    Unknown Type: LL_LENGTH_REQ (0x14)
```

carlescufi, 2019-03-05:
I believe we might need the same fix we did for slave feature request but also for the rest of control procedures that are not known to those old controllers.
Agree @cvinayak?

carlescufi, 2019-03-05:
@Olivier-ProGlove 

>A workaround to this issue seems to be to disable all LL commands before LL_ENC_REQ:
># Disable LL_SLAVE_FEATURE_REQ:
>CONFIG_BT_CTLR_SLAVE_FEAT_REQ=n

It should no longer be required to have to disable that one. Can you confirm that you only need to disable `CONFIG_BT_PHY_UPDATE` and `CONFIG_BT_DATA_LEN_UPDATE`?

Olivier-ProGlove, 2019-03-06:
@carlescufi As I mentioned later in @cvinayak's pull-request https://github.com/zephyrproject-rtos/zephyr/pull/13983#issuecomment-469174447 and https://github.com/zephyrproject-rtos/zephyr/issues/12362#issuecomment-468704354, the PR does not actually fix this issue. The issue looks to be timing dependent, and I might have been "lucky" when I tested the potential fix.

In the archive attached to the first message of this new issue ([sniffer-traces.zip](https://github.com/zephyrproject-rtos/zephyr/files/2927420/sniffer-traces.zip)), I actually have traces with `CONFIG_BT_CTLR_SLAVE_FEAT_REQ=y` (and  @cvinayak's fix) when the reconnection works and when it does not work:

### Successful reconnection with CONFIG_BT_CTLR_SLAVE_FEAT_REQ=y

Sniffer Trace: test1-successful.pcapng

[image]

### Failing reconnection with CONFIG_BT_CTLR_SLAVE_FEAT_REQ=y

Sniffer Trace: test2-fail-with-LL_SLAVE_FEATURE_REQ.pcapng

[image]

```
[Frame 5] Master > Slave
    Data Header: 0x170f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [OK]
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 23
    Control Opcode: LL_ENC_REQ (0x03)

[Frame 6] Slave > Master
    Data Header: 0x091b
        .... ..11 = LLID: Control PDU (0x3)
        .... .0.. = Next Expected Sequence Number: 0
        .... 1... = Sequence Number: 1 [OK]
        ...1 .... = More Data: True
        000. .... = RFU: 0
        Length: 9
    Control Opcode: LL_SLAVE_FEATURE_REQ (0x0e)

[Frame 8] Slave > Master
    Data Header: 0x0606
        .... ..10 = LLID: Start of an L2CAP message or a complete L2CAP message with no fragmentation (0x2)
        .... .1.. = Next Expected Sequence Number: 1
        .... 0... = Sequence Number: 0 [OK]
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 6
    [L2CAP Index: 0]

[Frame 9] Master > Slave
    Data Header: 0x020f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [OK] <----- Wireshark says the sequence number is OK
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 2
    Control Opcode: LL_UNKNOWN_RSP (0x07)
    Unknown Type: LL_SLAVE_FEATURE_REQ (0x0e)

[Frame 10] Master > Slave
    Data Header: 0x020f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [Wrong] <----- Wireshark says the sequence number is WRONG!!!
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 2
    Control Opcode: LL_UNKNOWN_RSP (0x07)
    Unknown Type: LL_SLAVE_FEATURE_REQ (0x0e)

[Frame 11] Master > Slave
    Data Header: 0x020f
        .... ..11 = LLID: Control PDU (0x3)
        .... .1.. = Next Expected Sequence Number: 1
        .... 1... = Sequence Number: 1 [Wrong] <--- Wireshark says the sequence number is WRONG!!!
        ...0 .... = More Data: False
        000. .... = RFU: 0
        Length: 2
    Control Opcode: LL_UNKNOWN_RSP (0x07)
    Unknown Type: LL_SLAVE_FEATURE_REQ (0x0e)
```

Olivier-ProGlove, 2019-03-06:
Tested latest upstream (https://github.com/zephyrproject-rtos/zephyr/commit/cb874235fd741a0d9431cc79c301bf749224cd99) without any configuration change with Chuwi tablet (Realtek Semiconductor Driver).

- Sniffer Trace: [chuwi-test1-reconnection-fail.pcapng.zip](https://github.com/zephyrproject-rtos/zephyr/files/2936541/chuwi-test1-reconnection-fail.pcapng.zip)
- Same issue "Wrong Sequence Number": [image]

Olivier-ProGlove, 2019-03-06:
When using ASUS-BT400 Bluetooth dongle, the reconnection always works and trace are quite different:

[image]

[image]

Sniffer Trace for these two reconnections: [asus-reconnection.zip](https://github.com/zephyrproject-rtos/zephyr/files/2936767/asus-reconnection.zip)

carlescufi, 2019-03-06:
@Olivier-ProGlove OK so what I think is happening is that the original fix that @cvinayak posted is not covering all possible sequences, so we need to take another look at it. The other issues when you see it failing because of `PHY` or `LEN` requests I think it must be that we are not allowing that right now. So this need a bit more assessment from @cvinayak.

FrankGorgenyi, 2019-03-07:
We have previously observed that some radios did not serialize the Link Layer Encryption procedure to be atomic compared to other LL data and control PDUs.

The spec has the following quote on this matter:

[1] “If at any time during the encryption start procedure, the Link Layer of the
master or the slave receives an unexpected Data Channel PDU from the peer
Link Layer, it shall immediately exit the Connection State, and shall transition to
the Standby State. The Host shall be notified that the link has been
disconnected with the error code Connection Terminated Due to MIC Failure
(0x3D).” (Core Spec Vol. 6 Part B. Section 5.1.3.1)

Here, the “Data Channel PDU” refers to both LL Data PDUs and LL Control PDUs that are not related to the encryption procedure (thus “unexpected”). The radio is only allowed to resume sending other LL PDUs after the encryption procedure is completed. From the spec:

[2] “When the Link Layer of the master receives the LL_START_ENC_RSP PDU,
the connection is encrypted. The Link Layer can now send LL Data PDUs and
LL Control PDUs; these PDUs will be encrypted.” (Core Spec Vol. 6 Part B. Section 5.1.3.1)

One suggested spec-compliant sequence when a race occurs is:

1. Radio A (Master) starts the encryption procedure by sending LL_ ENC_REQ
2. Radio B (Slave) sends a non-encryption LL PDU request, including LL_SLAVE_FEATURE_REQ (a race with the encryption procedure)
3. Radio A (Master) receives non-encryption LL PDU but should not respond since it knew it just started the encryption procedure and must wait for the encryption procedure to complete before continuing other LL procedures [1]
4. Radio B (Slave) receives LL_ ENC_REQ and respond with LL_ ENC_RSP
5. Radio A and Radio B continues the encryption procedures by exchanging LL_START_ENC_REQ and LL_START_ENC_RSP
6. Radio A (Master) receives LL_START_ENC_RSP which marks the completion of the encryption procedure
7. Radio A (Master) can then resume any paused LL procedures, such as the previous LL_SLAVE_FEATURE_REQ. It can respond accordingly, such as with LL_UNKNOWN_RSP, which will be now encrypted [2].
8. Radio B (Slave) receives the LL_UNKNOWN_RSP to complete their original procedure.

There is no good and reliable way to workaround this without a fix in the radio. Changing the timing by disabling some LL procedures can potentially reduce the repro rate with a particular system, but is not guaranteed to be a reliable solution. This is why not every device fails to reconnect even on systems that have this issue, due to the nature of the timing.

If you are encountering the race condition described above on a specific radio and it’s not handling it correctly, we should raise this issue with the radio manufacturer to fix it. Marvell already has a fix in the pipeline. However, your observation that the same issue exists on a Realtek radio is new information to us. We can raise this issue with them also, however we cannot comment on timeline for a fix from them.

cvinayak, 2019-03-12:
> 
> 
> I believe we might need the same fix we did for slave feature request but also for the rest of control procedures that are not known to those old controllers.
> Agree @cvinayak?

The fix allows unknown response for any opcode to be permitted during encryption setup, has the PR applied correctly? can the latest upstream master be used to confirm?

cvinayak, 2019-03-12:
@Olivier-ProGlove could you please email (in commit messages) me, so we can arrange for a Skype meeting to reproduce and fix, if possible without breaking the BT Spec, the IOP issue?

cvinayak, 2019-03-22:
@Olivier-ProGlove Please give a try with #14836 with added print of the flags where the MIC failure is generated in ctrl.c. I have moved the resetting of the enc_wa flag to the end of Enc Setup Procedure now.

joerchan, 2019-03-27:
The issue here is dependent on the exact sequence of control packets between the master and slave.

This sequence is this:
M -> S LL_ENC_REQ
M <- S LL_LENGTH_REQ/LL_SLAVE_FEATURE_REQ (not really important which)

M -> S LL_UNKNOWN_RSP

The reason why the workaround for #13983 did not work was that the LL has enqueued the LL_ENC_RSP when it receives LL_UNKNOWN_RSP, and it has already started to receive encrypted packet.

I have discussed this with @cvinayak and I'm currently working on a fix.

joerchan, 2019-03-27:
The UNKNOWN RSP has been added in Core Spec version 5.0
So this is no longer a workaround, but a spec requirement.

> The Link Layer of the slave shall then send an LL_ENC_RSP PDU. The Link
> Layer of the slave shall then notify the Host with the Rand and EDIV fields.
> After having sent the LL_ENC_RSP PDU, the Link Layer of the slave can
> receive an LL_UNKNOWN_RSP PDU corresponding to a LL Control PDU sent
> by the slave. The slave should not disconnect the link in this case.

joerchan, 2019-03-27:
@Olivier-ProGlove
Can you confirm that the fix in #14913 resolves this issue?

cvinayak, 2019-03-27:
@FrankGorgenyi  @carlescufi @Olivier-ProGlove FYI https://github.com/zephyrproject-rtos/zephyr/issues/14044#issuecomment-477160491

commit 474e8c5f (2019-03-26): Bluetooth: controller: Enable enc on RX after ENC_RSP acked

The first encrypted sent by the central is the START_ENC_REQ
packet after the peripheral has sent START_ENC_REQ.
The slave enqueued both ENC_RSP and START_ENC_REQ and enabled
encryption on rx side.
This caused MIC failure when the master sends an UNKNOWN_RSP
or REJECT_IND to terminate any colliding LL control procedures

Fixes #14044
commit 252c96a1 (2019-03-27): Bluetooth: controller: Fix unknown rsp received during enc procedure

The master is using unknown rsp to terminate slave side initiated
procedures that has collided with the encryption procedure initiated by
the master.
We need to handle an unknown response that is sent in unencrypted during
the encryption procedure, even though we have already set up to receive
encrypted packets.

Fixes #14044
commit c6082c76 (2019-03-27): Bluetooth: controller: Fix unknown rsp received during enc procedure

The master is using unknown rsp to terminate slave side initiated
procedures that has collided with the encryption procedure initiated by
the master.
We need to handle an unknown response that is sent in unencrypted during
the encryption procedure, even though we have already set up to receive
encrypted packets.

Fixes #14044
commit 273796e1 (2019-03-28): Bluetooth: controller: Fix unknown rsp received during enc procedure

The master is using unknown rsp to terminate slave side initiated
procedures that has collided with the encryption procedure initiated by
the master.
We need to handle an unknown response that is sent in unencrypted during
the encryption procedure, even though we have already set up to receive
encrypted packets.

Fixes #14044