zephyrproject-rtos/zephyr #14063: net: ipv6: Neighbor table management improvements

lbise, 2019-03-05:
I've been using a setup where potentially a large number of neighbor can be added to the nbr table. With this I noticed that the neighbor management generally speaking could use some improvements and fixes.

Some issues I found:

Currently when a neighbor is added to the table, it is never ever removed. I think the intent was to remove neighbors from the table on 3 conditions, if it reaches the stale state and stays stale for a certain amount of time, if it reaches the probe state and the probe fails or if it reaches the incomplete state and the probe fails.
However currently all these 3 cases do not remove the neighbors from the table. The culprit is in the ipv6_nbr.c function ipv6_nd_reachable_timeout (https://github.com/zephyrproject-rtos/zephyr/blob/69836455521656fbaeb16c35f48621f26b56a9a5/subsys/net/ip/ipv6_nbr.c#L1256).

Stale entries are not remove because ipv6_nd_reachable_timeout NET_IPV6_NBR_STATE_STALE is never actually executed, this is because when going from the REACHABLE to the STALE state the reachable timer is not set therefore the nbr will always be skipped (this is checked at the beginning of the function).

Probe entries are a bit different, the nbr can be removed but only if it is a router? Not really sure I understand why (https://github.com/zephyrproject-rtos/zephyr/blob/69836455521656fbaeb16c35f48621f26b56a9a5/subsys/net/ip/ipv6_nbr.c#L1345). Otherwise nothing is done and the nbr is left in the table.

Incomplete entries, which are similar functionally to probe are again not working properly, in that only the first probe will be sent, after that none will be sent again because the reachable timer is not set (https://github.com/zephyrproject-rtos/zephyr/blob/69836455521656fbaeb16c35f48621f26b56a9a5/subsys/net/ip/ipv6_nbr.c#L1303).

Another issue I found is that, when attempting to free a nbr they use the nbr_free function, but I think they should be using net_ipv6_nbr_rm because otherwise if the nbr was part of a route, the reference will not be removed.

Finally, I think the restart of the delayed work could be improved, currently the first work not elapsed will trigger a work submit (https://github.com/zephyrproject-rtos/zephyr/blob/69836455521656fbaeb16c35f48621f26b56a9a5/subsys/net/ip/ipv6_nbr.c#L1281), so if it's timeout is bigger than the next item, the second item will be delayed by this much. This might not be noticeable but if stale timeout is added, and it's several minutes it might not work properly.

I think that sums up the issues I found.

laperie, 2019-03-05:
This looks more like a bug; hence changing labels and assigning to @jukkar. 
Thanks @lbise  for reporting!

nashif, 2019-03-06:
@laperie set priority please

jukkar, 2019-03-07:
@lbise Thanks for the detailed analysis. Did it cause any major issues when nbr table got full with stale neighbors? I am pondering if we could do the improvements in 1.15 release.

laperie, 2019-03-07:
While waiting for @lbise to answer, setting priority:low as we have no other reports of this problem

lbise, 2019-03-08:
The system kept working it was just incapable of communicating with the neighbors that could not be added to the table. But nothing major like crash or anything like that.

jukkar, 2019-03-08:
> The system kept working it was just incapable of communicating with the neighbors that could not be added to the table.

Ok, this is indeed bad too. Lets try to fix this as there is certainly impact to functionality of the device.

rveerama1, 2019-03-15:
@lbise I tried to address as much as possible (https://github.com/zephyrproject-rtos/zephyr/pull/14557). Can you please test this with your environment and let me know?

lbise, 2019-03-18:
Thank you @rveerama1. I'll give it a try and report back.

commit f361859f (2019-03-26): net: ipv6: Fix IPv6 neighbor table

This patch fixes following issues.

  * If IPv6 neighbor table is full, stack can not add any new
    neighbors. So stale counter is introduced. Whenever neighbor
    enters into STALE state, stale counter will be incremented
    by one. When table is full and if stack wants to add new
    neighbor, oldest neighbor in STALE state will be removed
    and new neighbor will be added.

  * When neighbor is in PROBE state and when it exceeds max
    number of PROBEs, only neighbor with router is removed.
    As per RFC 4861 Appendix C, entry can be discarded. Now [... cut]
closed 2019-03-26 by nashif