zephyrproject-rtos/zephyr #14065: Save bonds in workqueue thread instead on tx thread

pdunaj, 2019-03-05:
**Is your enhancement proposal related to a problem? Please describe.**
When central connects to peripheral it needs to store bonding information. This happens in tx thread context and as settings subsystem uses a lot of RAM it significantly increases stack size required by tx thread.

**Describe the solution you'd like**
Bond store could be moved to workqueue thread. This would allow reducing stack size in tx thread while workqueue thread stack should be sufficient.

**Describe alternatives you've considered**
Tx thread stack is increased at the moment but this causes around 400B of RAM to be wasted. This is unacceptable for some SoCs.

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/issues/13585#issuecomment-467448270

jori-nordic, 2024-02-09:
@pdunaj I think this might be addressed by https://github.com/zephyrproject-rtos/zephyr/pull/68394

Could you confirm?

jori-nordic, 2024-02-29:
I think this is fixed. Re-open if still applicable.

closed 2024-02-29 by jori-nordic