zephyrproject-rtos/zephyr #14066: stm32: power: system power management support

KwonTae-young, 2019-03-05:
**Is your feature request related to a problem? Please describe.**
`STM32` soc must be able to use the `system power management` function.

**Describe the solution you'd like**
I would like to switch between `sleep mode`,` stop mode`, and `standby mode` in the application.

**Describe alternatives you've considered**
I want to use the `system power management` function in `STM32` in the form [samples/boards/nrf52/power_mgr](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/boards/nrf52/power_mgr).

**Additional context**
I have tried testing ` system power management` with [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html).
What I want to do is switch and recover the sleep mode in the application.

I have enabled the `CONFIG_SYS_POWER_MANAGEMENT` setting in Zephyr's Kernel configuration.
And when I try to compile, the following error is output.
```
root:build# pwd
/tmp/zephyrproject/zephyr/samples/hello_world/build
root:build# 
root:build# cat zephyr/.config | grep SYS_POWER_MANAGEMENT
CONFIG_SYS_POWER_MANAGEMENT=y
root:build# 
root:build# make -j8
[  0%] Built target syscall_macros_h_target
[  1%] Built target driver_validation_h_target
[  2%] Built target kobj_types_h_target
[  3%] Generating misc/generated/syscalls.json
[  6%] Built target syscall_list_h_target
[  7%] Built target offsets
[  7%] Built target offsets_h
[  8%] Built target linker_script_target
[  9%] Built target app
[ 19%] Built target lib__libc__minimal
[ 22%] Built target arch__arm__core__cortex_m__mpu
[ 23%] Built target drivers__gpio
[ 29%] Built target arch__arm__core__cortex_m
[ 30%] Building ASM object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/cpu_idle.S.obj
[ 32%] Built target boards__arm__stm32f4_disco
[ 35%] Built target drivers__pwm
[ 36%] Building ASM object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/isr_wrapper.S.obj
[ 37%] Built target drivers__serial
[ 63%] Built target zephyr
[ 83%] Built target kernel
/tmp/zephyrproject/zephyr/arch/arm/core/isr_wrapper.S: Assembler messages:
/tmp/zephyrproject/zephyr/arch/arm/core/isr_wrapper.S:68: Error: cannot represent T32_OFFSET_IMM relocation in this object file format
/tmp/zephyrproject/zephyr/arch/arm/core/isr_wrapper.S:83: Error: cannot represent T32_OFFSET_IMM relocation in this object file format
zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/build.make:186: recipe for target 'zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/isr_wrapper.S.obj' failed
make[2]: *** [zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/isr_wrapper.S.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
/tmp/zephyrproject/zephyr/arch/arm/core/cpu_idle.S: Assembler messages:
/tmp/zephyrproject/zephyr/arch/arm/core/cpu_idle.S:75: Error: cannot represent T32_OFFSET_IMM relocation in this object file format
/tmp/zephyrproject/zephyr/arch/arm/core/cpu_idle.S:94: Error: cannot represent T32_OFFSET_IMM relocation in this object file format
zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/build.make:137: recipe for target 'zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/cpu_idle.S.obj' failed
make[2]: *** [zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/cpu_idle.S.obj] Error 1
CMakeFiles/Makefile2:813: recipe for target 'zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/all' failed
make[1]: *** [zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
root:build# 
```

I searched soc that supports `system power management` with the following command.
As far as I'm concerned, I think that only `nordic` and `silabs` are currently supported.(ARM)
```
root:zephyr# pwd
/tmp/zephyrproject/zephyr
root:zephyr# 
root:zephyr# grep -r "SYS_POWER_MANAGEMENT" soc/arm
soc/arm/nordic_nrf/nrf91/Kconfig.defconfig.series:config SYS_POWER_MANAGEMENT
soc/arm/nordic_nrf/nrf52/Kconfig.defconfig.series:config SYS_POWER_MANAGEMENT
soc/arm/nordic_nrf/nrf52/CMakeLists.txt:zephyr_sources_ifdef(CONFIG_SYS_POWER_MANAGEMENT
soc/arm/nordic_nrf/nrf51/Kconfig.defconfig.series:config SYS_POWER_MANAGEMENT
soc/arm/nordic_nrf/nrf51/CMakeLists.txt:zephyr_sources_ifdef(CONFIG_SYS_POWER_MANAGEMENT
soc/arm/silabs_exx32/Kconfig.defconfig:if SYS_POWER_MANAGEMENT
soc/arm/silabs_exx32/Kconfig.defconfig:endif # SYS_POWER_MANAGEMENT
soc/arm/silabs_exx32/common/CMakeLists.txt:zephyr_sources_ifdef(CONFIG_SYS_POWER_MANAGEMENT soc_power.c)
root:zephyr# 
```

Does the `STM32` platform currently support `system power management`?
Or is it an error due to the immaturity of my Zephyr usage?

I'm sorry I do not have enough English.
Thank you.

kzyapkov, 2019-03-05:
This is a much-needed feature by many, us included. Proper power management support would also require device power management and is not a trivial effort. However, maybe it would be possible to at least add an example implementing the needed functions out-of-tree, with a plain sample application.

gon1332, 2019-03-05:
The Zephyr Power Management support seems to be under changes for this period. Locally, I have some samples consindering entering/exiting to/from LPMs on STM32L4, but they use the STM32 LL drivers. If you want me to upload anything in my local Zephyr fork, let me know.

Otherwise, I'm planning to create a PR on this, utilizing Zephyr Power Management.

kzyapkov, 2019-03-05:
@gon1332 I also have a somewhat-working sample for STOP1/STOP2 -- let's compare notes and maybe collaborate on a PR which shows how to do the basic power management integration. The issues I'm facing are around systick management and device power states.

gon1332, 2019-03-05:
@kzyapkov I agree the STOP modes, due to the fact that the system doesn't reset (as in Standby or Shutdown) are a bit tricky. That's because, one should set the clock frequency and the systick appropriately after exiting from them. As for the device power management, I haven't gone deeper.

So yes, we can share some knowledge. Maybe a PR would be perfect.

KwonTae-young, 2019-03-06:
Hi. @kzyapkov, @gon1332
I saw examples of using STM32's LL driver in @gon1332's [lpm_stm32](https://github.com/gon1332/zephyr/tree/lpm_stm32) branch.
I have a [stm32f4_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html) board.
I will test the power management by referring to [lpm_stm32](https://github.com/gon1332/zephyr/tree/lpm_stm32) on my board and share it here.

@kzyapkov,
Can you share the sample code in which STOP1 / STOP2 works?
I could not find it in your repository.
I would like to test it by referring to it.

gon1332, 2019-03-06:
@KwonTae-young 
I believe that the https://github.com/gon1332/zephyr/tree/lpm_stm32 is rather unstable and there's high possibility that isn't working. If you don't mind, I'll upload other simpler examples of how to enter an LPM using STM32's LL API.

KwonTae-young, 2019-03-06:
@gon1332
Thank you.
[lpm_stm32](https://github.com/gon1332/zephyr/tree/lpm_stm32) is a good example for me.
First of all, I'm going to test it in its current state.
If you upload a simple example, it will help me a lot.

KwonTae-young, 2019-03-07:
I tested the standby mode by referring to [lpm_stm32](https://github.com/gon1332/zephyr/tree/lpm_stm32).
I worked with the `STM32 LL Driver` in my application.

STOP1/STOP2 does not exist in STM32F4.
https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_ll_pwr.h#L115-L128
``` c
/** @defgroup PWR_LL_EC_MODE_PWR Mode Power
  * @{
  */
#define LL_PWR_MODE_STOP_MAINREGU             0x00000000U                    /*!< Enter Stop mode when the CPU enters deepsleep */
#define LL_PWR_MODE_STOP_LPREGU               (PWR_CR_LPDS)                  /*!< Enter Stop mode (with low power Regulator ON) when the CPU enters deepsleep */
#if defined(PWR_CR_MRUDS) && defined(PWR_CR_LPUDS) && defined(PWR_CR_FPDS)
#define LL_PWR_MODE_STOP_MAINREGU_UNDERDRIVE  (PWR_CR_MRUDS | PWR_CR_FPDS)                 /*!< Enter Stop mode (with main Regulator in under-drive mode) when the CPU enters deepsleep */
#define LL_PWR_MODE_STOP_LPREGU_UNDERDRIVE    (PWR_CR_LPDS | PWR_CR_LPUDS | PWR_CR_FPDS)   /*!< Enter Stop mode (with low power Regulator in under-drive mode) when the CPU enters deepsleep */
#endif /* PWR_CR_MRUDS && PWR_CR_LPUDS && PWR_CR_FPDS */
#if defined(PWR_CR_MRLVDS) && defined(PWR_CR_LPLVDS) && defined(PWR_CR_FPDS)
#define LL_PWR_MODE_STOP_MAINREGU_DEEPSLEEP  (PWR_CR_MRLVDS | PWR_CR_FPDS)                 /*!< Enter Stop mode (with main Regulator in Deep Sleep mode) when the CPU enters deepsleep */
#define LL_PWR_MODE_STOP_LPREGU_DEEPSLEEP    (PWR_CR_LPDS | PWR_CR_LPLVDS | PWR_CR_FPDS)   /*!< Enter Stop mode (with low power Regulator in Deep Sleep mode) when the CPU enters deepsleep */
#endif /* PWR_CR_MRLVDS && PWR_CR_LPLVDS && PWR_CR_FPDS */
#define LL_PWR_MODE_STANDBY                   (PWR_CR_PDDS)                  /*!< Enter Standby mode when the CPU enters deepsleep */
```

The source of the tested program is shown below.
The main operations are as follows.
1. The message is output every 1 second.
2. Press and hold User Buttons(PA0) for 3 seconds to switch to Standby mode.
3. In Standby mode, press the User Buttton(PA0) to wake up.
``` c
#include <zephyr.h>
#include <misc/printk.h>
#include <stm32f4xx_ll_pwr.h>
#include <stm32f4xx_ll_cortex.h>
#include <device.h>
#include <gpio.h>

static struct device *gpioa;
static struct gpio_callback gpiocb;

static struct k_timer timer;

static int standby_mode_cnt = 0;

void sleep_mode_button(struct device *gpioc, struct gpio_callback *cb,
                    uint32_t pins)

{
	k_timer_start(&timer, 100, 0);
}

void configure_gpio(void)
{
	gpioa = device_get_binding("GPIOA");
	if (!gpioa) {
		printk("Error\n");
		return;
	}

	gpio_pin_configure(gpioa, 0, GPIO_DIR_IN
		                        | GPIO_INT | GPIO_INT_EDGE
		                        | GPIO_INT_ACTIVE_HIGH);
	gpio_init_callback(&gpiocb, sleep_mode_button, BIT(0));
	gpio_add_callback(gpioa, &gpiocb);
	gpio_pin_enable_callback(gpioa, 0);
}

/*
 * Press and hold the user button(LL_PWR_WAKEUP_PIN1: PA0) for 3 seconds to enter standby mode.
 */
static void sleep_mode_timer(struct k_timer *work)
{
	int value;

	gpio_pin_read(gpioa, 0, &value);
	if (value == 1) {
		standby_mode_cnt++;
		if (standby_mode_cnt == 30) {
			/*
			 * In stm32f4_disco, LL_PWR_WAKEUP_PIN1 is PA0.
			 * PA0 is also built in as a User Button.
			 */
			LL_PWR_DisableWakeUpPin(LL_PWR_WAKEUP_PIN1);
			LL_PWR_EnableWakeUpPin(LL_PWR_WAKEUP_PIN1);
			LL_PWR_ClearFlag_WU();
			if (LL_PWR_IsActiveFlag_WU()) {
				LL_PWR_ClearFlag_WU();
			}
			LL_PWR_SetPowerMode(LL_PWR_MODE_STANDBY);

			printk("Now call LL_LPM_EnableDeepSleep().\n");
			LL_LPM_EnableDeepSleep();
		}
	
		k_timer_start(&timer, 100, 0);
	} else {
		standby_mode_cnt = 0;
	}

}

void main(void)
{
	printk("standby mode test: %s\n", CONFIG_BOARD);

	configure_gpio();

	k_timer_init(&timer, sleep_mode_timer, NULL);

	while (1) {
		k_sleep(1000);
		printk("You can not see this message in standby mode.\n");
	}
}
```

The operation is shown in the following video.
https://youtu.be/tMrZcj1lf6Q

`Stop Mode` has not been tested yet.
I have not measured the current yet.

gon1332, 2019-03-07:
@KwonTae-young That's the spirit. :) 
I hope today that I also provide with some examples here for STOP modes.

KwonTae-young, 2019-03-08:
I tested `STOP mode`.
The tested source has been uploaded to my [lpm_stm32](https://github.com/KwonTae-young/zephyr/tree/lpm_stm32) branch.

One strange thing was that I woke up from `STOP mode' automatically after about 1 second even though I did not issue an interrupt.
So I used [LL_SYSTICK_DisableIT()](https://github.com/zephyrproject-rtos/zephyr/blob/b1798721cc68112962bb388eaba81dcbcb3bca9a/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_ll_cortex.h#L294-L302).
I do not know the exact reason yet.
https://github.com/KwonTae-young/zephyr/blob/lpm_stm32/samples/boards/stm32f4_disco/stop_mode/src/main.c#L77-L84
``` c
/*
 * When the user button interrupts, it wakes up in STOP mode.
 * However, STOP Mode is automatically released after about 1 second even if the User Button does not generate an interrupt.
 * I thought there was a hardware interrupt that I do not know.
 * So I disabled SYSTICK by referring to https://community.st.com/s/question/0D50X00009XkXphSAF/unwanted-stop-mode-wakeup-stm32l052k8.
 * I do not know the exact reason yet.
 */
LL_SYSTICK_DisableIT();
```

Below is a test video.
https://youtu.be/07QW2vB4T6A

gon1332, 2019-03-08:
@KwonTae-young I remember experiencing the same behaviour but with Standby mode. So I clear  my wake up flags on initialization, eg.

```c
if (LL_PWR_IsActiveFlag_WU2) {
    LL_PWR_ClearFlag_WU2();
}
```

KwonTae-young, 2019-03-08:
@gon1332
I also did the same in `standby mode`.
https://github.com/KwonTae-young/zephyr/blob/lpm_stm32/samples/boards/stm32f4_disco/standby_mode/src/main.c#L58-L60
``` c
if (LL_PWR_IsActiveFlag_WU()) {
        LL_PWR_ClearFlag_WU();
}
```
However, there is no effect on `stop mode`.

kzyapkov, 2019-03-08:
Yes, the systick remains ticking in stop mode for some reason. If you have logging enabled, it schedules a wakeup every second:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/logging/log_core.c#L685

but even without async logging I see wakeups every second. So one must stop the systick for stop mode and then re-enable it on wakeup. I see other platforms have alternative systick sources (like nrf_rtc_timer). So an alternative timer should be used while the systick is stopped, and the passed time should be z_clock_announce()-d. I could not get this to work reliably though.

KwonTae-young, 2019-03-08:
@kzyapkov 
I understood that [LL_SYSTICK_DisableIT()](https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_ll_cortex.h#L294-L302) was needed in `stop mode`.
Later, I think we need [LL_SYSTICK_DisableIT()](https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_ll_cortex.h#L294-L302) when adding stop mode to [sys_set_power_state()](https://docs.zephyrproject.org/latest/reference/power_management/index.html?highlight=sys_power_state_post_ops#_CPPv319sys_set_power_state12power_states).
https://github.com/gon1332/zephyr/blob/lpm_stm32/soc/arm/st_stm32/stm32l4/power.c#L19-L62
``` c
/* Invoke Low Power/System Off specific Tasks */
void sys_set_power_state(enum power_states state)
{
	switch (state) {
#ifdef CONFIG_SYS_POWER_DEEP_SLEEP
#ifdef CONFIG_SYS_POWER_STATE_DEEP_SLEEP_SUPPORTED
		case SYS_POWER_STATE_DEEP_SLEEP:
            /* Enable Backup Access for RTC registers */
            // LL_PWR_EnableBkUpAccess();

            /*
             * Configure wake up sources
             */
            /* Disable all used wakeup sources: WKUP pin*/
            LL_PWR_DisableWakeUpPin(LL_PWR_WAKEUP_PIN2);

            /* Clear wakeup Flag */
            LL_PWR_ClearFlag_WU2();

            LL_PWR_SetWakeUpPinPolarityHigh(LL_PWR_WAKEUP_PIN2);

            /* Enable wakeup pin WKUP2 */
            LL_PWR_EnableWakeUpPin(LL_PWR_WAKEUP_PIN2);

            /*
             * Enter Standby mode
             */
            /* Set STANDBY mode when CPU enters deepsleep */
            LL_PWR_SetPowerMode(LL_PWR_MODE_STANDBY);

            /* Set SLEEPDEEP bit of Cortex System Control Register */
            LL_LPM_EnableDeepSleep();

            /* Request Wait For Interrupt */
            __WFI();

			break;
#endif
#endif
		default:
			LOG_ERR("Unsupported power state %u", state);
			break;
	}
}
```

psidhu, 2019-04-16:
Has there been any movement on creating a PR for this? I can help test on the STM32L4+ series.

erwango, 2019-07-04:
^^FRASTM

erwango, 2019-07-26:
@KwonTae-young, @kzyapkov, @gon1332, @psidhu:

We're starting to investigate zephyr power management support for STM32, with a first focus on STM32WB. Since this is a pretty wide topic and there are many possible use cases and ways to implement them, we'd like to have information about the use cases STM32 users would be interested to use in Zephyr.

It would be nice to have an overview of the use cases you'd like to be supported, with, for each use case:
- switched off peripherals
- sleep duration
- wake up source
- wake up time constraints if any
- optionally: the STM32 low power modes you expect to be supported (if already known).

Thanks

psidhu, 2019-07-26:
@erwango Hi. Thanks for taking the time to get into this. I'm actually thinking of using the STM32WB55 for my next project, but I'll need to investigating FCC etc licensing first (needs to be pre-certified). Anyways, what I mean is I can help when possible.

> It would be nice to have an overview of the use cases you'd like to be supported, with, for each use case:
> 
>     * switched off peripherals

I think this is most important if going into Standby mode. I'm not sure how it's done today in Zephyr, but I would assume that a 'parent' peripheral can't be suspended until child nodes are suspended (SPI devices, then the SPI peripheral itself).

Having power 'profiles' that are on-the-fly configurable would support this very well. E.g. keep bt le radio on if connected when going to STOP1, but off if nothing was connected before going to STOP1. Application code could configure this.

>     * sleep duration

A configurable sleep duration that is cancelled if woken early is important. Specifically using RTC alarms, even in shutdown mode. This may have to be a manual step.

>     * wake up source

Configuring these before sleep is important to me (EXTI or WKUP pins).

>     * wake up time constraints if any
>     * optionally: the STM32 low power modes you expect to be supported (if already known).

Going into Standby/Shutdown mode is required, with a continuous LSE so the RTC stays alive. On wake, don't erase current date.

Going into at least STOP1 and STOP2 is required I think. Imagine a sensor device that wakes every .1s to take a reading and go back into STOP2. When a user comes along to wake it, they can grab sensor data (e.g. connect via Bluetooth).

I think it would be interesting to have a method to switch from Run to Low-Power Run modes. I'm not sure how the clocking should get done as there's no real clock tree configuration in Zephyr today.

KwonTae-young, 2019-07-29:
I am mainly using STM32F4.
In STM32F4, `stadnby mode` and `watchdog` can not be used at the same time.
The power was automatically reset by the `watchdog` even if switching to `standby mode`.

I am using `RTC Backup Register` to solve this problem because I did not use RTC function.
The RTC Backup Register is a register that can not be initialized by System reset.
So I am using the RTC Backup Register as a flag to solve this problem.  
[image]

In the `L4` series, I know that `IWDG_STDBY` is supported in the `FLASH_OPTR` register.
I also tried to check if `IWDG_STDBY` exists in `STM32WB`.
But I could not find the datasheet of `STM32WB` and could not check it.
[image]

gon1332, 2019-07-29:
Thanks for the time to get our opinions.
We are using the STM32L4 series for our product.
As for your request:

> * switched off peripherals

Like @psidhu 

> * sleep duration

We use both fixed and unfixed sleep duration. Unfixed sleep duration is used e.g. when we want the device to be waken up from a radio module, when the network is awake, or a sensor that has a strange reading. OTOH, fixed sleep duration are implemented with the help of RTC.

> * wake up source

It is crucial to know who woke up the device. In most, if not all cases, more than one peripherals would wake up a device, so like @psidhu 

> * wake up time constraints if any

> * optionally: the STM32 low power modes you expect to be supported (if already known).

We are mainly using `STOP2` mode, so this is a must. Along with this, `STANDBY` and `SHUTDOWN` modes are also important.
Moreover, it would be really nice if frequency scaling was there in order to unlock `LPRUN` and `LPSLEEP` modes.

StefJar, 2019-08-22:
@KwonTae-young thx for open this topic. Since the last post has some time passed.
@erwango what is the current state of the stm32 low power/sleep implementation?
I am currently looking to implement deep sleep and wakeup via button on a stm32f412 MCU.

r2r0, 2019-08-27:
@erwango 
It would be great to support following three sleep use cases:
1. All peripherals down but (LP) timer; sleep for 1-10s; wake up from: LP timer or  I/O pin; wake up time up to ~10ms; mode STOP2. 
1. All peripherals down but: (LP) timer, BLE (periodic advertisement), LPUART; sleep for 100-1000ms; wake up from: timer,  BLE (advertising), I/O pin, LPUART RX, USB resume ; wake up time to satisfy BLE; mode STOP2.
1. Like 2 but additionally enabled USB (suspend state); sleep for 100-1000ms; wake up from: also from USB resume ; wake up time to also satisfy USB; mode STOP1.

In all cases, it would be helpfull to have possibility to configure main clock after wake-up (static onfiguration or callback). I.e. in my case I need aways to switch to the highest clock (64MHz from PLL).

Additionally there is also a deep sleep use case:
- All peripherals down, sleep time not limited, wakeup from EXTI; wake up time not critical; mode SHUTDOWN

erwango, 2019-08-27:
Thanks all for the inputs.
We've started to look at PM support for STM32. As shown by discussions in current issue this is not an easy topic and people might have different expectations, depending on application needs and targets. Besides, STM32 devices offer a wide range of possible use cases.
Since we had to start from somewhere, we decided to:
- target low power devices as a start (L4 for the very beginning)
- propose a LPTIM based kernel tick source as a first brick (so we can exit stop modes)
- build on top of that bit by bit

This is starting in following PR: https://github.com/zephyrproject-rtos/zephyr/pull/18582
This is functional but this is the very first step so obviously needs to be refined.

StefJar, 2019-10-28:
For all who needs to deal with the stm32F4xx series. I figured out a work around. It's a bit dirty but it works well for me.

I put the blueprint of the solution to my answer on StackOverflow:

https://stackoverflow.com/questions/58222886/how-to-setup-the-standby-mode-on-a-stm32f4-mcu-running-an-rtos-and-waking-it-up

@erwango I am excited to see the official solution.

FRASTM, 2020-05-19:
I have a PR https://github.com/zephyrproject-rtos/zephyr/pull/19026  for adding the low power on stm32l4r5. This is based on a low power timer LPTIM and supports SLEEP Mode 1,2,3 as STOP 0,1,2
DEEP sleep states as STANDBY and Shutdown requires a wakup through user button (or RTC auto-wakeup feature which is not implemented yet)

erwango, 2021-10-19:
@KwonTae-young I propose to close this point.
STM32 PM implementation is followed by https://github.com/zephyrproject-rtos/zephyr/issues/19755 and being implemented bits by bits along with Zephyr PM subsystem evolutions.
Please don't re-open but if needed open a new issue with detailed description of the use case not supported.