zephyrproject-rtos/zephyr #14068: Allow better control on SPI pin settings

pdunaj, 2019-03-05:
**Is your enhancement proposal related to a problem? Please describe.**
This issue concerns Nordic SoCs SPI. At the moment there is no possibility to control MISO pull per SPI instance.

**Describe the solution you'd like**
It would be great to be able to control pull for each bus pin separately. This will requite a companion change in nrfx.

**Describe alternatives you've considered**
At the moment switch in Kconfix is added to allow MISO pull control for SPI type.

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/issues/13901

5frank, 2021-12-07:
Another possible workaround is the `miso-pull-up` and `miso-pull-down` property. Example
```
&spi 0 {
    compatible = "nordic,nrf-spi";
    miso-pull-up;
    ...
}
```

gmarull, 2022-08-11:
This is now possible using pinctrl, closing.

closed 2022-08-11 by gmarull