zephyrproject-rtos/zephyr #14087: serial/stm32: uart_stm32_fifo_fill can't transmit data complete

bzd2132, 2019-03-06:
**Describe the bug**
 Enable UART interrupt support, and STM32 uart can't transmit data complete with function uart_fifo_fill.

**To Reproduce**
1. Select "Enable UART Interrupt support" int menuconfig "Device Drivers -> Serial Drivers"
2. Build application and download to target board  NUCLEO_STM32L452RE.
3. In the application it transmit 8 bytes on time, but only 2 bytes can be received from UART.

**Environment (please complete the following information):**
 OS:  Linux
 Toolchain:  Zephyr SDK,

**Additional context**
It's found the function "uart_stm32_fifo_fill" in uart_stm32.c can't transmit all the bytes, because TXE flag check in wrong time.
``` 
	while ((size - num_tx > 0) &&
	       LL_USART_IsActiveFlag_TXE(UartInstance)) {
		/* TXE flag will be cleared with byte write to DR register */

		/* Send a character (8bit , parity none) */
		LL_USART_TransmitData8(UartInstance, tx_data[num_tx++]);
	}

```
After I modified the TXE check position , then it can transmit 8 bytes complete.
```
	while (size - num_tx > 0) {
		/* TXE flag will be cleared with byte write to DR register */

		/* Send a character (8bit , parity none) */
		if(LL_USART_IsActiveFlag_TXE(UartInstance)){
			LL_USART_TransmitData8(UartInstance, tx_data[num_tx++]);
		}
	}
```

pfalcon, 2019-03-06:
uart_stm32_fifo_fill() isn't supposed to "transmit data complete". It's supposed to fill FIFO, and in the case of STM32, the FIFO depth is 1. Please read the documentation: https://docs.zephyrproject.org/latest/reference/peripherals/uart.html?highlight=uart_fifo_fill#_CPPv314uart_fifo_fillP6devicePK4u8_ti .

And it's internal API function, I'd recommend against using it.

bzd2132, 2019-03-06:
@pfalcon Thank you for your clarify.  I reference to some sample application in zephyr, and make misunderstanding of this API.

closed 2019-03-06 by bzd2132