zephyrproject-rtos/zephyr #14088: sanitycheck --coverage: Running with different -T options results in different line coverage

cinlyooi-intel, 2019-03-06:
Not sure whether it is a bug.

If I run sanitycheck --coverage with different -T options, the total lines reported by the coverage report differs, should it be the same?

To reproduce, check out 259c2409c093e7c9b2f3cb5bbb7524ce772ce8b3, run the two commands below and  compare the coverage report for `kernel/sched.c` 
```
 scripts/sanitycheck --coverage  --coverage-platform mps2_an385 -p mps2_an385  --coverage-platform qemu_x86  -p qemu_x86 --gcov-tool /opt/zephyr-sdk-0.9.5/sysroots/x86_64-pokysdk-linux/usr/bin/
i586-zephyr-elf/i586-zephyr-elf-gcov   -T tests -T samples 
```
and
```
scripts/sanitycheck --coverage  --coverage-platform mps2_an385 -p mps2_an385  --coverage-platform qemu_x86  -p qemu_x86 --gcov-tool /opt/zephyr-sdk-0.9.5/sysroots/x86_64-pokysdk-linux/usr/bin/
i586-zephyr-elf/i586-zephyr-elf-gcov   -T tests/kernel -T tests/subsys/logging 
```
I got 334 and 329 respectively

aescolar, 2019-03-06:
> the total lines reported by the coverage report differs

If you are compiling different images with different options, different parts of the code will be compiled in. So your total amount of code which could possibly be covered does actually change.

cinlyooi-intel, 2019-03-06:
@aescolar Thank you very much. It matches my expectation.

I noticed another thing that you might be able to help to explain. I was about to raise another ticket just in case it has something to do with the way zephyr reads on the gcda data. I noted that _sometimes_ the report skips reporting a line while it could not possibly do, e.g.

```
# first number is line number
# second number is the number of time the line is used

1:10: void functionA() {
2:10:     func1();
3:        func2();
4:10:     func3();
5:    }
```
total line covered = 3 because lcov/gcov did not count `func2()`. I am expecting 4 because it is clear that logically it is not possible to skip `func2()` if `functionA()` was called.

aescolar, 2019-03-06:
@cinlyooi-intel : Are you compiling without optimizations?

cinlyooi-intel, 2019-03-08:
@aescolar  Thank you very much for reminding me to check optimizations. Will do. Thanks

aescolar, 2019-03-16:
@cinlyooi-intel : Would you be happy to close this issue?

cinlyooi-intel, 2019-03-19:
Yes. Sorry for not closing earlier.

closed 2019-03-19 by cinlyooi-intel