zephyrproject-rtos/zephyr #14096: Timeslicing is broken

pdunaj, 2019-03-06:
**Describe the bug**
I created a simple debug app to test timeslicing. Two threads are created both preemptible with priority of 0. Both in the loop are printing a log and execute `k_busy_loop(100000)` (full code below). I expect that these two threads will be executed in round-robin fashion and I will see logs from both threads.

Now what I see is that at the beginning everything is fine for random time (usually from 10s to 60s) and then everything stops. When I execute from debugger I get following trace:
```
(gdb) bt
#0  k_cpu_idle () at /home/pdunaj/work/ncs/zephyr/arch/arm/core/cpu_idle.S:139
#1  0x00001f5a in _SysFatalErrorHandler (reason=<optimized out>, pEsf=<optimized out>) at /home/pdunaj/work/ncs/zephyr/arch/arm/core/sys_fatal_error_handler.c:70
#2  0x00001a42 in _oops () at /home/pdunaj/work/ncs/zephyr/arch/arm/core/swap_helper.S:380
#3  <signal handler called>
#4  0x20003958 in timeout_list ()
#5  0x00030100 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

It seems that issue is easier to replicate if I set the timeslice to 1 ms. Usually there are no logs at all. When run from debugger I see following trace:
```
#150 log_output_flush (log_output=log_output@entry=0x59fc <log_output>) at /home/pdunaj/work/ncs/zephyr/subsys/logging/log_output.c:142
#151 0x00001118 in panic (backend=<optimized out>) at /home/pdunaj/work/ncs/zephyr/subsys/logging/log_backend_rtt.c:241
#152 0x00000c5e in log_backend_panic (backend=<optimized out>) at ../../../include/logging/log_backend.h:176
#153 log_panic () at /home/pdunaj/work/ncs/zephyr/subsys/logging/log_core.c:397
#154 0x00001c40 in _Fault (esf=0x1ffffe68, exc_return=<optimized out>) at /home/pdunaj/work/ncs/zephyr/arch/arm/core/fault.c:676
#155 0x00001eae in __usage_fault () at /home/pdunaj/work/ncs/zephyr/arch/arm/core/fault_s.S:147
#156 <signal handler called>
#157 0x20003958 in timeout_list ()
#158 0x00030100 in ?? ()
```

**To Reproduce**
Steps to reproduce the behavior:
Build and run app as below. I run it on nrf52840_PCA10056.

**Expected behavior**
Scheduling must be stable.

**Impact**
Showstopper for anybody willing to use slicing and preemption.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
ncs zephyr 1d6219ffec845a43f66469354d997fbea2f283a4 from cf85618

**Additional context**
`main.c`:
```
#include <zephyr.h>                                                                                                      
#include <misc/util.h>                                                                                                   
#include <logging/log.h>                                                                                                 
                                                                                                                         
LOG_MODULE_REGISTER(sched_demo, 4);                                                                                      
                                                                                                                         
#define THREAD_COUNT 2                                                                                                   
#define STACK_SIZE 1024                                                                                                  
                                                                                                                         
static K_THREAD_STACK_ARRAY_DEFINE(stack, THREAD_COUNT, STACK_SIZE);                                                     
static struct k_thread thread[THREAD_COUNT];                                                                             
                                                                                                                         
                                                                                                                         
static void thread_fn(void *p1, void *p2, void *p3)                                                                      
{                                                                                                                        
        size_t id = (size_t)p1;                                                                                          
                                                                                                                         
        while (true) {                                                                                                   
                LOG_INF("%zu: %p", id, k_current_get());                                                                 
                k_busy_wait(100000);                                                                                     
        }                                                                                                                
}                                                                                                                        
                                                                                                                         
[... 17 lines cut ...]
CONFIG_HEAP_MEM_POOL_SIZE=4096                                                                                           
CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000                                                                                      
CONFIG_NUM_COOP_PRIORITIES=20                                                                                            
CONFIG_NUM_PREEMPT_PRIORITIES=20                                                                                         
CONFIG_TIMESLICE_SIZE=100                                                                                                
CONFIG_SPEED_OPTIMIZATIONS=y                                                                                             
CONFIG_ASSERT=y                                                                                                          
CONFIG_ASSERT_LEVEL=2                                                                                                    
CONFIG_SYS_POWER_MANAGEMENT=n                                                                                            
CONFIG_LOG=y                                                                                                             
CONFIG_LOG_DEFAULT_LEVEL=2                                                                                               
CONFIG_LOG_PRINTK=y                                                                                                      
CONFIG_LOG_IMMEDIATE=y                                                                                                   
CONFIG_LOG_BACKEND_RTT=y                                                                                                 
CONFIG_LOG_BACKEND_RTT_MODE_DROP=y                                                                                       
CONFIG_LOG_BACKEND_RTT_MESSAGE_SIZE=128                                                                                  
CONFIG_LOG_BACKEND_RTT_BUFFER=0                                                                                          
CONFIG_LOG_BACKEND_SHOW_COLOR=y                                                                                          
CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP=y                                                                                    
CONFIG_CONSOLE=y                                                                                                         
CONFIG_USE_SEGGER_RTT=y                                                                                                  
CONFIG_SEGGER_RTT_BUFFER_SIZE_UP=4096                                                                                    
CONFIG_RTT_CONSOLE=y                                                                                                     
CONFIG_UART_CONSOLE=n
```

pdunaj, 2019-03-06:
I tried to look what may cause it and added some debug traces in timeout. I do not see however crash coming from there. I thought that the problem can be caused by logger and so I removed it. Without logger I have not seen the issues. Then I tried following and issue hit again:
```
static void thread_fn(void *p1, void *p2, void *p3)                                                                      
{                                                                                                                        
        size_t id = (size_t)p1;                                                                                          
                                                                                                                         
        while (true) {                                                                                                   
                k_mutex_lock(&lock, K_FOREVER);                                                                          
                k_busy_wait(10000);                                                                                      
                k_mutex_unlock(&lock);                                                                                   
                k_busy_wait(100000);                                                                                     
        }                                                                                                                
}
```
Logger is using locks that in some cases may not be available. It seems the problem is with how timeslicing and locking interact with each other.

pdunaj, 2019-03-06:
Btw - issue does not seem to replicate itself if I disable timeslice and yield instead.

pdunaj, 2019-03-06:
Hi @andyross , is this anything you know of?

pdunaj, 2019-03-06:
@pizi-nordic , @cvinayak fyi

andyross, 2019-03-06:
There are known stability issues on ARM with userspace enabled, does your app have it enabled?  There is some discussion in #12553.  #12559 is one that we can reproduce in simulation using mps2_an385 which sounds similar -- 1ms tick rate + tickless + userspace.

andyross, 2019-03-12:
I'm not able to reproduce this.  So... to begin with I have no idea have the Segger RTT stuff works.  I'm just copying the binary to the device via MBED and reading the output on the UART console.

And when I do that (i.e. take your prj.conf and comment out anything with RTT or SEGGER in it and enabled UART_CONSOLE), I get the following, which... I gotta be honest that looks pretty clearly like a bug in the logging subsystem.  Maybe the assertion is not being reported cleanly out of RTT?  (Needless to say, when I replace LOG_INF() with printk() it works as expected):

```
***** Booting Zephyr OS v1.14.0-rc1-1044-g8104c6c7dfc6 *****
[00:00:00.004,425] <inf> sched_demo: Thread 0 created
[00:00:00.010,101] <inf> sched_demo: Thread 1 created
[00:00:00.015,777] <inf> sched_demo: Start
[00:00:00.020,507] <inf> sched_demo: 0: 0x20000000
[00:00:00.118,164] <inf> sched_demo: 1: 0x2000006c
[00:00:00.223,449] <inf> sched_demo: 0: 0x20000000
[00:00:00.321,044] <inf> sched_demo: 1: 0x2000006c
[00:00:00.426,086] <inf> sched_demo: 0: 0x20000000
[00:00:00.523,620] <inf> sched_demo: 1: 0x2000006c
[00:00:00.628,631] <inf> sched_demo: 0: 0x20000000
[00:00:00.726,196] <inf> sched_demo: 1: 0x2000006c
[00:00:00.831,207] <inf> sched_demo: 0: 0x20000000
[00:00:00.928,741] <inf> sched_demo: 1: 0x2000006c
[00:00:01.033,721] <inf> sched_demo: 0: 0x20000000
[00:00:01.131,286] <inf> sched_demo: 1: 0x2000006c
[00:00:01.236,236] <inf> sched_demo: 0: 0x20000000
[00:00:01.333,770] <inf> sched_demo: 1: 0x2000006c
[00:00:01.390,136] <inf> sched_demo: 0: 0x20000000
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ /home/ajross/zephyr/subsys/logging/log_output.c:101
```

As to the mutex example, what symptoms are you seeing exactly (there's no console output in the function you posted)?  It seems to work for me, but note that the time in that test between the calls to k_mutex_lock() (which is a blocking operations and will change priorities and cause context switches) is (if I'm counting right) exactly one slice time, so almost no timeslicing is going to be happening there.

Can you try to pare this down to a smaller test that shows the error, and post the whole project (with instructions I can duplicate with the uart console and MBED flashing, heh)?

carlescufi, 2019-03-26:
@pdunaj can you give an update on this one?

pdunaj, 2019-03-27:
Hi @carlescufi , I will try to replicate it again. I am positive I have similar issue when I used locks only (no RTT logging) but let me try again, I should have the code around.

rljordan, 2019-04-02:
@pdunaj  Were you able to reproduce this?

pdunaj, 2019-04-02:
Hi @rljordan , have no time yet but will get back to it this week.

pdunaj, 2019-04-03:
I checked it today using ncs zephyr `1937b41459ca722848abe88cf84067945e8cf0c2` (upstream  `d3bb3cf`) and pca10040.

Issue is still present but it was more difficult to replicate. Logs seems to be reworked so characters are printed synchronously. When I added log issue was not replicable for the time I run test. When I set timeslice to some larger value I was also unable to replicate the problem.

Since logs are fixed crash can be seen:
```
***** Booting Zephyr OS v1.13.99-ncs1-5783-ge9f04d5cfe99 *****
[00:00:03.273,223] <inf> sched_demo: Thread 0 created
[00:00:03.901,977] <inf> sched_demo: Thread 1 created
[00:00:04.530,670] <inf> sched_demo: Start
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x0
***** Hardware exception *****
Current thread ID = 0x20000088
Faulting instruction address = 0x492a
Fatal fault in thread 0x20000088! Aborting.
***** HARD FAULT *****
  Fault escalation (see below)
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x0
***** Hardware exception *****
Current thread ID = 0x20000088
Faulting instruction address = 0x428c
Fatal fault in ISR! Spinning...
```

The code is as above apart from the thread_fn which is:
```
static void thread_fn(void *p1, void *p2, void *p3)                                                                      
{                                                                                                                        
        size_t id = (size_t)p1;                                                                                          
        static struct k_mutex lock;                                                                                      
                                                                                                                         
        while (true) {                                                                                                   
                k_mutex_lock(&lock, K_FOREVER);                                                                          
                k_busy_wait(1000);                                                                                       
                k_mutex_unlock(&lock);                                                                                   
                k_busy_wait(100000);                                                                                     
        }                                                                                                                
}
```
Following time slice is used: `CONFIG_TIMESLICE_SIZE=1`

pdunaj, 2019-04-03:
My guess is that this issue is caused by the access to mutex around timeslice (recheduling event in general maybe).

I changed code a bit to replicate bug with higher value of timeslice.
```
#define THREAD_COUNT 10                                                                                                  
#define STACK_SIZE 1024                                                                                                  
                                                                                                                         
static K_THREAD_STACK_ARRAY_DEFINE(stack, THREAD_COUNT, STACK_SIZE);                                                     
static struct k_thread thread[THREAD_COUNT];                                                                             
                                                                                                                         
                                                                                                                         
static void thread_fn(void *p1, void *p2, void *p3)                                                                      
{                                                                                                                        
        size_t id = (size_t)p1;                                                                                          
        static struct k_mutex lock;                                                                                      
                                                                                                                         
        while (true) {                                                                                                   
                k_mutex_lock(&lock, K_FOREVER);                                                                          
                k_mutex_unlock(&lock);                                                                                   
        }                                                                                                                
}                                                                                                                        
                                                                                                                         
void main(void)                                                                                                          
{                                                                                                                        
        for (size_t i = 0; i < ARRAY_SIZE(thread); i++) {                                                                
                k_thread_create(&thread[i],                                                                              
                                stack[i], STACK_SIZE,                                                                    
                                (k_thread_entry_t)thread_fn,                                                             
                                (void *)i, NULL, NULL,                                                                   
                                K_PRIO_PREEMPT(0), 0, i);                                                                
                LOG_INF("Thread %zu created", i);                                                                        
        }                                                                                                                
        LOG_INF("Start");                                                                                                
}
```

andyross, 2019-04-03:
For clarity: this still requires the log subsystem with the segger backend to reproduce?  Or can you make it happen with printk?

pdunaj, 2019-04-03:
Hi @andyross , you don't need any logging, only mutex lock/unlock in started threads. I placed the snippet that seems to replicate the issue right away (https://github.com/zephyrproject-rtos/zephyr/issues/14096#issuecomment-479434661). The only logged messages are from main thread that is cooperative. All started started threads will run after all code from main is done.

andyross, 2019-04-03:
OK, I can make something happen.

FWIW: the problem you were seeing with that specific code is actually a test bug.  You're missing a call to k_mutex_init() on the lock object.  So eventually a thread will be sliced away while the mutex is held, causing future threads to pend themselves inside the k_mutex_lock() call, at which point they will crash because the wait_q is uninitialized.

But even fixing that, I eventually see failures in the scheduler, where it seems like threads are being scheduled again (or maybe twice) after having been pended.   I actually did find some minor races in the mutex code, but they aren't the root cause.

andyross, 2019-04-03:
Heh, actually it was just user error.  I stuff the k_mutex_init() into the thread function instead of main, so each thread would clobber the mutex state and thus multiply lock it.  Doing it right works for me on all platforms I tried.

The race I found in k_mutex has a patch up, but on further reflection the code as written wouldn't have been actually racy on a uniprocessor platform, so it's unrelated.

I think we're fine here.  I'm going to close this bug given the impending release.  Please reopen if needed with details and test case.

closed 2019-04-03 by andyross

pdunaj, 2019-04-04:
Hi @andyross , thanks for pointing this one out. It was probably just that as I am unable to replicate other issues.