zephyrproject-rtos/zephyr #14105: Race condition in k_delayed_work_submit_to_queue()

pizi-nordic, 2019-03-06:
From k_delayed_work_submit_to_queue():
```
        /* Submit work directly if no delay.  Note that this is a                                                                         
         * blocking operation, so release the lock first.                                                                                 
         */                                                                                                                               
        if (!delay) {                                                                                                                     
                k_spin_unlock(&work_q->lock, key);                                                                                        
                k_work_submit_to_queue(work_q, &work->work);                                                                              
                return 0;                                                                                                                 
        }                                                                                                                                 
```

The lock is released before submitting work to the queue. As result the k_delayed_work_cancel() called in the very moment between  k_spin_unlock() and k_work_submit_to_queue() will return 0 signalling that the work was cancelled. However it will be eventually submitted to the queue and executed.

pizi-nordic, 2019-03-06:
FYI: @nashif, @carlescufi, @pdunaj

andyross, 2019-03-06:
How is that a bug?  Calling cancel() before submit() has returned is a clear error, no?

pizi-nordic, 2019-03-07:
The k_delayed_work_*() API is locked, so concurrent calls to submit and cancel are serialized. This means that you either get 0 or -EINVAL from cancel informing that the work was cancelled or not. 
In the mentioned case, the cancel return value of 0 indicating that the work was cancelled, but it will be submitted and executed.

If you think that concurrent submit and cancel is something we should not care about, feel free to remove locking which only occupies space and consumes CPU resources in such case.

andyross, 2019-03-07:
The locking strategy is (modulo the one other bug you found where in SMP we have one lock trying to cover two queues because the original lock was global) the same as it's always been.  It could surely be improved.

But no, you can't cancel something until the submit function has returned.

pizi-nordic, 2019-03-08:
Please either remove locking and specify in the docs that this needs external locking or eliminate race condition.

carlescufi, 2019-03-08:
>How is that a bug? Calling cancel() before submit() has returned is a clear error, no?

I think this depends on whether we consider the API to be thread-safe. If it is, I would expect calls to be atomic and sequential in effect, meaning that cancelling while we are still submitting should cancel the current submission or error out as described here. i.e. I want to be able to  submit() and cancel() at will from different contexts and the last one called will take effect. 
@ioannisg thoughts?

pizi-nordic, 2019-03-08:
FYI: @pdunaj

pdunaj, 2019-03-08:
> you can't cancel something until the submit function has returned

I agree with @andyross here. The other (similar) race would be when concurrent cancel executes just before submission does anything. The end effect would be exactly the same - cancel succeeded while work is submitted. This is just a race.

> I  want to be able to submit() and cancel() at will from different contexts and the last one called will take effect. 

@carlescufi true. The common case may be when I have work that periodically resubmits itself while at the same time I have some code that cancels it. And for that I have a comment.

According to `k_delayed_work_cancel` documentation it returns `-EINPROGRESS` to inform that work is already pending (I guess author meant actually being executed) and so cannot be canceled. The problem is that this error code is not used in the implementation, meaning caller will have no idea anyway if work was correctly canceled or not. And so will have to use other means to make sure work should not execute anymore. Our application does that.

pdunaj, 2019-03-08:
btw; documentation states

> A delayed work item can only be canceled while its countdown is still underway.

Meaning that once timeout is done you cannot do anything with the work. In some point of view submitting a delayed work with 0 delay could be seen as submitting something that cannot be canceled anyway.

pizi-nordic, 2019-03-08:
@pdunaj: There is no issue with the fact that at certain states work is not cancelled.
The problem is that k_delayed_work_cancel() says that work was cancelled, which is not true in the presented case.

andyross, 2019-03-08:
OMG.  Are we really going to do this?  I'll add a line to the docs.

It's not a threadsafety issue, it's just a state bug.  A work item that has not returned from a submit call is, well, unsubmitted.  The cancel operation is allowed to assume (again, remember Zephyr doesn't typically do defensive coding and this isn't a userspace API that needs checking) that the item passed to it is in a valid state.  Are there seriously any real world circumstances where we'd see wrong behavior that isn't trivially fixed by "don't do that"?

PR #14202 (2019-03-08): kernel/work_q: Clarify docs for k_delayed_work_cancel()
You can't cancel what hasn't been submitted.  Clarification added
following minor bike shed in github.  Fixes #14105
commit d7ae2a81 (2019-03-08): kernel/work_q: Clarify docs for k_delayed_work_cancel()

You can't cancel what hasn't been submitted.  Clarification added
following minor bike shed in github.  Fixes #14105
closed 2019-03-08 by nashif