zephyrproject-rtos/zephyr #14121: gaps between app shared memory partitions can waste a lot of space

andrewboie, 2019-03-06:
**Describe the bug**

On systems which require power-of-two size/alignment, we are not ordering the app shared memory partitions in an optimal way, which results in a lot of memory being wasted.

A good example is running the app memory size script from #14120 on the kernel produced by tests/subsys/jwt on mps2_an385, notice the huge gap in between z_libc_partition and k_mbedtls_partition:

```
$ python3 ~/projects/zephyr4/scripts/appmem_report.py zephyr/zephyr.elf 
Userspace memory partition report:
Partition                     Total       Used        Wasted      Efficiency
----------------------------------------------------------------------------
ztest_mem_partition           0x40        0x34        0xc         81.25%
  gap 0x0         
z_libc_partition              0x20        0x4         0x1c        12.50%
  gap 0x7fa0      
k_mbedtls_partition           0x8000      0x7fe0      0x20        99.90%

total                         0x10000     0x8018      0x7fe8      50.04%
```

Ordering the app shared memory partitions in reverse order of total size should greatly reduce these gaps.

We don't know the full sizes of these partitions when gen_app_partitions.py runs since the zephyr_prebuilt.elf has not been linked yet, but since we are examining the individual object files we might be able to assemble a total anyway. Or we could somehow figure out a way to get the linker to reorganize these partitions for us, but that might require them being placed in individual sections.

andrewboie, 2019-03-06:
If we try to solve this in gen_app_partitions.py, any solution for this will need to consider the memory used not only in .o files scanned, but libraries added via the --library option to gen_app_partitions.py

galak, 2019-03-07:
The linker has SORT_BY_ALIGNMENT which might allow us to sort the partitions at link time.  Just requires that each partition ends up in its own section.  Not sure what the implications or requirements in the linker script are related to that.

I know we need to keep the partitions together, that should be easy enough.  @ioannisg had some additional constraints we needed to ensure.

Maybe first step is just to get each partition in its own linker section and ensuring the proper constraints.

dcpleung, 2019-03-09:
SORT_BY_ALIGNMENT can only be applied to input sections. With app shared memory, the sorting needs to be done to across different output sections. The issue is due to ARM's MPU alignment requirements, where a 64KB region must be aligned on 64KB. So `k_mbedtls_partition` in the description has to be aligned at 32KB since its partition is 32KB. Hence there is a big gap between `z_libc_partition` and `k_mbedtls_partition` because of all the padding.

PR #14243 (2019-03-13): linker: sort app shared mem partition by alignment
If CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT is enabled,
the app shared memory partition may cause waste of memory
due to the need for padding.

For example, tests/subsys/jwt and board mps2_an385:

```
  z_test_mem_partition: addr 0x20000000, size 52
  z_libc_partition: addr 0x20000040, size 4
  k_mbedtls_partition : addr 0x20008000, size 32736

    ending at 0x2000ffff, taking up 65536 bytes
```

With power-of-two size and alignment requirement,
k_mbedtls_partition takes up 32KB memory and needs to be
aligned on 32KB boundary. If the above p [... cut]
commit 212ec9a2 (2019-03-13): linker: sort app shared mem partition by alignment

If CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT is enabled,
the app shared memory partition may cause waste of memory
due to the need for padding.

For example, tests/subsys/jwt and board mps2_an385:

  z_test_mem_partition: addr 0x20000000, size 52
  z_libc_partition    : addr 0x20000040, size 4
  k_mbedtls_partition : addr 0x20008000, size 32736

    ending at 0x2000ffff, taking up 65536 bytes

With power-of-two size and alignment requirement,
k_mbedtls_partition takes up 32KB memory and needs to be
aligned on 32KB boundary. If the above part [... cut]
closed 2019-03-13 by nashif