zephyrproject-rtos/zephyr #14123: Particle boards need a board initialization module for antenna configuration

pabigot, 2019-03-06:
The Particle boards have a SKY13351 SPDT switch that selects between a PCB antenna and an external uFL antenna.  On the Xenon the Zephyr board initialization probably leaves this in an unconfigured mode, resulting in neither antenna being connected and so very poor RF performance.  Driving P0.24 high enables the PCB antenna, which adds more than 10 dB.  Driving P0.25 high enables the uFL antenna.

For Xenon to be useful as a Bluetooth/6LowPAN device it needs a board initialization function that configures P0.24 for write and sets the GPIO.  Whether API to select the external antenna is necessary is up to whoever solves this.

I discovered this while doing performance analysis of my non-Zephyr framework, and have not looked at Argon or Boron.  By the schematic they both have the same switch, but the pins to select the antenna are different on Argon, and Boron has a single signal that also needs to be driven.

pabigot, 2019-03-06:
@mike-scott Assigning to you because you seem to be actively using Xenon for real stuff, probably would care about poor RF performance, and may be able to address this before I get back to Zephyr whenever that is.  If it's still unresolved when I do come back I'll deal with it then.

mike-scott, 2019-03-06:
@pabigot Funny story about my using the Particle boards for real stuff: I was trying to get a large # of BLE 6lowpan and OpenThread devices running to do gateway load tests (perform FOTA updates on all devices at once, etc) when I started stumbling across the various issues with the Particle boards.   Typical Zephyr :P  Start one thing and end up fixing 2-3 things along the way.

I'll see what I can do.  I was also looking at the modem driver for Boron (u-blox sara r41xx).  But, my work has also been very busy lately.

tautologyclub, 2019-04-11:
@mike-scott See #15355

closed 2019-05-30 by nashif