zephyrproject-rtos/zephyr #14127: netusb: TX path doesn't work in RNDIS driver

aurel32, 2019-03-06:
**Describe the bug**
When using the RNDIS driver communication with the host doesn't work. Enabling some debugging shows that the driver is able to receive data, but not to send it. Enabling more debugging shows that the RNDIS driver try to do more USB bulk IN transfer while the current one hasn't finished.

**To Reproduce**
Build any net sample with the RNDIS driver for the interface. Tested on Nucleo L432KC and SAM E70 Xplained.

**Expected behavior**
IP communication works out of the box.

**Impact**
The driver is unusable.

**Screenshots or console output**
```
uart:~$ net ping 192.0.2.2
S[00:00:27.310,000] <dbg> usb_net.netusb_send: Send pkt, len 42
[00:00:27.310,000] <dbg> usb_rndis.rndis_send: send pkt 0x20003968 len 42
[00:00:27.310,000] <dbg> usb_rndis.net_pkt_hexdump: <
[00:00:27.310,000] <dbg> usb_rndis: 
ff ff ff ff ff ff 00 00 |........
5e 00 53 00 08 06       |^.S...  
[00:00:27.310,000] <dbg> usb_rndis: 
00 01 08 00 06 04 00 01 |........
00 00 5e 00 53 00 c0 00 |..^.S...
02 01 00 00 00 00 00 00 |........
c0 00 02 02             |....    
[00:00:27.310,000] <dbg> usb_rndis.rndis_hdr_add: type 1 len 86 payload offset 36 payload len 42
[00:00:27.310,000] <dbg> usb_net: USB <
01 00 00 00 56 00 00 00 |....V...
24 00 00 00 2a 00 00 00 |$...*...
00 00 00 00 00 00 00 00 |........
00 00 00 00 00 00 00 00 |........
00 00 00 00 00 00 00 00 |........
00 00 00 00 ff ff ff ff |........
ff ff 00 00 5e 00 53 00 |....^.S.
08 06 00 01 08 00 06 04 |........
[00:00:27.310,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x82, len 64
[00:00:27.310,000] <dbg> usb_net: USB <
00 01 00 00 5e 00 53 00 |....^.S.
c0 00 02 01 00 00 00 00 |........
00 00 c0 00 02 02       |......  
[00:00:27.310,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x82, len 22
[00:00:27.310,000] <err> usb_dc_stm32: Unable to write ep 0x82 (-16)
ent a ping to 192.0.2.2
[00:00:27.310,000] <wrn> usb_net: Error writing to ep 0x82 ret -16
Ping timeout
[00:00:27.310,000] <err> usb_dc_stm32: Unable to write ep 0x82 (-16)
[00:00:27.310,000] <wrn> usb_net: Error writing to ep 0x82 ret -16
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Debian
 - Commit SHA: 35b7fde8a60b55ca0806a50e5b8cb4cca83f1b84

aurel32, 2019-03-06:
It looks like the RNDIS driver should be switch to use the transfer API instead of reinventing the wheel with the `try_write` function. Something similar to ccf0b46a3120afd8156dc183eb3149c82cd24d4c.

finikorg, 2019-03-11:
> It looks like the RNDIS driver should be switch to use the transfer API instead of reinventing the wheel with the `try_write` function. Something similar to [ccf0b46](https://github.com/zephyrproject-rtos/zephyr/commit/ccf0b46a3120afd8156dc183eb3149c82cd24d4c).

`try_write()` was before new USB API, so it was not reinvented. However I have added patch moving to new API here:  PR #13985

PR #13985 (2019-03-13): USB: netusb: Cleanup and switch to USB transfer API
Cleanup and fixes.

Fixes #14127 
Fixes #14416
commit db311047 (2019-03-13): usb: rndis: Use usb_transfer() for data transfer

Use USB transfer API for data transfer. Simplify notification count,
removing delayed work, use usb_transfer() logic instead.

Fixes #14127
closed 2019-03-13 by galak