zephyrproject-rtos/zephyr #14139: nsim failed in tests/subsys/jwt/libraries_encoding

vonhust, 2019-03-07:
**Describe the bug**
arc nsim_em and nsim_sem failed in tests/subsys/jwt/library_encoding.

It's caused by the wrong calculation of _timer_int_handler of  arcv2_timer0 driver

The address of _timer_int_handler in zephyr.elf and zephyr_prebuilt.elf are different. (It should be same).
In tests/kernel , there are no problems, ,the address of _timer_int_handler in zephyr.elf and zephyr_prebuilt is the same.

I compared the map files of zephyr.elf and zephyr_prebuilt.elf and found there are difference in linking of some functions in libgcc.a

***info of zephyr_prebuilt.elf***
text            0x0000000000000000    0x1383c
                0x0000000000000000                _image_rom_start = .
                0x0000000000000000                _image_text_start = .
 *(.exc_vector_table)
 .exc_vector_table
                0x0000000000000000       0x40 arch/arch/arc/core/libarch__arc__core.a(vector_table.c.obj)
                0x0000000000000000                _VectorTable
 *(.exc_vector_table.*)
 *(.gnu.linkonce.irq_vector_table)
 .gnu.linkonce.irq_vector_table
                0x0000000000000040       0x10 libzephyr.a(isr_tables.c.obj)
                0x0000000000000040                _irq_vector_table
 *(.text)
 .text          0x0000000000000050      0x378 /opt/zephyr-sdk/sysroots/arc-zephyr-elf/usr/lib/arc-zephyr-elf/6.2.1/em4_dmips/libgcc.a(_udivdi3.o)
                0x0000000000000050                __udivdi3
 .text          0x00000000000003c8       0xf8 /opt/zephyr-sdk/sysroots/arc-zephyr-elf/usr/lib/em4_dmips/libc.a(lib_a-memcmp-bs-norm.o)

***info of zephyr.elf***
text            0x0000000000000000    0x138a8
                0x0000000000000000                _image_rom_start = .
                0x0000000000000000                _image_text_start = .
 *(.exc_vector_table)
 .exc_vector_table
                0x0000000000000000       0x40 arch/arch/arc/core/libarch__arc__core.a(vector_table.c.obj)
                0x0000000000000000                _VectorTable
 *(.exc_vector_table.*)
 *(.gnu.linkonce.irq_vector_table)
 .gnu.linkonce.irq_vector_table
                0x0000000000000040       0x10 CMakeFiles/kernel_elf.dir/isr_tables.c.obj
                0x0000000000000040                _irq_vector_table
 *(.text)
 .text          0x0000000000000050       0x30 /opt/zephyr-sdk/sysroots/arc-zephyr-elf/usr/lib/arc-zephyr-elf/6.2.1/em4_dmips/libgcc.a(_millicodethunk_st.o)
                0x0000000000000050                __st_r13_to_r25
                0x0000000000000054                __st_r13_to_r24
                0x0000000000000058                __st_r13_to_r23
                0x000000000000005c                __st_r13_to_r22
                0x0000000000000060                __st_r13_to_r21
                0x0000000000000064                __st_r13_to_r20
                0x0000000000000068                __st_r13_to_r19
                0x000000000000006c                __st_r13_to_r18
                0x0000000000000070                __st_r13_to_r17
                0x0000000000000074                __st_r13_to_r16
                0x0000000000000078                __st_r13_to_r15
 .text          0x0000000000000080       0x3c /opt/zephyr-sdk/sysroots/arc-zephyr-elf/usr/lib/arc-zephyr-elf/6.2.1/em4_dmips/libgcc.a(_millicodethunk_ret.o)
                0x0000000000000080                __ld_r13_to_r25_ret
                0x0000000000000084                __ld_r13_to_r24_ret
                0x0000000000000088                __ld_r13_to_r23_ret
                0x000000000000008c                __ld_r13_to_r22_ret
                0x0000000000000090                __ld_r13_to_r21_ret
                0x0000000000000094                __ld_r13_to_r20_ret
                0x0000000000000098                __ld_r13_to_r19_ret
                0x000000000000009c                __ld_r13_to_r18_ret
                0x00000000000000a0                __ld_r13_to_r17_ret
                0x00000000000000a4                __ld_r13_to_r16_ret
                0x00000000000000a8                __ld_r13_to_r15_ret
                0x00000000000000ac                __ld_r13_to_r14_ret
 .text          0x00000000000000bc      0x378 /opt/zephyr-sdk/sysroots/arc-zephyr-elf/usr/lib/arc-zephyr-elf/6.2.1/em4_dmips/libgcc.a(_udivdi3.o)
                0x00000000000000bc                __udivdi3

The root cause is unknown,

**To Reproduce**
Steps to reproduce the behavior:
0. cd tests\subsys\jwt\library_encoding
1. mkdir build; cd build
2. cmake -DBOARD=nsim_em ..
3. make run
4. see error
5. comapre the value  (the first entry)in isr_table.c  and the value of _timer_int_handler in zephyr.map.

**Expected behavior**
If the address caculation is ok.  The test will pass as shown below
```
***** Booting Zephyr OS v1.7.99-15745-g00c0cb066c5f *****
Running test suite lib_jwt_test
===================================================================
starting test - test_jwt
JWT:
eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCJ9.eyJhdWQiOiJpb3Qtd29yay0xOTk0MTkiLCJleHAiOjE1MzAzMTIwMjYsImlhdCI6MTUzMDMwODQyNn0.VHKDT-dvWW3UDBGokkttP7P6DNDnd4jjz_wLsOvn9RVBoDAAVG1mQnlROPtOHyRTotWzFYf2g02-VyZoh326yDFXM-66U7ycqyUsJdWBZWYqub6iZ17l0hjodoSpZBJqPB6F9c8Q5u9x1gWVw2l2XNASQyTDIZ3db_GiQbx1SYnNtyxsospoKClIebee-EQkcut9jBPNyiKrzZRPuX0xCiPJYTumHdHSp0UPIDIij2hJl_tacXs-hz_ckqT5HZvJPsxgJCeXz3tP7IlgbeGbPFMFmaYNX4UxYD4XS10MPXUK4_qp8wK4EqjT2YBNF7xXi_sjWMj7isc0i0YSIl6_nQ
len: 459
PASS - test_jwt
===================================================================
Test suite lib_jwt_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

andrewboie, 2019-03-11:
Might be a symptom of something very bad going on -- there's supposed to be no difference in symbol offsets between zephyr.elf and zephyr_prebuilt.elf. I'll take a look.

andrewboie, 2019-03-14:
As part of working on this I'm going to try to implement some test that proves memory addresses are not shifting between zephyr_prebuilt.elf and zephyr.elf, this is something we have seen before (with various root causes)

andrewboie, 2019-03-14:
I've reproduced this and indeed, the zephyr final kernel binary is pulling in all this __st_XXX_to_YYY and __ld_XXX_to_YYY_ret functions. They were not present in zephyr_prebuilt.elf. Every function after them is getting shifted, starting with __udivdi3.

I don't comprehend yet what these "millicodethunk" functions do.

Presumably, these are getting pulled in by functions that are only present in zephyr.elf and not zephyr_prebuilt.elf. I'm still investigating but I believe this is just the functions generated by gperf. What I don't understand yet is why it's just this test that is having problems.

andrewboie, 2019-03-14:
From the GCC manual:

```
       -mno-millicode
           When optimizing for size (using -Os), prologues and epilogues
           that have to save or restore a large number of registers are
           often shortened by using call to a special function in libgcc;
           this is referred to as a millicode call.  As these calls can pose
           performance issues, and/or cause linking issues when linking in a
           nonstandard way, this option is provided to turn off millicode
           call generation.
```

Now I could just add this to ARC's cflags...but it doesn't make sense that these are there in zephyr.elf and not zephyr_prebuilt.elf

PR #14545 (2019-03-15): userspace: don't modify optimization options
The gperf hashtables and support functions are created
after the initial zephyr_prebuilt.elf is created, using
data found within it.

A fixed-sized amount of memory is reserved for the
program text of these support functions, as their true
size isn't possible to predict in advance and we don't
want memory addresses after them to shift. To minimize
the amount of space reserved, it seemed reasonable to
hard-code -Os.

However, on ARC, building with -Os can cause various
millicode functions from libgcc to be included in the
binary which would not be pr [... cut]
commit a514898e (2019-03-15): userspace: don't modify optimization options

The gperf hashtables and support functions are created
after the initial zephyr_prebuilt.elf is created, using
data found within it.

A fixed-sized amount of memory is reserved for the
program text of these support functions, as their true
size isn't possible to predict in advance and we don't
want memory addresses after them to shift. To minimize
the amount of space reserved, it seemed reasonable to
hard-code -Os.

However, on ARC, building with -Os can cause various
millicode functions from libgcc to be included in the
binary which would not be p [... cut]
closed 2019-03-15 by galak